aws/amazon-freertos #482: Syntax error in aws_iot_network_manager.c (feature/ble-beta branch)

bhusang, 2019-04-04:
**Describe the bug**
The following code in aws_iot_network_manager.c does not compile. This is in the feature/ble-beta branch and you need to also define IOT_BLE_ENABLE_WIFI_PROVISIONING = 0

line 356
    bool ret = true;
    static const WIFINetworkParams_t network =
    {
        .pcSSID           = clientcredentialWIFI_SSID;
        .ucSSIDLength     = sizeof(clientcredentialWIFI_SSID);
        .pcPassword       = clientcredentialWIFI_PASSWORD;
        .ucPasswordLength = sizeof(clientcredentialWIFI_PASSWORD);
        .xSecurity        = clientcredentialWIFI_SECURITY;
    };

    /* Setup parameters. */

**System information**
- VSCode
- MacOS
- Amazon FreeRTOS v1.4.7-438-g48c8c93d
- Project [Shadow]
- code snippet above

**Expected behavior**
Code should compile cleanly when IOT_BLE_ENABLE_WIFI_PROVISIONING = 0

**Fix**
    bool ret = true;
    static const WIFINetworkParams_t network =
    {
        .pcSSID           = clientcredentialWIFI_SSID,
        .ucSSIDLength     = sizeof(clientcredentialWIFI_SSID),
        .pcPassword       = clientcredentialWIFI_PASSWORD,
        .ucPasswordLength = sizeof(clientcredentialWIFI_PASSWORD),
        .xSecurity        = clientcredentialWIFI_SECURITY
    };

    /* Setup parameters. */

abhidixi11, 2019-04-04:
@bhusang 
Thanks for pointing it out we will fix it soon.

abhidixi11, 2019-04-04:
Pushed changes : https://github.com/aws/amazon-freertos/pull/483

closed 2019-04-04 by abhidixi11