aws/amazon-freertos #486: WIFI_Scan - unnecessary clearing STARTED_BIT

chegewara, 2019-04-04:
Hi,
in esp32 version of WIFI_Scan there is unnecessary `STARTED_BIT` cleared in here:
https://github.com/aws/amazon-freertos/blob/feature/ble-beta/lib/wifi/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_wifi.c#L497

this is blocking for example WIFI_Provision in this line:
https://github.com/aws/amazon-freertos/blob/feature/ble-beta/lib/wifi/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_wifi.c#L218

Im not sure it is only in ble-test branch, because in master this issue does not exists.

abhidixi11, 2019-04-05:
Hello @chegewara,
FreeRTOS code does not use WIFI_Provision() , we will look into it internally and update the issue when we have the answer.

ravibhagavandas, 2019-04-14:
Hi @chegewara
We looked at it and I agree with you that the STARTED_BIT  should not be cleared. We will fix this soon.

closed 2019-04-14 by chegewara