aws/amazon-freertos #493: [question] option to turn off auto-connect wifi

chegewara, 2019-04-05:
Hi, is there option to stop auto-reconnect wifi even if wifi credentials are correct? I am using `WIFI_Disconnect` but before i can do something app is reconnecting to AP.
Thanks.

abhidixi11, 2019-04-06:
Hello @chegewara,
1. Which branch are you referring to?  ble-beta or master?
2. Which app you are referring to ?
Thanks.

chegewara, 2019-04-06:
Hi,
1. modified ble-beta branch
2. this app should be enough to reproduce this issue
https://github.com/chegewara/esp32-AWSFreeRTOS-wifi-provisioning-demo

From awsfreertos i am using basically only main.c, without apps from DEMO_RUNNER_RunDemos

abhidixi11, 2019-04-06:
Hello @chegewara,
Thanks for quick response. I didn't see the edit earlier about WIFI_NetworkAdd().  We will look into it when we get a chance, since ble-beta is still in 'beta'  and we are actively working on it,  it may take some time, due to other other priorities.  We will update once we find more information.
Thanks.

chegewara, 2019-04-06:
Thanks,
i can try to test against master and see if this problem exists in master too if this will help.

abhidixi11, 2019-04-06:
Hello @chegewara,
Yes, if you can verify it, it will be great. 
Thanks!

chegewara, 2019-04-07:
Sorry, but it cant be tested against master branch. This is the reason:
https://github.com/aws/amazon-freertos/blob/master/lib/wifi/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_wifi.c#L530

`WIFI_NetworkAdd` is not implemented yet.

abhidixi11, 2019-04-07:
Yes, that's correct.
AwsIotNetworkManager is new feature in ble-beta and WIFI_NetworkAdd  is the new update from ESP. 
We will look into it as time permits and update the the issue. 
Thanks for your patience!

chegewara, 2019-04-07:
Thanks for support.
Its not very urgent, because i still need some time before i deliver full app to the client.

From this log `WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 8` i would say that NetworkManager should not try to reconnect when disconnect reason is 8 (`WIFI_REASON_ASSOC_LEAVE`).

chegewara, 2019-04-07:
Further investigations shows me that the issue is because of this task:
https://github.com/aws/amazon-freertos/blob/feature/ble-beta/demos/common/wifi_provisioning/aws_wifi_connect_task.c#L79

which is also described in here:
https://github.com/aws/amazon-freertos/tree/feature/ble-beta/demos/common/wifi_provisioning#provisioning-sequence

abhidixi11, 2019-04-08:
Hell @chegewara ,

The WiFi Connect task is designed to keep WiFi connected that's why you are experiencing this behavior. 
You have two options:
1. Comment out the task here:
[](https://github.com/aws/amazon-freertos/blob/feature/ble-beta/demos/common/network_manager/aws_iot_network_manager.c#L402)
2. You can programmatically : disable WIFI off by using network manager API AwsIotNetworkManager_DisableNetwork() and providing AWSIOT_NETWORK_TYPE_WIFI as the parameter.  You can turn the WIFI back on by using AwsIotNetworkManager_EnableNetwork( AWSIOT_NETWORK_TYPE_WIFI ). 
Let us know if you have any questions.
Thanks.

chegewara, 2019-04-08:
Thanks for help and support, i found solution for my requirements. I am using 2 variables (1 for auto-connect and 1 for auto-reconnect):
```
static void prvWiFiNetworkStateChangeCallback( uint32_t ulNetworkType, AwsIotNetworkState_t xNetworkState, void* pvContext )
{
    if( xNetworkState == eNetworkStateDisabled && autoReconnect )
    {
        xSemaphoreGive( xWiFiConnectLock );
    }
}

void prvWiFiConnectTask( void * pvParams )
{
    uint16_t ulNumNetworks, ulNetworkIndex;
    TickType_t xWifiConnectionDelay = pdMS_TO_TICKS( wifiConnectDELAY_SECONDS * 1000 );
    BaseType_t xWiFiConnected;

    for(;;)
    {
        if( xSemaphoreTake( xWiFiConnectLock, portMAX_DELAY ) == pdTRUE )
        {
            xWiFiConnected = pdFALSE;
            while( xWiFiConnected == pdFALSE )
            {
                ulNumNetworks = IotBleWifiProv_GetNumNetworks();
                if( ulNumNetworks > 0 && autoConnect)
```

closed 2019-04-08 by chegewara