aws/amazon-freertos #497: make flash fails on OSX due to hwcrypto/aes.h: No such file or directory

mluis, 2019-04-06:
**Describe the bug**
While running `make flash monitor` from the tutorial the procedure fails.

**System information**
- Which hardware board or part numbers? 
ESP32 WROOM-32
- Operating System
OSX
- Version of Amazon FreeRTOS 
v1.4.7-39-ge1c0f7f7
- Project 
MQTT

**Expected behavior**
Procedure of `make flash monitor` show proceed.

**Additional context**
```
amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code/mbedtls/port/include/aes_alt.h:31:26: fatal error: hwcrypto/aes.h: No such file or directory
compilation terminated.
make[1]: *** [lib/tls/aws_tls.o] Error 1
make: *** [component-amazon-freertos-common-build] Error 2
```

Thank you!

abhidixi11, 2019-04-06:
Hello @mluis ,

The file should be part of your repo. Can you please verify that you have aes.h your repository here  :
lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/include/hwcrypto/aes.h ?
Thanks!

mluis, 2019-04-06:
Hello @abhidixi11 ,

The file is part of the repo.
```
$ file lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/include/hwcrypto/aes.h
lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/include/hwcrypto/aes.h: c program text, ASCII text
```

abhidixi11, 2019-04-06:
Hello @mluis,
Have you tried building with the latest master branch ?
Also can you please try 'make all' and see if the build goes through.
Thanks!

mluis, 2019-04-06:
Hi @abhidixi11 ,

Yes, just tried it.

```
[mluis@macb0x:~/research/aws/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make]$ git pull
Already up to date.
[mluis@macb0x:~/research/aws/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make]$ make all
Toolchain path: /Users/miguelluis/research/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/miguelluis/research/esp/esp-idf/requirements.txt are satisfied.

CC build/amazon-freertos-common/lib/tls/aws_tls.o
In file included from /Users/miguelluis/research/aws/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/common/config_files/FreeRTOSConfig.h:40:0,
                 from /Users/miguelluis/research/aws/amazon-freertos/lib/include/FreeRTOS.h:56,
                 from /Users/miguelluis/research/aws/amazon-freertos/lib/tls/aws_tls.c:27:
/Users/miguelluis/research/esp/esp-idf/components/esp32/include/rom/ets_sys.h:1:2: warning: #warning rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead [-Wcpp]
 #warning rom/ets_sys.h is deprecated, please use esp32/rom/ets_sys.h instead
  ^
In file included from /Users/miguelluis/research/aws/amazon-freertos/lib/third_party/mbedtls/include/mbedtls/aes.h:111:0,
                 from /Users/miguelluis/research/aws/amazon-freertos/lib/third_party/mbedtls/include/mbedtls/ctr_drbg.h:36,
                 from /Users/miguelluis/research/aws/amazon-freertos/lib/tls/aws_tls.c:40:
/Users/miguelluis/research/aws/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code/mbedtls/port/include/aes_alt.h:31:26: fatal error: hwcrypto/aes.h: No such file or directory
compilation terminated.
make[1]: *** [lib/tls/aws_tls.o] Error 1
make: *** [component-amazon-freertos-common-build] Error 2
[mluis@macb0x:~/research/aws/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make]$
```

mluis, 2019-04-07:
Just tried on Linux and the problem persists...

mluis, 2019-04-07:
Under linux this is my tree:
```
mluis@vub0x:~/esp$ tree -L 2
.
├── amazon-freertos
│   ├── CHANGELOG.md
│   ├── LICENSE
│   ├── README.md
│   ├── checksums.json
│   ├── demos
│   ├── directories.txt
│   ├── lib
│   ├── tests
│   └── tools
├── esp-idf
│   ├── CMakeLists.txt
│   ├── CONTRIBUTING.rst
│   ├── Kconfig
│   ├── LICENSE
│   ├── README.md
│   ├── add_path.sh
│   ├── components
│   ├── docs
│   ├── examples
│   ├── make
│   ├── requirements.txt
│   └── tools
└── xtensa-esp32-elf
    ├── bin
    ├── include
    ├── lib
    ├── libexec
    ├── share
    └── xtensa-esp32-elf
```

```
mluis@vub0x:~/esp$ printenv IDF_PATH
/home/mluis/esp/esp-idf
```
And I'm also exporting correctly, I think..
```
export PATH="$HOME/esp/xtensa-esp32-elf/bin:$PATH"
```

abhidixi11, 2019-04-07:
Hello @mluis,

OK. Can please try following commands and let me know if it builds ?
make -C /<your_root>/demos/espressif/esp32_devkitc_esp_wrover_kit/make clean
make -j4 -C /<your_root>/demos/espressif/esp32_devkitc_esp_wrover_kit/make V=1
Thanks.

mluis, 2019-04-07:
Hi @abhidixi11 ,

Yes, it builds without a problem.

abhidixi11, 2019-04-07:
Cool ! 
Now can you please try flash ?

mluis, 2019-04-07:
I'm sorry, something went wrong on the -j4... I couldn't see it at first.

```
$ make -j4 -C demos/espressif/esp32_devkitc_esp_wrover_kit/make V=1 
......
/Users/miguelluis/research/aws/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code/mbedtls/port/include/aes_alt.h:31:26: fatal error: hwcrypto/aes.h: No such file or directory
compilation terminated.
make[1]: *** [lib/tls/aws_tls.o] Error 1
make: *** [component-amazon-freertos-common-build] Error 2
make: *** Waiting for unfinished jobs....
```

abhidixi11, 2019-04-07:
OK then, it looks like the same problem:
aes_alt.h:31:26: fatal error: hwcrypto/aes.h: No such file or directory
Have you followed the the Getting Started Guide?
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html
I tried it here on our build machine, and the build goes through.
If the problem persists for you,  we will investigate it  next week.
Thanks.

mluis, 2019-04-07:
Yes, that's the guide I'm following.

I'm on the `make flash monitor` part.

abhidixi11, 2019-04-07:
OK.  We'll try to see what's going wrong on Monday.  
Thanks for your patience!

mluis, 2019-04-08:
Wich esp-idf git repo hash are you using? Currently I'm on latest master.

abhidixi11, 2019-04-08:
Hello @mluis ,
Can you please check if IDF_PATH is set somewhere in your source tree?  If it is, it will not point to lib/third_party/mcu_vendor/espressif/esp-idf
In makefile for demos: (demos/espressif/esp32_devkitc_esp_wrover_kit/make/Makefile) we define IDF_PATH as follows:

> ifndef IDF_PATH
export IDF_PATH := $(AMAZON_FREERTOS_PATH)/lib/third_party/mcu_vendor/espressif/esp-idf
endif

So if you have IDF_PATH defined somewhere else, you will start seeing compilation problems.
We are using latest master to make sure build works.
Thanks.

mluis, 2019-04-08:
It seems that I had previously set the IDF_PATH so it was making the error happening.

Thank you for your support.
Best Regards

closed 2019-04-08 by mluis