aws/amazon-freertos #505: Microchip PIC32MZEF demos will not compile on latest MPLAB X

DexterHaslem, 2019-04-10:
**Describe the bug**
After a fresh clone of master, opening `demos\microchip\curiosity_pic32mzef\mplab` project and trying to compile fails. I've tried xc32 2.10 and 2.15, error is the same. It fails pretty immediately on bootloader

```
make  -f nbproject/Makefile-pic32mz_ef_curiosity.mk dist/pic32mz_ef_curiosity/production/aws_bootloader.X.production.hex
make[4]: Entering directory 'C:/Users/Dexter/Documents/GitHub/amazon-freertos/demos/microchip/curiosity_pic32_bl/aws_bootloader.X'
"C:\Program Files (x86)\Microchip\xc32\v2.10\bin\xc32-gcc.exe" -g -x c -c -mprocessor=32MZ2048EFM100 -ffunction-sections -O1 -I../common/microchip_code -I../common/microchip_code/bsp -I../common/microchip_code/framework -I../../../../lib/third_party/mcu_vendor/microchip/harmony/v2.05/framework -I../../../../lib/third_party/mcu_vendor/microchip/harmony/v2.05/framework/driver/usart -I../common/include -I../../../common/ota/bootloader/include -I../../../../lib/third_party/tinycrypt/lib/include -I../../../../lib/third_party/tinycrypt/asn1 -I../../../common/ota/bootloader/flash/portable/microchip/curiosity_pic32mzef -I../config_files -MMD -MF build/pic32mz_ef_curiosity/production/_ext/17033345/bsp.o.d -o build/pic32mz_ef_curiosity/production/_ext/17033345/bsp.o ../common/microchip_code/bsp/bsp.c -DXPRJ_pic32mz_ef_curiosity=pic32mz_ef_curiosity -legacy-libc 
In file included from ../../../../lib/third_party/mcu_vendor/microchip/harmony/v2.05/framework/peripheral/pcache/processor/pcache_p32xxxx.h:625:0,
                 from ../../../../lib/third_party/mcu_vendor/microchip/harmony/v2.05/framework/peripheral/pcache/processor/pcache_processor.h:33,
                 from ../../../../lib/third_party/mcu_vendor/microchip/harmony/v2.05/framework/peripheral/pcache/plib_pcache.h:123,
                 from ../common/microchip_code/framework/system/devcon/src/sys_devcon_pic32mz.c:56:
../../../../lib/third_party/mcu_vendor/microchip/harmony/v2.05/framework/peripheral/pcache/processor/pcache_p32mz2048efm100.h:55:71: fatal error: ../templates/pcache_PFMAddressWaitStateEnable_Unsupported.h: No such file or directory
 #include "../templates/pcache_PFMAddressWaitStateEnable_Unsupported.h"
                                                                       ^
compilation terminated.
```

This is on the latest version of MPLAB X v5.15, It keeps modifying project files with no UI interaction every few minutes with the the following changes (patch below) which is likely the culprit? Any idea which MPLAB version to use or how to fix would be appreciated. 

```
diff --git a/demos/microchip/curiosity_pic32_bl/aws_bootloader.X/nbproject/project.xml b/demos/microchip/curiosity_pic32_bl/aws_bootloader.X/nbproject/project.xml
index 0c814c71..aa8117ee 100644
--- a/demos/microchip/curiosity_pic32_bl/aws_bootloader.X/nbproject/project.xml
+++ b/demos/microchip/curiosity_pic32_bl/aws_bootloader.X/nbproject/project.xml
@@ -13,6 +13,35 @@
             <asminc-extensions/>
             <sourceEncoding>ISO-8859-1</sourceEncoding>
             <make-dep-projects/>
+            <sourceRootList>
+                <sourceRootElem>../common/microchip_code</sourceRootElem>
+                <sourceRootElem>../../../../lib/third_party/mcu_vendor/microchip/harmony/v2.05/framework/driver/tmr/src/dynamic</sourceRootElem>
+                <sourceRootElem>../../../../lib/third_party/mcu_vendor/microchip/harmony/v2.05/framework/driver/usart/src/dynamic</sourceRootElem>
+                <sourceRootElem>../../../../lib/third_party/mcu_vendor/microchip/harmony/v2.05/framework/system/tmr/src</sourceRootElem>
+                <sourceRootElem>../../../../lib/third_party/mcu_vendor/microchip/harmony/v2.05/framework/system/int/src</sourceRootElem>
+                <sourceRootElem>../config_files</sourceRootElem>
+                <sourceRootElem>../common</sourceRootElem>
+                <sourceRootElem>../../../common/ota/bootloader/crypto/tinycrypt</sourceRootElem>
+                <sourceRootElem>../../../common/ota/bootloader/include</sourceRootElem>
+                <sourceRootElem>../../../common/ota/bootloader/flash/portable/microchip/curiosity_pic32mzef</sourceRootElem>
+                <sourceRootElem>../../../common/ota/bootloader/loader</sourceRootElem>
+                <sourceRootElem>../../../common/ota/bootloader/logging</sourceRootElem>
+                <sourceRootElem>../../../../lib/third_party/tinycrypt/lib</sourceRootElem>
+                <sourceRootElem>../../../../lib/third_party/tinycrypt/asn1</sourceRootElem>
+                <sourceRootElem>../../../common/ota/bootloader/crypto/asn1utility</sourceRootElem>
+                <sourceRootElem>../../../../lib/third_party/mbedtls/library</sourceRootElem>
+                <sourceRootElem>../../../../lib/third_party/mbedtls/include/mbedtls</sourceRootElem>
+                <sourceRootElem>../../../common/ota/bootloader/utility/user-config</sourceRootElem>
+            </sourceRootList>
+            <confList>
+                <confElem>
+                    <name>pic32mz_ef_curiosity</name>
+                    <type>2</type>
+                </confElem>
+            </confList>
+            <formatting>
+                <project-formatting-style>false</project-formatting-style>
+            </formatting>
         </data>
     </configuration>
 </project>
diff --git a/demos/microchip/curiosity_pic32mzef/mplab/nbproject/project.xml b/demos/microchip/curiosity_pic32mzef/mplab/nbproject/project.xml
index ce07d70d..b33cefec 100644
--- a/demos/microchip/curiosity_pic32mzef/mplab/nbproject/project.xml
+++ b/demos/microchip/curiosity_pic32mzef/mplab/nbproject/project.xml
@@ -15,6 +15,43 @@
             <make-dep-projects>
                 <make-dep-project>../../curiosity_pic32_bl/aws_bootloader.X</make-dep-project>
             </make-dep-projects>
+            <sourceRootList>
+                <sourceRootElem>../../../../lib/third_party/mcu_vendor/microchip</sourceRootElem>
+                <sourceRootElem>../../../../lib/FreeRTOS/portable/MPLAB</sourceRootElem>
+                <sourceRootElem>../../../../lib/secure_sockets</sourceRootElem>
+                <sourceRootElem>../../../common/include</sourceRootElem>
+                <sourceRootElem>../../../../lib/pkcs11/portable/microchip/curiosity_pic32mzef</sourceRootElem>
+                <sourceRootElem>../../../../lib/wifi/portable/microchip/curiosity_pic32mzef</sourceRootElem>
+                <sourceRootElem>../../tmp</sourceRootElem>
+                <sourceRootElem>../common/application_code</sourceRootElem>
+                <sourceRootElem>../common/config_files</sourceRootElem>
+                <sourceRootElem>../../../../lib/third_party/tracealyzer_recorder</sourceRootElem>
+                <sourceRootElem>../../../../lib/FreeRTOS-Plus-TCP</sourceRootElem>
+                <sourceRootElem>../../../common/demo_runner</sourceRootElem>
+                <sourceRootElem>../../../common/mqtt</sourceRootElem>
+                <sourceRootElem>../../../common/greengrass_connectivity</sourceRootElem>
+                <sourceRootElem>../../../common/shadow</sourceRootElem>
+                <sourceRootElem>../../../common/tcp</sourceRootElem>
+                <sourceRootElem>../../../../lib/pkcs11/portable/microchip/pic32mzef</sourceRootElem>
+                <sourceRootElem>../../../common/logging</sourceRootElem>
+                <sourceRootElem>../../../common/ota</sourceRootElem>
+                <sourceRootElem>../../../../lib/ota</sourceRootElem>
+                <sourceRootElem>../../../../lib/third_party/tinycbor</sourceRootElem>
+                <sourceRootElem>../../../../lib/pkcs11/mbedtls</sourceRootElem>
+                <sourceRootElem>../../../../lib/cbor</sourceRootElem>
+                <sourceRootElem>../../../../lib/defender</sourceRootElem>
+                <sourceRootElem>../../../common/defender</sourceRootElem>
+                <sourceRootElem>../../../common/devmode_key_provisioning</sourceRootElem>
+            </sourceRootList>
+            <confList>
+                <confElem>
+                    <name>pic32mz_ef_curiosity</name>
+                    <type>2</type>
+                </confElem>
+            </confList>
+            <formatting>
+                <project-formatting-style>false</project-formatting-style>
+            </formatting>
         </data>
     </configuration>
 </project>
```

DexterHaslem, 2019-04-10:
After some more investigation with same setup working on Linux and OSX, I found root cause.
To fix this error on windows, move the project closer to root of drive, eg moving to `C:\amazon-freertos` and opening project from there worked on windows!

closed 2019-04-10 by DexterHaslem

GnUfTw, 2020-04-29:
I ran into this issue (../templates/* source files not being found) when creating a bootloader project for an existing application. I had the source in my user directory (C:\Users\user\work\company\project\repo) and couldn't get the bare, new bootloader project to compile. Indeed, moving the source tree closer to the root of my drive (C:\dev\repo) allows the bootloader project to compile. This issue was experienced w/ xc32 v2.40, harmony v2.06 and mplabx v5.35. 

Thanks for your time investigating, it has saved me a ton of time today.