aws/amazon-freertos #510: ble with shadow

eldo-frederick, 2019-04-11:
7 1302 [MainDemoTask] [ERROR][NET][13020] Failed to establish new connection.
8 1303 [MainDemoTask] [ERROR][MQTT][13030] Failed to establish new MQTT connecti
on, error NETWORK ERROR.
9 1303 [MainDemoTask] Shadow_ClientConnect unsuccessful, returned 2.
10 1304 [MainDemoTask] Failed to initialize Shadow client.

ravibhagavandas, 2019-04-12:
@eldo-frederick ,
 Thanks for raising the issue. Could you provide more details on which board you are running the demo and which network type you are using ? Are you running the demo from feature/ble-beta branch ? If so which commit ID are you using?

These will help us troubleshoot the issue.

danhorner, 2019-04-22:
@eldo-frederick,

Are you still experiencing trouble with demos in the ble-beta branch? Please feel free to reopen this issue and let us know which port you are working with. We'd love to hear more about your experience.

closed 2019-04-22 by danhorner