aws/amazon-freertos #531: ST ble feature does not compile.

personlast, 2019-04-16:
I am currently looking to setup the ble demo with the STM32L475 Discovery Kit.

I have followed the Aws guide here regarding setting up ble demo: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_st.html

Steps I took to set up the project:

1. Downloaded the source code zip
2. Created eclipse workspace for use with stm32 workbench
3. Imported the existing project under 
{project-root}/demos/st/B-L475E-IOT01A/SW4STM32/
4. Selected the Project Tab -> Build All
5. IDE console output is below

[image]

**System information**
I'm currently on MacOS 
using the stm32 workbench IDE
targeting the STM32L475 Discovery Kit board
My version of Amazon FreeRTOS is amazon-freertos VERSION 1.4.7
I'm targeting the MQTT over ble demo example

Any info or help on this will be appreciated, thanks.

danhorner, 2019-04-19:
Hi @personlast,
 
Thanks for your interest in Amazon FreeRTOS. We are aware of this issue and
are working on improving support for BLE on the STM32L475 boards.
 
 It's also important to note that BLE support is only functional on STM32L4
Discovery Kit IoT Nodes with the STBTLE-1S module. You might wish to verify
that this is the specific module on your board.

gordonwang0, 2019-05-08:
Hi @personlast,

The ble-beta branch has been deprecated in favor of release-1.5.  Builds on that branch are handled through CMake.

I am closing this issue; please re-open if you have any more questions.

closed 2019-05-08 by gordonwang0