aws/amazon-freertos #533: custom mqtt topic (non default) is failing

varshakiragi, 2019-04-17:
In the demo application, if the MQTT topic is freertos/demos/echo, it works fine. But if I change the topic name(echoTOPIC_NAME) : freertos/demos/eche or freertos/demos/echod or "test".. all are failing.

Is there anything to be modified to change the topic name?
Below is the policy details:
{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": "iot:Connect",
      "Resource": "arn:aws:iot:us-west-2:587219694816:*"
    },
    {
      "Effect": "Allow",
      "Action": "iot:Publish",
      "Resource": "arn:aws:iot:us-west-2:587219694816:*"
    },
    {
      "Effect": "Allow",
      "Action": "iot:Subscribe",
      "Resource": "arn:aws:iot:us-west-2:587219694816:*"
    },
    {
      "Effect": "Allow",
      "Action": "iot:Receive",
      "Resource": "arn:aws:iot:us-west-2:587219694816:*"
    }
  ]
}

danhorner, 2019-04-17:
Hello, 
Thanks for opening this issue.

Your policy document would likely be rejected by the create-policy API. There should be an asterisk or a topic at the end of the resource name:  "arn:aws:iot:us-west-2:587219694816:*"

If that doesn't fix things, can you provide some details about the specific target board and branch you are testing so that we can investigate further?

danhorner, 2019-04-17:
For more information about resource naming, check out this section of the  IOT Developer guide: https://docs.aws.amazon.com/iot/latest/developerguide/action-resources.html

danhorner, 2019-04-22:
@varshakiragi 

My aplogies: the github markdown parser dropped the asterisks in your policy. Your policy is correct and will allow access to all client IDs and topics. 

Were you able to change the default topic? If not, what was the symptom of failure?

varshakiragi, 2019-04-25:
I am able to change the default topic.
Anyway Thanks.

closed 2019-04-25 by varshakiragi