aws/amazon-freertos #545: Amazon-FreeRTOS porting

meixuxunxue, 2019-04-22:
I want to ask that all the tests must be passed in “Amazon FreeRTOS Qulification Developer Guide”  or is only for reference  when porting a new mcu-wifi chip on Amazon-FreeRTOS, thanks !

danhorner, 2019-04-22:
Hi @meixuxunxue, 

When qualifying a port of Amazon FreeRTOS to a new MCU wifi chip, all of the relevant tests in the  [Amazon FreeRTOS Qualification Developer Guide](https://github.com/aws/amazon-freertos/blob/master/tests/Amazon%20FreeRTOS%20Qualification%20Developer%20Guide.pdf) should pass.  Tests for features that are not supported by the device (ie. BLE) may be skipped. 

Please let us know if we've answered your question or can provide more information.

meixuxunxue, 2019-04-23:
@danhorner ,

OK.
When I ran vStartOTAUpdateDemoTask( ) preparing for " End-to-end OTA Tests ", I could download image from remote but it always failed when prvUpdateJobStatus. What are the possible causes of this problem? 

In my case, the return of prvPAL_GetPlatformImageState() is not eOTA_PAL_Image_PendingCommit, I also want to know that what usage of eOTA_PAL_Image_PendingCommit image state.

some log is following：

[prvPAL_CloseFile] signature verification passed.
[prvIngestDataBlock] File receive complete and signature is valid.
[prvUpdateJobStatus] Msg: {"status":"IN_PROGRESS","statusDetails":{"self_test":"ready","updatedBy":State: Active  Received: 506   Queued: 506   Processed: 499   Dropped: 0
State: Active  Received: 506   Queued: 506   Processed: 499   Dropped: 0
[prvUpdateJobStatus] Failed: $aws/things/pc_test/jobs/AFR_OTA-ivy5661_OTA_TESTS_42/update

danhorner, 2019-04-24:
@meixuxunxue,

The PAL should return status `eOTA_PAL_ImageState_PendingCommit` *after reboot* to indicate that firmware image download is complete and the firmware is ready for self-test. 

Can you share some more information about your platform and use case? If you are working on enabling OTA on a new platform, have you already implemented your portable layer and passed the PAL tests as outlined in [lib/ota/portable/README.md](https://github.com/aws/amazon-freertos/blob/master/lib/ota/portable/README.md)?  

Feel free to reach out on the AWS forums if you need us to contact you privately.

meixuxunxue, 2019-04-25:
@danhorner ,
OK, I understand, thank you!

meixuxunxue, 2019-04-29:
@danhorner ,

There is an error when I ran ota_e2e.py. Setting the parameter `aws_signer_platform`  in board.json both `Windows Simulator` and `AWSIoTGeneric(default)` would cause this error. Is it related to the parameter?I am working on porting a new mcu-wifi platform and I can acheive ota update through aws console using Windows Simulator platform signing for my image.  thank you!

Error: OTA update creation failed for OTA update ID b8060cb1-47bd-4b87-bef8-ea6e578146b6
Error Info: ResourceNotFoundException:Platform with id Windows Simulator does not exist (Service: AWSSigner; Status Code: 404; Error Code: ResourceNotFoundException; Request ID: efc10e4b-6a52-11e9-a2be-b1dd054eb2ca)...

dcgaws, 2019-04-29:
@meixuxunxue, hello, our recommended signature cipher spec is ECDSA with curve P256 and SHA256. To produce a signature of that spec, the AWS signer platform name you want is AmazonFreeRTOS-Default.

FYI, you can use the following AWS CLI command to see that:
C:\>aws signer list-signing-platforms
...
            "displayName": "Amazon FreeRTOS SHA256-ECDSA",
            "platformId": "AmazonFreeRTOS-Default",
            "signingConfiguration": {
 ...

closed 2019-04-29 by dcgaws