aws/amazon-freertos #549: JITP

chegewara, 2019-04-23:
Hi,
i am working on ble-beta branch on esp32 and i am having problems with using JITP procedure.
For creating all certificates i am using this tutorial and from cli all is working fine:
https://aws.amazon.com/blogs/iot/setting-up-just-in-time-provisioning-with-aws-iot-core
but, when i use device certificates in my code then connection is failing with `TIMEOUT`:
```
11 788 [MQTTEcho] [INFO ][NET][788] New network connection established.
13 788 [MQTTEcho] [INFO ][MQTT][788] Establishing new MQTT connection.
14 788 [MQTTEcho] [INFO ][MQTT][788] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
15 788 [MQTTEcho] [INFO ][MQTT][788] Waiting for operation CONNECT to complete.
16 808 [MqttRecv] [ERROR][NET][808] No data could be read from the socket. Returned -30848.
...
17 50791 [MQTTEcho] [INFO ][MQTT][50791] MQTT operation CONNECT complete with result TIMEOUT.
18 50791 [MQTTEcho] [ERROR][MQTT][50791] Failed to establish new MQTT connection, error TIMEOUT.
19 50791 [MQTTEcho] [ERROR][prvMqttPublishTask][50791] Failed to create an MQTT connection.
```

Now, when i register device certificate from mosquitto with CLI then esp32 can connect with that certificate.

dcgaws, 2019-04-25:
@chegewara since the BLE branch is still in Beta, and in the interest of eliminating experimental variables, would you please consider getting your JITP setup working first in our master branch (i.e. using ESP32 over Wi-Fi)? 

Regarding keyJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM, it is needed in both cases. With it, the AWS IoT MQTT gateway doesn't know to which account to route the inbound request from the device.

chegewara, 2019-04-25:
Hi @dcgaws 
i thought mqtt gateway is using `xxxxxxxxxxxx-ats.iot.eu-central-1.amazonaws.com` to know which account to route.

Ok, when i provide `keyJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM` then device certificate and device itself are registered with JITP, i should have test it before i started this issue.

One more question, not really related to aws-freertos. How can i setup IoT thing name generating certificate to provide my own name. I know it is parameter in provisioning-template.json:
```
{
 "templateBody":"{ \"Parameters\" : { \"AWS::IoT::Certificate::Country\" : { \"Type\" : \"String\" }, \"AWS::IoT::Certificate::Id\" : { \"Type\" : \"String\" } }, \"Resources\" : { \"thing\" : { \"Type\" : \"AWS::IoT::Thing\", \"Properties\" : { \"ThingName\" : {\"Ref\" : \"AWS::IoT::Certificate::Id\"}, \"AttributePayload\" : { \"version\" : \"v1\", \"country\" : {\"Ref\" : \"AWS::IoT::Certificate::Country\"}} } }, \"certificate\" : { \"Type\" : \"AWS::IoT::Certificate\", \"Properties\" : { \"CertificateId\": {\"Ref\" : \"AWS::IoT::Certificate::Id\"}, \"Status\" : \"ACTIVE\" } }, \"policy\" : {\"Type\" : \"AWS::IoT::Policy\", \"Properties\" : { \"PolicyDocument\" : \"{\\\"Version\\\": \\\"2012-10-17\\\",\\\"Statement\\\": [{\\\"Effect\\\":\\\"Allow\\\",\\\"Action\\\": [\\\"iot:*\\\"],\\\"Resource\\\" : [\\\"*\\\"]}]}\" } } } }",
 "roleArn":"arn:aws:iam::xxxxxxxx:role/aws-iot-jitp"
}
```

Thanks for help.

dcgaws, 2019-04-25:
@chegewara to confirm, what you're observing when using the above provisioning-template.json is that the Thing name ends up being the SHA256 hash of the client certificate? (I'm referencing https://docs.aws.amazon.com/iot/latest/developerguide/jit-provisioning.html, which doesn't exactly say that, but I believe that's what I've observed.)

Note that the above documentation page also gives you the option of using AWS::IoT::Certificate::CommonName in your provisioning template. You control the CN (aka the Subject Name) of your client certificates when you create them in the first place. Does that capability address your request? If not, can you please give me a more detailed example of what you'd like to see happen?

chegewara, 2019-04-25:
> @chegewara to confirm, what you're observing when using the above provisioning-template.json is that the Thing name ends up being the SHA256 hash of the client certificate? (I'm referencing [...]

Yes, thats exactly what i am getting. ThingName is the same as device certificate name and is SHA256.

> Note that the above documentation page also gives you the option of using AWS::IoT::Certificate::CommonName in your provisioning template. You control the CN (aka the Subject Name) of your client [...]

I think this can be what i am looking for, thanks for help, appreciate it.

closed 2019-04-25 by chegewara

monkeytronics, 2019-11-06:
@chegewara I just posted a question in issue [https://github.com/aws/amazon-freertos/issues/1119] that you will probably know the answer to right away! I'm still feeling my way around freeRTOS.