aws/amazon-freertos #559: DHCP client ignores last DHCP option

unixdj, 2019-04-25:
**Describe the bug**
DHCP client ignores the last DHCP option in the reply packet

**System information**
- Which hardware board or part numbers?
ESP32
- IDE used
none
- Operating System [Windows|Linux|MacOS]
Linux
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)
v1.4.7-59-ga2b064f6
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
Custom Application, but can be tested with MQTT
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
When the board connects to WiFi and obtains an IP address over DHCP, all DHCP options must be parsed.  E.g., if the DHCP server puts DNS configuration as the last option, the DNS client should use the configured resolver and not the fallback resolver.

**Screenshots or console output**

**To reproduce**
Steps to reproduce the behavior:

1. Put debug prints in `lib/FreeRTOS-Plus-TCP/source/FreeRTOS_DHCP.c`, function `prvProcessDHCPReplies()`
2. Observe which DHCP options are in the packet but aren't parsed

**Code to reproduce the bug**

**Additional context**

Fix:

```
diff --git a/lib/FreeRTOS-Plus-TCP/source/FreeRTOS_DHCP.c b/lib/FreeRTOS-Plus-TCP/source/FreeRTOS_DHCP.c
index 4ec3e5dc..db34f3eb 100644
--- a/lib/FreeRTOS-Plus-TCP/source/FreeRTOS_DHCP.c
+++ b/lib/FreeRTOS-Plus-TCP/source/FreeRTOS_DHCP.c
@@ -647,9 +647,9 @@ const uint32_t ulMandatoryOptions = 2ul; /* DHCP server address, and the correct
 				/* Walk through the options until the dhcpOPTION_END_BYTE byte
 				is found, taking care not to walk off the end of the options. */
 				pucByte = &( pxDHCPMessage->ucFirstOptionByte );
-				pucLastByte = &( pucUDPPayload[ lBytes - dhcpMAX_OPTION_LENGTH_OF_INTEREST ] );
+				pucLastByte = &( pucUDPPayload[ lBytes ] );
 
-				while( pucByte < pucLastByte )
+				while( pucByte <= pucLastByte - dhcpMAX_OPTION_LENGTH_OF_INTEREST )
 				{
 					ucOptionCode = pucByte[ 0 ];
 					if( ucOptionCode == dhcpOPTION_END_BYTE )
@@ -671,7 +671,7 @@ const uint32_t ulMandatoryOptions = 2ul; /* DHCP server address, and the correct
 						ucLength = pucByte[ 1 ];
 						pucByte += 2;
 
-						if( pucByte >= pucLastByte - ucLength )
+						if( pucByte > pucLastByte - ucLength )
 						{
 							break;
 						}
```

dcgaws, 2019-04-25:
@unixdj nice bug. I'm taking a look.

closed 2019-05-01 by dcgaws

unixdj, 2019-05-02:
Thanks for fixing this!

unixdj, 2019-05-02:
@dcgaws Thanks for fixing this!