aws/amazon-freertos #569: MT7697Hx aws iot demo issue

cs3sw, 2019-04-29:
Hi,

We use MT7697Hx to run mqtt demo(aws_hello_world), compiler tool by using gnu gcc
When demo start running, it will show error as below:
 [wrap] ERROR: Failed to provision certificate 48
 [wrap] ERROR: Failed to provision private key 48
Do you know what problem will cause this issue?

Detail log as below:
loader enable 30s timeout
loader init
fota: TMP is empty, skip upgrade
jump to (0x10079000)
0 10 [wrap] Write certificate...
1 10 [wrap] ERROR: Failed to provision certificate 48
2 11 [wrap] ERROR: Failed to provision private key 48
3 217 [wrap] WiFi module initialized. Connecting to AP...
4 218 [wpa_supplicant] ========= Supplicant Ready =======5 4697 [wrap] WiFi Connected to AP. Creating tasks which use network...
6 4697 [wrap] Creating MQTT Echo Task...
7 4697 [MQTTEcho] MQTT echo attempting to connect to aiab19o4pmx7b-ats.iot.us-east-1.amazonaws.com.
8 4941 [MQTT] ERROR: Object with label Device Priv TLS Key not found.
9 4941 [MQTT] ERROR: Loading credentials from flash into TLS context failed with error 6.
10 4941 [MQTT] ERROR: TLS_Connect failed with error code 6
11 4941 [MQTT] TLS_Connect fail (0x7d1, aiab19o4pmx7b-ats.iot.us-east-1.amazonaws.com)
12 4952 [MQTTEcho] ERROR:  MQTT echo failed to connect with error 1.
13 4952 [MQTTEcho] MQTT echo test could not connect to broker.
14 4952 [MQTTEcho] MQTT echo demo finished.
15 4952 [MQTTEcho] ----Demo finished----

Thank you!
Jack Ho

dcgaws, 2019-04-29:
@cs3sw, yes, please take a look at vDevModeKeyProvisioning in X. That function is called from vApplicationDaemonTaskStartupHook in main.c for most of our projects (including the MediaTek board demos). 

The purpose of vDevModeKeyProvisioning is to do a one-time import of the device client certificate and private key. That's for lab testing purposes only, since we don't recommend having those objects statically compiled in your production firmware. 

In your case, error code 48 corresponds to CKR_DEVICE_ERROR, That is likely being returned by PKCS11_PAL_SaveObject in lib\pkcs11\portable\mediatek\mt7697hx-dev-kit\aws_pkcs11_pal.c. The latter function is used to write the two cryptographic objects I mentioned above to flash storage using the MediaTek driver. 

Have you correctly defined keyCLIENT_CERTIFICATE_PEM and keyCLIENT_PRIVATE_KEY_PEM in demos\common\include\aws_clientcredential_keys.h to valid PEM encoded objects? If so, and you're still seeing the error above, I recommend stepping through PKCS11_PAL_SaveObject in your debugger.

cs3sw, 2019-05-06:
Hi,
I defined keyCLIENT_CERTIFICATE_PEM and keyCLIENT_PRIVATE_KEY_PEM correctly. it still had same problem, then I put debug message in PKCS11_PAL_SaveObject, I discover it will call hal_file_write function, In hal_file_write function, first it will run ret = hal_file_find(pcFileName, &fIdx);
the pcFileName is FreeRTOS_P11_Certificate.dat, it seems hal_file_find function cannot find match one and return FALSE, do you know why cannot find the match file?

Thank you

dcgaws, 2019-05-06:
@cs3sw it looks that hal_file_find is being used to locate either an existing entry by that name in the file table or the first free entry. The first time PKCS11_PAL_SaveObject is called after your board has been re-flashed, I believe you should expect the latter case: hal_file_find will return FALSE and *pfIdx is set to a free file index (i.e. something other than -1). 

After that call, hal_file_write should make two calls to hal_flash_write: one for the file header and one for the file data.

How far in that sequence are you getting?

cs3sw, 2019-05-07:
Hi,
I put debug message, in  ret = hal_file_find(pcFileName, &fIdx);
I got ret =0, fidx = -1
In hal_file_find function,  it never go into if and else if condition as below, I got *pfIdx =-1 
//-------------------------------------------------------
   for ( idx=0; idx<MAX_FILE_NUM; idx++)
    { 
        if ( strcmp(gFiles[idx].fname, pfName) == 0 )
        {
            if ( pfIdx )
                *pfIdx = idx;
            return pdTRUE;
        }    
        else if ( (*pfIdx== -1) && ( (strlen(gFiles[idx].fname)==0) || (gFiles[idx].fname[0]==0xff)) )
        {    
            if ( pfIdx )
                *pfIdx = idx; //Free File Index
        }         
    }
//-------------------------------------------------
So In hal_file_write function as below , when fIdx = -1, it will directly return FALSE and never call hal_flash_write function.
//-------------------------------------------------
if (ret==pdFALSE) 
    {
        if (fIdx==-1)
        {
           //into here
            return pdFALSE;
        }
        strcpy(gFiles[fIdx].fname,  pcFileName);
    }    
//-------------------------------------------------
Do you know what is the problem?
Thank you.

dcgaws, 2019-05-07:
Hello @cs3sw it sounds like your flash layout isn't producing the conditions expected by hal_file_find: basically, that your file table starts out with either 0x00 or 0xFF in the first byte of the fname field of every vacant entry, or, that the table is pre-initialized with the expected filenames for the crypto library. 

It's tough for me to give more specific guidance than that since, while we've extensively tested the Keil project we offer for the MT7697Hx, we don't currently have support for the GNU tool chain for that Kit.

cs3sw, 2019-05-09:
Hi 
So you mean this aws_iot project run on Keil with no problem, right?
In GNU part, it will have this problem. 
So maybe I need to modify flash layout compare with Keil part to fix this issue.
Am I mention correct?

dcgaws, 2019-05-09:
Yes, flash layout in GNU could be the issue. Another possibility is that you're not erasing flash each time you write your image. As per the code above, if that flash location is left uninitialized, the app will fail as you've observed.

cobusve, 2019-05-16:
I am closing this issue as we have not heard anything back and it seems like this is caused by the flash memory not being cleared before trying to store the keys. If you need more help please feel free to comment here.

closed 2019-05-16 by cobusve