aws/amazon-freertos #599: what is use of NVM in PKCS11 module

varshakiragi, 2019-05-07:
I have implemented PKCS11 module without using NVM. I am able to subscribe and publish to AWS cloud.
But if I see other implementations all the APIs in aws_pkcs11_pal.c like PKCS11_PAL_SaveObject, PKCS11_PAL_FindObject, PKCS11_PAL_GetObjectValue uses NVM.

What is the use case in which NVM is required?

dcgaws, 2019-05-07:
@varshakiragi hello, regarding your PKCS #11 module, is your implementation for a hardware based crypto engine? That will help me answer your question more precisely. Thank you.

varshakiragi, 2019-05-08:
Yes. My implementation is for a hardware based crypto engine.

dcgaws, 2019-05-08:
Can you please take a look at https://github.com/aws/amazon-freertos/blob/fix/pkcs11_compliance/lib/pkcs11/ReadMe.md? That document describes the rationale for implementing either a complete "monolithic" PKCS #11 module, or instead implementing the PAL_ routines that you listed above. 

Please let me know any follow-up questions you have. Our documentation in this area can definitely be improved.

varshakiragi, 2019-05-09:
Thanks for the above information.
For me, according to above link **Implement a platform abstraction layer (PAL)** will be applied. 
But my question is : Here what is the need of storing private keys and certificate to perstitant storage?
Every time when the application starts or main() starts executing, private key and certificate will be provided using vDevModeKeyProvisioning().

Then why to store them in NVM?

dcgaws, 2019-05-09:
The vDevModeKeyProvisioning routine should only be used during your initial prototyping phase. 

For production, I recommend using dedicated cryptographic hardware for storing device private keys. I think that approach offers the best defense against device cloning and data theft. Crypto hardware can also simplify provisioning. 

However, crypto hardware can drive up BOM cost past the point that some products can bear. That's why, as an alternative, we provide the PAL interface for flash-based storage of the device private key and certificate. With this approach, the intended model is for the key and certificate to be written once to flash during provisioning. Thereafter, the key and cert should always be read from flash by the app. 

In order to reduce the risk of key theft when using the PAL model, I recommend using MCU onboard flash and disabling JTAG.

varshakiragi, 2019-05-10:
Thank you very much for the detailed explaination. Now I understand that storing the keys and certificate in NVM is good and rsik-free for production phase.

closed 2019-05-10 by varshakiragi