aws/amazon-freertos #614: [OTA] ESP32 notify the OTA job with failure although it accepted the new firmware.

ahmedwahdan, 2019-05-08:
I'm trying to get the OTA functionality to work with ESP32 and I have a strange behavior.
ESP32 notify the OTA job with failure although it accepted the new firmware.

And this is the error msg I get:
`https://forums.aws.amazon.com/ Job in self test but platform state is not!
40 1318 https://forums.aws.amazon.com/ https://forums.aws.amazon.com/ Msg: {"status":"FAILED","statusDetails":{"reason":"rejected: 0x26000000"}}
41 1332 https://forums.aws.amazon.com/ State: Ready Received: 1 Queued: 1 Processed: 0 Dropped: 0 `

gordonwang0, 2019-05-09:
Hi @ahmedwahdan,

Can you post the full logs from your OTA update?

Thanks.

cobusve, 2019-05-17:
Without the logs we were unable to reproduce this issue so I am going to have to close it. 

If you manage to reproduce this and can provide the logs please feel free to post it here and we will re-open it.

closed 2019-05-17 by cobusve