aws/amazon-freertos #615: Scan list on WiFi Provisioning over BLE comes back to app 50% of the time

mhattiga, 2019-05-08:
**Describe the bug**
A clear and concise description of what the bug is.

**System information**
- Which hardware board or part numbers?
- IDE used
- Operating System [Windows|Linux|MacOS]
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 

**To reproduce**
Steps to reproduce the behavior:

1. ...
2. ...

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
Add any other context about the problem here.

Thank you!

mhattiga, 2019-05-08:
Scan list on WiFi Provisioning over BLE comes back to app 50% of the time.

Branch is feature/OTAandHelloWorld.

This bug has been there for sometime.

ravibhagavandas, 2019-05-16:
@mhattiga 
To help reproduce the issue, can you mention with which sdk ( android or IOS ) you saw this issue and the version ( commit ID ) of the SDK you were using.

mhattiga, 2019-05-16:
It was with Android. But I'm not sure what versions at this point. It was the latest on the date that the ticket was submitted. Next time I'll include the versions. Sorry about that.

qiutongs, 2019-07-12:
Sorry for no respond for such a long time. Is this still an issue?

ravibhagavandas, 2019-07-16:
Closing this as the issue was not reproducible in later trials. Please feel free to re-open if you still see the issue.

closed 2019-07-16 by ravibhagavandas