aws/amazon-freertos #620: Misc.py still has old path name that include the "common" directory

mhattiga, 2019-05-09:
**Describe the bug**
In preparing to use this demo:
https://github.com/aws/amazon-freertos/tree/feature/OTAandHelloWorld

I had to modify this file:
~/afr/tools/aws_config_quick_start/misc.py

The modification is to remove "common" from all the path names because what was
~/demos/common/include
is now
~/demos/include

tgsong, 2019-05-16:
Hi, thanks for reporting the issue! We're making folder structure changes for the upcoming release, which caused this incompatibility with existing tools. We're working on fixing all the broken paths now and will address this issue in future release.

mradulan, 2019-06-03:
This has been resolved in the PR #692. Closing this issue. Please reopen if the issue persists.

closed 2019-06-03 by mradulan