aws/amazon-freertos #624: Telnet server support

cs3sw, 2019-05-13:
Hi,

Is it possible to add telnet server service(telnetd) on freertos in the feature?
Thank you!

htibosch, 2019-05-13:
If you like, you can try out this [simple telnet server](https://sourceforge.net/p/freertos/discussion/382005/thread/b3ece8be94/?limit=250#22ca)
I wrote it as an example of how to use semaphores in combination with sockets.
It allows you to send data to a particular client, or to all (as for logging purposes). Hein

cs3sw, 2019-05-15:
Hi 
I will try it, we want run telnet server on freertos and let PC use telnet client to connect to MCU device
to run test command.

Thank you.

htibosch, 2019-05-16:
That should be possible. Please follow the example code in `telnet_main.c`.
Any questions can be asked here ( or on the [FreeRTOS forum](https://sourceforge.net/p/freertos/discussion/382005) ).

cobusve, 2019-05-17:
There are also a number of basic networking examples that come with FreeRTOS+TCP which you can download from the FreeRTOS.org site. Most of those are implemented as a single task which should be easy to add to your application.

Inside of this repo (amazon-freertos) we try to focus our examples on typical IoT applications connecting to AWS, and it is usually frowned upon to accept incoming connections like telnetd to execute code on an IoT device due to the security risks this opens up so we are not planning to add such an example to this repo.

Accordingly I am going to close the issue here. 

If you need more help it would be best to start a thread on the FreeRTOS forum as suggested by htibosh above so that we can answer it there where other people will also be able to benefit from the discussion.

closed 2019-05-17 by cobusve