aws/amazon-freertos #625: extern "C" should be added to port.c

ppejic, 2019-05-13:
**Describe the bug**
I am trying to compile STM32F429ZIT6 project with g++. Although build is passing and everything is linked correctly, after calling "prvPortStartFirstTask" I end up in 
"Default_Handler:
Infinite_Loop:
  b  Infinite_Loop"
due to the g++ function name mangling.

**System information**
- Which hardware board or part numbers?
STM32F429ZI, STM32CubeMX
- IDE used
VSCode
- Operating System [Windows|Linux|MacOS]
Linux
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)
V10.0.1
Compiled with GNU Tools for Arm Embedded Processors 7-2018-q3-update
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 

**To reproduce**
Generate template project for NUCLEO STM32F429ZIT - do not change anything after that! Now change CC and AS variables in Makefile to use g++ instead of gcc. Program is compiled but is not running correctly since it got stuck in Infinite_Loop.

**Additional context**
How to solve it?

Use extern "C" on static function declarations and implementations in port.c

Thank you!

htibosch, 2019-05-14:
It is not clear to me why you get a problem with a static function within a .C file.
The `extern "C"` expression is only recognised by a C++ compiler. Your are not compiling `port.c` as a cpp file, do you?
Already for many years I compiled FreeRTOS in C++ projects, and there should not be uncertainty.
I am using `arm-none-eabi-g++` for c++ files, and `arm-none-eabi-gcc` for C and assembler files.

Default_Handler: isn't that called when you get an interrupt that is not defined?

aggarg, 2019-05-15:
The problem you are facing is not because of ```static``` functions but because of handlers defined in ```port.c``` file: 

```c
/*
 * Exception handlers.
 */
void xPortPendSVHandler( void ) __attribute__ (( naked ));
void xPortSysTickHandler( void );
void vPortSVCHandler( void ) __attribute__ (( naked ));
```

I was able to re-pro it when I compiled all the files (including port.c) using ```arm-none-eabi-g++```. The reason is that the symbols for Handlers in ```port.c``` get mangled and the linker fails to override the default handlers defined in the assembler file. 

As mentioned by @htibosch , you should use ```arm-none-eabi-gcc``` for compiling C files. In case you still want to use ```arm-none-eabi-g++```, you can move the above function declarations from ```port.c``` to ```portmacro.h``` and this should fix your problem (as ```portmacro.h``` already has ```extern "C"```). 

Let us know if you need anything else.

Thanks.

closed 2019-08-16 by dcgaws