aws/amazon-freertos #631: esp32 + SPIRAM issue

chegewara, 2019-05-15:
Hi, on esp32 wrover with SPIRAM turned on in menuconfig NVS storage partition is not initialized to read thing key and cert. Here you can see logs:
```
I (3222) WIFI: SYSTEM_EVENT_STA_GOT_IP
7 3128 [IP-task] vDHCPProcess: offer c0a82ba2ip
8 3134 [iot_thread] [INFO ][DEMO][3134] Successfully initialized the demo. Network type for the demo: 1
9 3134 [iot_thread] [INFO ][MQTT][3134] MQTT library successfully initialized.
10 3136 [iot_thread] [INFO ][Shadow][3136] Shadow library successfully initialized.
11 3136 [iot_thread] [INFO ][DEMO][3136] Shadow Thing Name is test (length 4).
12 3685 [iot_thread] [ERROR][NET][3685] Failed to establish new connection.
13 3686 [iot_thread] [ERROR][MQTT][3686] Failed to establish new MQTT connection, error NETWORK ERROR.
14 3686 [iot_thread] [ERROR][DEMO][3686] MQTT CONNECT returned error NETWORK ERROR.
15 3686 [iot_thread] [INFO ][Shadow][3686] Shadow library cleanup done.
16 3686 [iot_thread] [INFO ][MQTT][3686] MQTT library cleanup done.
17 3686 [iot_thread] [INFO ][DEMO][3686] Demo minimum ever free heap: 4273540 bytes.
18 3693 [iot_thread] [INFO ][COMMON][3693] Common libraries cleanup done.
```
and here are logs with SPIRAM turned off:
```
7 3116 [IP-task] vDHCPProcess: offer c0a82ba2ip
8 3122 [iot_thread] [INFO ][DEMO][3120] Successfully initialized the demo. Network type for the demo: 1
9 3122 [iot_thread] [INFO ][MQTT][3122] MQTT library successfully initialized.
10 3122 [iot_thread] [INFO ][Shadow][3122] Shadow library successfully initialized.
11 3122 [iot_thread] [INFO ][DEMO][3122] Shadow Thing Name is test (length 4).
I (3768) PKCS11: Initializing NVS partition: "storage"
12 8139 [iot_thread] [INFO ][MQTT][8139] Establishing new MQTT connection.
13 8140 [iot_thread] [INFO ][MQTT][8140] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
14 8142 [iot_thread] [INFO ][MQTT][8142] (MQTT connection 0x3ffd292c, CONNECT operation 0x3ffd2a44) Waiting for operation completion.
15 8444 [iot_thread] [INFO ][MQTT][8444] (MQTT connection 0x3ffd292c, CONNECT operation 0x3ffd2a44) Wait complete with result SUCCESS.
```
As you can see in failed case we are missing NVS initialization from PKCS11 `I (3768) PKCS11: Initializing NVS partition: "storage"`

Setup:
- esp32 wrover
- aws-freertos ble-beta branch commit 4fa09aac8140ac05e89d3b7fb88571a6c8318e55

cobusve, 2019-05-15:
Looking at the code I think the only way that it would behave like this would be if the call to mbedtls_ssl_config_defaults from TLS_Connect in aws_tls.c returns an error. Regretably we are not printing an error message in this case so we will go ahead and add some code to print this error when it happens.

Going a bit deeper it looks like this could fail on a memory allocation error inside of mbed_tls, but I cannot confirm this without reproducing the problem.

I am not 100% sure which version of the code you are using. If you can let me know which commit on which branch I can try to reproduce it and also give you a specific line number, but can you please try the following in the meantime?

Add a print message in aws_tls.c around line 653 to print failure of this call and let us know if this is what you are seeing and what the return error code is.
```
 /* Start with protocol defaults. */
    if( 0 == xResult )
    {
        xResult = mbedtls_ssl_config_defaults( &pxCtx->xMbedSslConfig,
                                               MBEDTLS_SSL_IS_CLIENT,
                                               MBEDTLS_SSL_TRANSPORT_STREAM,
                                               MBEDTLS_SSL_PRESET_DEFAULT );

        if( 0 != xResult )
        {
            TLS_PRINT( ( "ERROR: Failed to set ssl config defaults %d \r\n", xResult ) );
        }
    }
```

chegewara, 2019-05-15:
Hi, thanks for your assistance.
You can find commit that i am using in OP: https://github.com/aws/amazon-freertos/commit/4fa09aac8140ac05e89d3b7fb88571a6c8318e55

I will add logs you are suggesting and let you know.

sarenameas, 2019-07-06:
Hello Chegewara, 

Have you been able to get the log as cobusve sugguested?

chegewara, 2019-07-06:
Hi,
currently i switched to master/release-1.5 and i didnt test it yet. I will close this issue and re-open it when i have more info.

closed 2019-07-06 by chegewara