aws/amazon-freertos #637: ESP32 - OTA Agent - Reboot Requirement - Firmware Rollback or Brick Device

sbertrand-witekio, 2019-05-17:
Our device is based on ESP32. It needs to wait before rebooting after an OTA.
The ResetDevice function has been change to send an event to request reboot of the system.
This delay leads to the OTA Agent performing the following queued OTA when a firmware is rejected and reboot is pending.

Is there any documentation about the reboot requirement of the OTA Agent ?
How should the OTA Agent implementation cope with delaying the reboot ?

The Firmware version is the following :
	 AFR : v1.4.7-45-geb0e75b3-dirty
	 IDF : v3.1.3-15-g97deba922

The behavior observed are different depending on the OTA slot where the OTA occurs.
```
V0.14-25-gae996f79-dirty-F		
V0.14-25-gae996f79-dirty-F	0	updating
V0.14-26-gf7822525-dirty-O	16	accepted
V0.14-26-gf7822525-dirty-O		
V0.14-26-gf7822525-dirty-O	16	updating
V0.14-27-g64c33d30-dirty-O	17	accepted
V0.14-27-g64c33d30-dirty-O		
V0.14-27-g64c33d30-dirty-O	17	updating
V0.14-25-gae996f79-dirty-O	16	rejected

The OTA_1 (17) slot get corrupted if the rejected update runs on OTA_0 (16) slot. 
OTA_0 (16) is not valid, but it booted since it has a valid image.
This leads to a Firmware Rollback to V0.14-25. Should not happen since we accepted V0.14-27

V0.14-25-gae996f79-dirty-O		
V0.14-25-gae996f79-dirty-O	16	updating
V0.14-28-g768df95a-dirty-O	17	accepted
V0.14-28-g768df95a-dirty-O		
```

```	
V0.14-25-gae996f79-dirty-F		
V0.14-25-gae996f79-dirty-F	0	updating
V0.14-26-gf7822525-dirty-O	16	accepted
V0.14-26-gf7822525-dirty-O		
V0.14-26-gf7822525-dirty-O	16	updating
V0.14-25-gae996f79-dirty-O	17	rejected

The OTA_0 (16) slot gets corrupted if the rejected update runs on OTA_1 (17) slot.
OTA_1 (17) is not valid. OTA_0 (16) is corrupted. The system has a factory app ( not standard).
The bootloader falls back to it.

V0.14-25-gae996f79-dirty-F		
V0.14-25-gae996f79-dirty-F	0	updating
V0.14-27-g64c33d30-dirty-O	16	accepted
V0.14-27-g64c33d30-dirty-O		
```

pvyawaha, 2019-05-20:
Hi sbertrand-witekio,

Thank you for reporting this, we are looking into the issue and trying to reproduce it.

sbertrand-witekio, 2019-05-22:
While the OTA Agent does not have a reboot pending flag, it has a `xInSelfTest` flags which can be use to avoid processing of the next OTA while it is not cleared.

The following patch is my changes to use that flag.

[0001-OTA-using-self-test-flag-before-processing-new-job.patch.txt](https://github.com/aws/amazon-freertos/files/3209611/0001-OTA-using-self-test-flag-before-processing-new-job.patch.txt)

pvyawaha, 2019-05-23:
Hi sbertrand-witekio,

When the platform is in self-test only operations related to self-test should be performed. 

The `xInSelfTest` flag was introduced to filter messages of type `eOTA_PubMsgType_Stream` when the platform is in self-test. This ensures that we do not process any data blocks received when we are in self-test. The flag is then reset when the image is accepted and its state is changed to `eOTA_ImageState_Accepted`.

In addition to that we should also ignore any new job received when the platform is in self-test and that is filtered in `prvProcessOTAJobMsg` by calling `prvInSelftest` which reads the platform state.

From the patch that you shared it seems you are shutting down and then restarting the OTA agent to delay the reboot. Is that correct?

We have a callback `App_OTACompleteCallback `which gets the event `eOTA_JobEvent_Activate` which then calls `OTA_ActivateNewImage`. This callback can be used to perform some operations (or wait ) before activating the image. Let me know your thoughts.

sbertrand-witekio, 2019-05-23:
> From the patch that you shared it seems you are shutting down and then restarting the OTA agent to delay the reboot. Is that correct?

The patch has  the addition of shutting down the OTA Agent, however this is commented. Shutting down the agent is not entirely valid, it does not guarantee anything. I am not shutting down nor restarting the OTA Agent in this patch.

### `App_OTACompleteCallback` Usage

The implementation could wait proper reboot condition before calling `OTA_ActivateNewImage`. This would be a suitable implementation. However, `OTA_ActivateNewImage` is not the only method and cause for the OTA Agent to request a reboot of the system.

The OTA Agent does provide any notification when the OTA Validation fails. The OTA Agent request a reboot directly.

### `prvProcessOTAJobMsg` & `prvInSelftest`

My patch change the use of `prvInSelftest` for `xInSelfTest` in `prvProcessOTAJobMsg`.

`prvInSelftest` provides the current state of the image. This return **Self Test** until the validation is performed. Once the validation is done, the state return will no longer be **Self Test** allowing for a new job to start transfer. `prvProcessOTAJobMsg` is the first step in starting the transfer by creating a file for reception `prvPAL_CreateFileForRx`.

The overwrite issue faced is due to `prvPAL_CreateFileForRx` being called by a failed-validation firmware. The currently firmware is pending reboot and should not be allows to download new firmware. The call is made since the current image is no longer in self test, it change to rejected. 

By changing  `prvInSelftest` for `xInSelfTest` in `prvProcessOTAJobMsg`, we check the very first initial state of the image ( **Self Test** or not ).  If we booted as Self Test, downloading will only be allow if the image is accepted and `xInSelfTest` is reset by call to `eOTA_ImageState_Accepted`.

The patch changes affectation of `xInSelfTest` during the initialization of the OTA Agent. Restart of the Agent would clear `xInSelfTest`. If the agent is started with `xInSelfTest` set, the flags is not updated.

pvyawaha, 2019-05-23:
Hi sbertrand-witekio,

In the `App_OTACompleteCallback` when we receive the `eOTA_JobEvent_Activate` event we should suspend and wait for the application specific event that you can add as per application logic  before calling `OTA_ActivateNewImage` to delay the reboot.  This is last step before reboot and OTA agent loop should not execute after the `App_OTACompleteCallback` is called with `eOTA_JobEvent_Activate` event.

>The implementation could wait proper reboot condition before calling OTA_ActivateNewImage. This would be a suitable implementation. However, OTA_ActivateNewImage is not the only method and cause for the OTA Agent to request a reboot of the system.

>The OTA Agent does provide any notification when the OTA Validation fails. The OTA Agent request a reboot directly.

When image is not executing in self-test OTA agent will only initiate a reboot after it validates the received image on other bank. We have the above mentioned user callback before this happens. 

When image is executing in self-test then self-test timer/watchdog (if used on the platform) can reboot the device  which is expected. The device will then fall back to the previous image.  Other than these two situations which other scenario does the OTA agent reboots directly?

>The overwrite issue faced is due to prvPAL_CreateFileForRx being called by a failed-validation firmware. The currently firmware is pending reboot and should not be allows to download new firmware. The call is made since the current image is no longer in self test, it change to rejected.

Lets consider we are executing from bank 1 and receiving image on bank 2 which fails validation. The bank 2 will now be erased and when next job doc is received `prvPAL_CreateFileForRx` is called to prepare for receiving image on bank 2. If image is in self-test then we have already filters in place to ignore job and data messages.

I think using `App_OTACompleteCallback` to wait for an application specific event on receiving `eOTA_JobEvent_Activate`  should solve the issue here. If not can you please also share how you are delaying the reboot and console logs ?

pvyawaha, 2019-06-19:
Hi sbertrand-witekio,

Closing this issues now, please update if you have more questions about reboot requirements.

Thanks.

closed 2019-06-19 by pvyawaha