aws/amazon-freertos #638: OTA Agent Callback Event Expansion

sbertrand-witekio, 2019-05-20:
The OTA Event Callback provides only very few information to the main application. The  `eOTA_JobEvent_Fail` event is only sent when the OTA Data ingest has an error.

Could this event be sent when the validation fails (due to rejection with error code, timeout, ...) ? 
Could other OTA Events be sent : start of download, download progress, end of download, state change, ... ?

pvyawaha, 2019-05-22:
Hi sbertrand-witekio,

We are actively looking into your feature request for adding more events and callbacks from OTA agent to the application. I will update this thread when I have more information.

Thank you for the feedback.

shreyasbharath, 2019-11-07:
Hi @pvyawaha has there been any progress on this? 

We are also after more fine-grained information about what the OTA agent is actually doing. This will allow us to log and diagnose issues faster.

shreyasbharath, 2019-11-07:
Also, I had a question. 

Can we query detailed error codes about why an OTA job failed for a thing as per this guide?

https://docs.aws.amazon.com/freertos/latest/userguide/ota-failure-codes.html

pvyawaha, 2019-12-18:
Hello,

We have added more OTA states with the release [201912.00](https://github.com/aws/amazon-freertos/releases/tag/201912.00) to provide more information to the application and we will continue expanding this further as required.
> Can we query detailed error codes about why an OTA job failed for a thing as per this guide?
> 
> https://docs.aws.amazon.com/freertos/latest/userguide/ota-failure-codes.html

Yes we can query error codes for the OTA jobs using the document you mentioned. Please let me know if you have any issues using the steps mentioned in this document.

pvyawaha, 2019-12-23:
Hello,

I am closing this issue now, if you have more questions on same topic please re-open and for any other topics please crate a new issue.

Thank you.

closed 2019-12-23 by pvyawaha