aws/amazon-freertos #649: [Feature] OTA Agent Initializing State

sbertrand-witekio, 2019-05-22:
**Is your feature request related to a problem? Please describe.**  
I have a state machine that starts the OTA Agent. The agent is started based on its state. However the agent remains in `eOTA_AgentState_NotReady` until its initialization is completed ( running the first operation of `prvOTAUpdateTask` ).
The state machine is unable to determine is the OTA Agent has failed to start or initialization is still ongoing.

**Describe the solution you would like.**  
Can the OTA Agent have a new state `eOTA_AgentState_Initializing` to describe its state when starting its thread ?

**Additional Behavior**
Once ready, can the agent notify the system with an event to the callback ? Perhaps `eOTA_AgentEvent_InitializationComplete`.

**Additional context**  
The initialization of the OTA Agent depends on MQTT connection which can incur delays.
Waiting for the OTA Agent to complete initialization is done with a task wait. This prevent the task running the state machine to get and process its events. 

Thank you!

pvyawaha, 2019-05-23:
Hi sbertrand-witekio,

The OTA_AgentInit can not be called before calling `OTA_AgentShutdown` which cleans up the context and deletes the task and this is also mentioned in source code comments. The OTA_AgentInit starts the `prvOTAUpdateTask` (and other init) and waits for the state to change from `eOTA_AgentState_NotReady` to  `eOTA_AgentState_Ready`. Before this state is set it subscribes to the OTA job notification topic by calling `prvSubscribeToJobNotificationTopics` which incurs the delay. I think your feature request to add such state for OTA initialization will be useful and I will update the thread when I have more information.

sbertrand-witekio, 2019-05-23:
Hi,

Thanks for considering adding an initialization state in the OTA Agent. Believe it would make the API more robust.

# OTA Agent Initialization State

> The `OTA_AgentInit` can not be called before calling `OTA_AgentShutdown` which cleans up the context and deletes the task and this is also mentioned in source code comments.

After reviewing the operation performed by `OTA_AgentShutdown`, there is unfortunately no cleanup nor task deletion performed by the `OTA_AgentShutdown`.  `OTA_AgentShutdown` is sending a event to the OTA Task to performed the cleanup and task deletion.

`OTA_AgentShutdown` suffers the same issue my state machine faces, there is no way of knowing if the OTA Agent task is currently performing initialization. `OTA_AgentShutdown` will report that the task is `eOTA_AgentState_NotReady` and allow the `OTA_AgentInit` to start a new task. 

My code was able to start 2 OTA Agent task with a 250ms wait time. The current implementation cannot guarantee a single instance of the OTA Agent.

# `OTA_AgentInit` Method

I feel `OTA_AgentInit` method has a number of flaws. I will list them for further consideration.
- The method return the state of the OTA Agent and not success or failure. The method has a number of checks and operation that are not reported to the caller 
  - if `pucThingName` is null, the return should be `invalid argument`
  - if `xTaskCreate` fails, the return should be given. I faced `errCOULD_NOT_ALLOCATE_REQUIRED_MEMORY` which is not reported
  - if `xOTA_Agent.eState` is valid, the return should inform of an invalid OTA Agent State
- The method changes multiple `xOTA_Agent` structure values. Since this object is global and used by the OTA Agent task, could any value changes be done if the OTA Agent State is `eOTA_AgentState_NotReady`. Thus moving any affectation `xOTA_Agent` of  under `if( xOTA_Agent.eState == eOTA_AgentState_NotReady )`

This would leads to a change in the API return value. I propose have `OTA_AgentStart` with the changes. The state is already given by `OTA_GetAgentState`. `OTA_AgentInit` would be deprecated and rewritten calling  `OTA_AgentStart`.

sbertrand-witekio, 2019-05-23:
This is my patch with the above comment changes : [0001-OTA-Update-OTA-Agent-Initialization.patch.txt](https://github.com/aws/amazon-freertos/files/3213789/0001-OTA-Update-OTA-Agent-Initialization.patch.txt)

pvyawaha, 2019-12-18:
Hi,

We have added a new state `eOTA_AgentState_Init` which will be the state of the OTA agent during initialization and after successful initialization it transitions to `eOTA_AgentState_Ready`. If the initialization is not successful it transitions to `eOTA_AgentState_Stopped`. The changes are released in [201912.00](https://github.com/aws/amazon-freertos/releases/tag/201912.00)

sbertrand-witekio, 2019-12-19:
Hi
Thank you for the update. I will review the changes when we migrate to the newer version.
Stan

pvyawaha, 2019-12-23:
Thank you Stan.

Please reopen the ticket or create a new one if you have any questions.

closed 2019-12-23 by pvyawaha