aws/amazon-freertos #652: Nordic NRF52840 RAM usage.

ljaynes87, 2019-05-22:
I have been developing my Nordic NRF52840 mqtt over BLE application using the ble-beta branch. With no modifications, the demo application used roughly 177KB of RAM. That is still quite a bit but I still had enough left over for my application. 

After switching to the release-1.5 branch I noticed the unmodified demo uses over 230KB of the 256KB available on the NRF52840. This is way to much to even fit my application into, making this release unusable for me in it's current state.

What caused the RAM usage to increase this much and is there anything I remove or modify to decrease some of the RAM usage?

gordonwang0, 2019-05-22:
Hi @ljaynes87,

The difference is due to the configuration of the FreeRTOS heap on release-1.5 vs feature/ble-beta.
- [On feature/ble-beta, the FreeRTOS heap is 120KB](https://github.com/aws/amazon-freertos/blob/4fa09aac8140ac05e89d3b7fb88571a6c8318e55/demos/nordic/nrf52840-dk/common/config_files/FreeRTOSConfig.h#L68).
- [On release-1.5, the FreeRTOS heap is 175KB](https://github.com/aws/amazon-freertos/blob/84fcfd97c7bfbfe630a4e944d4d0bd98180fb8ae/vendors/nordic/boards/nrf52840-dk/aws_demos/common/config_files/FreeRTOSConfig.h#L68)

We increased the size of the heap to occupy more of the available RAM.  You can decrease the size of this heap size, or make your application use this memory through `pvPortMalloc` and `vPortFree`.

ljaynes87, 2019-05-23:
Okay thank you. I missed the heap size increase. I will just decrease the heap size for now.

closed 2019-05-23 by gordonwang0