aws/amazon-freertos #654: OTA File Block Size

sbertrand-witekio, 2019-05-22:
The OTA Agent has a configuration to select the size of the data block : `otaconfigLOG2_FILE_BLOCK_SIZE`.

What are the valid file block size supported  ?
What block size are supported by the AWS Backend ?

I have tested 1024 and 512 bytes successfully.
I tried 2048 and 4096 bytes but the OTA failed with `kOTA_Err_MomentumAbort`.

In addition of increasing `otaconfigLOG2_FILE_BLOCK_SIZE` what other config needs to be change to support larger MQTT inbound buffer ?

  - `otaconfigLOG2_FILE_BLOCK_SIZE` is one
  - `mqttconfigRX_BUFFER_SIZE` may be one
  - `bufferpoolconfigBUFFER_SIZE` may be one
  - `ipconfigTCP_RX_BUFFER_LENGTH` may be one
  - any other to consider ?

pvyawaha, 2019-05-23:
Hi sbertrand-witekio,

The AWS backend supports blocks sizes from 256 bytes to 128 kilobytes but we might have limitations coming from the platform being used.

The error kOTA_Err_MomentumAbort happens when too many stream requests are sent to the AWS backend for data blocks without any response. 

I will try to reproduce this on ESP32 platform with block sizes similar to what you are using and update here.

Thanks.

sbertrand-witekio, 2019-05-23:
I was able to performed an OTA using 4k file block size following : https://forums.aws.amazon.com/thread.jspa?threadID=288974

I need to check which configuration values I used for the different buffer parameters.

It is indeed on ESP32.

Thanks

pvyawaha, 2019-05-24:
Ok, I am closing this ticket as it worked for you with the new configurations.

Thanks

closed 2019-05-24 by pvyawaha