aws/amazon-freertos #660: xTaskCheckForTimeOut() inconsistent behavior

macbany, 2019-05-24:
**Describe the bug**
When the function xTaskCheckForTimeOut() returns true, indicating a timeout, the value of pxTicksToWait is not consistently modified. If the timeout is due to an aborted delay or occurred in conjunction with a tick overflow, pxTicksToWait is not modified. However, pxTicksToWait is set to 0 when the timeout is "normal".

**System information**
- Version of Amazon FreeRTOS :  10.2.0

**Expected behavior**
The modification of pxTicksToWait should be consistent. Whenever true is returned from xTaskCheckForTimeOut() it implies that the calling function should no longer block. Therefore, pxTicksToWait should also be set to 0 whenever the return value is true.

Thank you!

RichardBarry, 2019-05-24:
Thanks for taking the time to report this.  I will start by looking at the situation when the delay is aborted.  xTaskCheckForTimeOut() is used internally by the kernel, but is also available as an API function for advanced users.  Is the behavior you observed when the function is used internally, or when you are calling it yourself?  If the latter I would be grateful if you could post just the shorted code snippet that demonstrates how you are using the function so we can replicate here - thanks.  In the mean time I will do my own tests too.

macbany, 2019-05-24:
Thanks for taking a look at it. I'm using it at the application layer and this a snippet that's similar to what I'm doing in my project.

```
// initialize timeout
TickType_t ticks_to_wait= pdMS_TO_TICKS(60000);
TimeOut_t timeout;
vTaskSetTimeOutState(&timeout);

// do work here for awhile

// check for timeout
xTaskCheckForTimeOut(&timeout, &ticks_to_wait);

// wait for EVENT_1...
// if it hasn't occurred, block until the timeout expires
// it it has already occurred, it should be handled regardless of the timeout

event_flags = xEventGroupWaitBits(event_group, EVENT_1, false, false, ticks_to_wait)

if (event_flags & EVENT_1)
{
    // do something
}
```

In the above code, if the timeout is "normal", ticks_to_wait will be set 0 and xEventGroupWaitBits won't block. However, if a tick overflow occurred, then ticks_to_wait doesn't get set to 0 and xEventGroupWaitBits will potentially block waiting for EVENT_1.

Of course it is possible to work around this at the application level by checking the return value of xTaskCheckForTimeOut and adjusting ticks_to_wait as needed but it seems like having consistent behavior from the API would be the best solution. Something like this at the bottom of xTaskCheckForTimeOut should do the trick assuming it doesn't mess with internal kernel behavior:

```
if (xReturn == pdTRUE)
{
    *pxTicksToWait = 0;
}
	
taskEXIT_CRITICAL();	

return xReturn;
```

macbany, 2019-05-24:
Interestingly, it looks like the case where pxTicksToWait gets set to 0 was added in v10.0.0. Prior to that pxTicksToWait was never changed if a timeout had occurred.

RichardBarry, 2019-05-28:
I've been looking into this and think this is a situation where your use of the API is not as we intended, mainly due to our documentation not being clear enough.  The pxTicksToWait parameter is only really intended to maintain state that the kernel uses itself, rather than be something that the application writer uses directly - the return value being the only value we intended the application writer to check.  Therefore pxTicksToWait only gets updated when there is no timeout, otherwise it just gets left.  As I said though, this is not at all clear from the way that parameter is documented.

On this line https://sourceforge.net/p/freertos/code/HEAD/tree/tags/V10.2.1/FreeRTOS/Source/tasks.c#l3234 the delay has been aborted, which is treated as a timeout, so pxTicksToWait is not updated.  I'm not sure setting it to 0 would be the right thing to do though as the time has not run down to 0, but (*pxTicksToWait -= xElapsedTime).

On this line https://sourceforge.net/p/freertos/code/HEAD/tree/tags/V10.2.1/FreeRTOS/Source/tasks.c#l3246 there is no timeout, so again I'm not sure what would be the right thing to set *pxTIcksToWait to.

On this line https://sourceforge.net/p/freertos/code/HEAD/tree/tags/V10.2.1/FreeRTOS/Source/tasks.c#l3258 the time has actually exceeded the requested timeout (presumably as the task had a priority below the task that was running at the time of the timeout), so it is a little tricky again.  *pxTicksToWait can't be set to a negative number and setting it to zero could be misleading if people are using the value.

I think we are all good in this case: https://sourceforge.net/p/freertos/code/HEAD/tree/tags/V10.2.1/FreeRTOS/Source/tasks.c#l3263

Which leaves the inconsistent case: https://sourceforge.net/p/freertos/code/HEAD/tree/tags/V10.2.1/FreeRTOS/Source/tasks.c#l3269 I'm still searching for the checkin that made the chance to see why it was done.  Its a very mature repo so its taking a while....

macbany, 2019-05-28:
I appreciate the effort you're putting into trying to come up with a good solution as I think the timeout feature is immensely useful. The way I was understanding and envisioning the function was that pxTicksToWait should be the remaining number of ticks that should be "waited" or allowed to block after xTaskCheckForTimeOut returns. So for the 5 cases you outlined:

1. https://sourceforge.net/p/freertos/code/HEAD/tree/tags/V10.2.1/FreeRTOS/Source/tasks.c#l3234 - The caller should not block. (ie. 0)

1. https://sourceforge.net/p/freertos/code/HEAD/tree/tags/V10.2.1/FreeRTOS/Source/tasks.c#l3246 - The caller should block indefinitely. (ie. portMAX_DELAY, since inf - x = inf)

1. https://sourceforge.net/p/freertos/code/HEAD/tree/tags/V10.2.1/FreeRTOS/Source/tasks.c#l3258 - Agreed that negative numbers wouldn't make sense but regardless, the caller should not block. (ie. 0)

1. https://sourceforge.net/p/freertos/code/HEAD/tree/tags/V10.2.1/FreeRTOS/Source/tasks.c#l3263 - This case is fine as is.

1. https://sourceforge.net/p/freertos/code/HEAD/tree/tags/V10.2.1/FreeRTOS/Source/tasks.c#l3269 - Again, a timeout has occurred so the caller should not block. (ie. 0)

In case it helps, the commit that introduced the change is r2519 but there doesn't seem to be any indication as to why it was changed.

RichardBarry, 2019-05-28:
The history.txt file just said "fix race condition in xTaskCheckForTimeOut", not very helpful (I was the one that wrote it too!).

aggarg, 2019-12-31:
As mentioned by Richard, this function is not well documented to reflect in the intended use of this API. I have updated the function documentation here: https://sourceforge.net/p/freertos/code/2786/tree//trunk/FreeRTOS/Source/include/task.h?diff=523769345fcbc93f2ac9582d:2785

It will be available in the next release. You can also refer to this page: https://www.freertos.org/xTaskCheckForTimeOut.html

Let us know if you have any other questions.

Thanks.

closed 2019-12-31 by aggarg