aws/amazon-freertos #676: Secure_Sockets test meet a problem when adding new target: Synopsys ARC board EMSDP

YuguoWH, 2019-05-27:
Hi, 
I had been porting Synopsys ARC board EMSDP to amazon_freertos for some time, but recently I met a problem passing one of the SECURE_SOCKETS tests, which is **AFQP_SECURE_SOCKETS_Threadsafe_SameSocketDifferentTasks**.

**System information:**

- Here's a github link to a clean version of the port:
    https://github.com/foss-for-synopsys-dwc-arc-processors/amazon-freertos/tree/arc_support

- Currently it is running on amazon-freertos v1.4.7, will update to v.1.4.8 later.

- Project built with _Synopsys MetaWare P2019.03_

 

**Description:**
Right now I could not locate the issue but during the first loop of the test, (Recv task _LARGE_BUFFER_HIGH_PRIORITY_; Sending task _SMALL_BUFFER_HIGH_PRIORITY_), after some time the recving task will suddenly stop ACK, causing the connection break.
At this case, the echo server will try to retransmit the package but after timeout it will send RST to reset. And at the same time, the echo server would also report error like this:
`2019/05/27 14:24:15 Error write tcp 192.168.137.1:9000->192.168.137.174:64242: wsasend: An existing connection was forcibly closed by the remote host. while sending data`

I also tried to run this test on a different type of ARC board EMSK which uses another WiFi module and driver. However the issue is still there. So I wonder if the issue reside at higher level?
The port adopted lwIP as tcp/ip stack, so I wonder if there could be any chance that my configuraion of lwIP went wrong?

Thank you!

dcgaws, 2019-05-29:
@YuguoWH nice job, it looks like you're close to completing a new port! Some questions:

1. What echo server implementation are you using?
2. When you run the non-TLS variation of the above test case (i.e. AFQP_SOCKETS_Threadsafe_SameSocketDifferentTasks) does it always pass?
3. Would it be possible for you to take a network packet capture on the echo server? Assuming you can get the Wi-Fi network interface to cooperate, the packet traces can tell you more concretely what's happening in TCP/IP (and TLS, pending the above).
4. What version of the lwIP library are you using?

Thank you.

YuguoWH, 2019-05-30:
@dcgaws 
1. I'm using two go language echo servers in tools/echo_server folder. 
2. Yes, the non-TLS version always pass.
3. Yes, I already have captured packets using wireshark. I'll double check that to make sure everything is right.
4. the version of LwIP in Amazon_freeRTOS is 2.0.3, and I was using version 2.1.0 on my original code. That's why I doubt when I port the code here, I could miss some configuration.

Thank you.

WRansohoff, 2019-06-04:
Are you able to run the tests successfully with one of the existing ports?

The most common type of issue that we encounter with TLS connections and tests are related to credential configurations, but we can probably rule that out if your tests pass on a qualified platform using the same credentials.

We also have a porting guide with a small section dedicated to lwip. It has a bit more information about porting that particular library:

https://docs.aws.amazon.com/freertos/latest/portingguide/porting-lwip.html

YuguoWH, 2019-06-06:
Hi @WRansohoff , I think the configurations are fine, all other tests but this one has succeed.

dcgaws, 2019-06-06:
@YuguoWH few more things to consider:

1. Are you using https://github.com/aws/amazon-freertos/blob/master/lib/secure_sockets/portable/lwip/aws_secure_sockets.c as your sockets API implementation?
2. If, in any case, you're also using the Amazon FreeRTOS port of mbedTLS, you can enable MBEDTLS_DEBUG_C at https://github.com/aws/amazon-freertos/blob/master/lib/third_party/mbedtls/include/mbedtls/config.h#L2121, rebuild, and re-run the failing tests. That will give you verbose serial console output at the TLS protocol client layer, which may be helpful in narrowing down the problem.
3. You can try https://docs.aws.amazon.com/freertos/latest/userguide/device-tester-for-freertos-ug.html#dev-test-versions-afr. Follow the instructions for testing a new in-progress port. I suggest this in particular because Device Tester has its own built-in echo server.
4. Finally, pending the above, my best theory is that there's a buffer management issue at the TCP/IP layer (i.e. whether caused by the app or not). When you're executing the multi-task TLS echo test, you're for some reason running out of netbuffers, and that causes the client to give up and disconnect.

aggarg, 2019-06-17:
Since we have haven't heard back, I am closing this issue. Please feel free to re-open in case you still have any problem.

Thanks.

closed 2019-06-17 by aggarg