aws/amazon-freertos #677: purpose of clientcredentialMQTT_BROKER_ENDPOINT

chegewara, 2019-05-28:
I am working with JITP on esp32 and i found that `clientcredentialMQTT_BROKER_ENDPOINT` has no purpose.
https://github.com/aws/amazon-freertos/blob/master/demos/common/include/aws_clientcredential.h#L38

The reason i am asking this question is because:
- for testing purpose i am using my development aws account,
- AWS IoT thing CA certificate, thing certificate and thing are registered under my account,
- thing shadow is changing values in thing on my account,
- clientcredentialMQTT_BROKER_ENDPOINT is set to my client account endpoint.

Code i am working on is based on `feature/ble-beta` branch.

From my tests i am guessing it is just DNS resolving address, because my and client endpoints both are in the same region, which is suggesting the name points to the same IP address.

mradulan, 2019-05-28:
clientcredentialMQTT_BROKER_ENDPOINT is the URL of the MQTT broker to connect to. Our MQTT agent has it as one of connection parameters. Yes it is DNS resolving address, but it is essential to provide the endpoint address.

chegewara, 2019-05-29:
> but it is essential to provide the endpoint address.

Like i mentioned, i can use endpoint from 2 different not related accounts and application is running without issues. I believe it is not a problem, because client is routed by certificates, but im not aws expert, so i decided i will report this strange behavior just in case.

If it is expected behavior then feel free to close this issue. Thanks.

aggarg, 2019-05-30:
Hi,

Some endpoints do overlap in the IPs they resolve to but all endpoints do not share the same set of IPs. The co-location of endpoints is not guaranteed and we strongly recommend that you use the endpoint corresponding to your account (the one you get using describe-endpoint) to connect to AWS IoT.

Thanks.

closed 2019-05-30 by chegewara