aws/amazon-freertos #678: OTA Agent : eImageState consistency

sbertrand-witekio, 2019-05-28:
**Describe the bug**
The value of the OTA Agent member `eImageState` is not consistent.
The value is set upon  validation of the OTA (rejected or accepted).
The value is not updated when a new OTA is processed.
When the OTA agent request activation of the image, the value return is what was set last ( Rejected or Accepted ) which is inconsistent with the OTA Agent initialization value : `eOTA_ImageState_Unknown`.

**Expected behavior**
When the OTA Agent requests activation, the `eImageState` provided by `OTA_GetImageState` should be consistent and equal to `eOTA_ImageState_Unknown`.
A new state could be made to identify a new image `eOTA_ImageState_New`.

**To reproduce**
1. Add a printout of the `eImageState` before performing activation.
1. Perform an OTA. Before calling activation, the value of `eImageState` should be 0.
2. Without any reboot after successful validation, perform another OTA
3. Before calling activation, read the value of `eImageState`. 

**Additional context**
The OTA are performed on ESP32.

Thank you!

sbertrand-witekio, 2019-05-31:
Patch to set the `eImageState` before calling the activation callback : 
[0002-OTA-Agent-image-state-inconsistent-fix.patch.txt](https://github.com/aws/amazon-freertos/files/3242061/0002-OTA-Agent-image-state-inconsistent-fix.patch.txt)

pvyawaha, 2019-12-24:
Hi,

Please let us know if you still have this issue on latest Amazon FreeRTOS release.

tgsong, 2020-01-07:
We're closing this issue for now, if you have further questions you can reopen this issue.

closed 2020-01-07 by tgsong