aws/amazon-freertos #679: If I have an MQTT agent connected and I'm trying to add a Shadow client - Connection fail

egoltzman, 2019-05-28:
I have a shadow connection that I used in my application, in parallel to that connection I need to publish some ad hoc topics for handling special application issues (non-shadow custom topics).
When I used my initial end point URL I was able to create both a shadow client as well as an MQTT one and implement this need.  Lately I created some more things (since then I used a single thing that was created a long time ago) and for them I was forced to use my end point with the "-ats" extension, but now, from some reason the shadow connection fail if the MQTT one is already connected.

My first question is why that and if there is a walk around it.

My second question is, based on the answer to [Issue #327](https://github.com/aws/amazon-freertos/issues/327), will I be able to use the shadow MQTT connection for those custom topics in the next version?

Thanks,
Eyal

mradulan, 2019-05-30:
Regarding the 1st question, AWS moved the endpoints to "-ats" because of web consortium move off of Verisign root CAs. But that should not cause the disconnection of shadow if MQTT is already connected.

egoltzman, 2019-06-01:
I know that but still, when I'm using my old thing that does not have to use the -ats I can create an MQTT connection and then a shadow connection.
If I add tho this same thing end point the -ats the same test does not pass, that is, the MQTT connection connected successfully and next, when I try to create the shadow connection it fail with:
"Shadow_ClientConnect unsuccessful, returned 2."

WRansohoff, 2019-06-10:
Can you double-check that you aren't using the same MQTT Client ID for both connections? MQTT Client IDs should be unique for each connection to the broker, and our demos often use a device's 'Thing name' for its client ID. If you used a similar approach and made multiple connections from the same device, that could cause issues because the MQTT broker only keeps track of one connection per client ID.

closed 2019-08-16 by dcgaws