aws/amazon-freertos #687: ESP32 ECHO Demo failing to connect to AWS IOT (MQTT)

alexanderturner, 2019-05-30:
**Describe the bug**
On running the AWSSetup.py script to produce certificates and header files a few times the ESP32 demo fails to connect to AWS IOT without meaningful failure messages. I've tried rerunning the script a number of times, have validated the certificates in `demos/common` as well as the endpoint being active in the console and correct in the app. 

**System information**
- Which hardware board or part numbers? ESP32-WROOM-32
- IDE used - text editor
- Operating System [Windows|Linux|MacOS] - MacOS
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it) v1.4.7-194-ga1e88df8
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server] - Echo demo 
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
Connect to Amazon MQTT broker

**Screenshots or console output**
```
I (3188) WIFI: SYSTEM_EVENT_STA_GOT_IP
2 305 [IP-task] vDHCPProcess: offer a00015fip
3 306 [main] WiFi Connected to AP. Creating tasks which use network...
4 307 [main] Write certificate...
I (3208) PKCS11: Initializing NVS partition: "storage"
5 312 [main] Write device private key...
6 322 [main] Creating MQTT Echo Task...
7 322 [MQTTEcho] MQTT echo attempting to connect to a2vir6x2xbxfvq-ats.iot.ap-southeast-2.amazonaws.com.
8 1823 [MQTTEcho] ERROR:  MQTT echo failed to connect with error 1.
9 1823 [MQTTEcho] MQTT echo test could not connect to broker.
10 1823 [MQTTEcho] MQTT echo demo finished.
11 1823 [MQTTEcho] ----Demo finished----
```

**To reproduce**
Build MQTT echo demo

**Code to reproduce the bug**

**Additional context**
Nil

Thank you!

mradulan, 2019-05-30:
Hi @alexanderturner , so the ECHO demo is failing to connect with return error code `eMQTTAgentFailure` . Would you be able to enable the debug logs in the file`demos/espressif/esp32_devkitc_esp_wrover_kit/common/config_files/aws_mqtt_config.h` on line 69 with the following code.
`#define mqttconfigENABLE_DEBUG_LOGS    1`
 That will help us check why the MQTT_AGENT_Connect failing.

alexanderturner, 2019-05-30:
thanks @mradulan :)
```
5 309 [main] Write device private key...
6 319 [main] Creating MQTT Echo Task...
7 319 [MQTTEcho] MQTT echo attempting to connect to a2vir6x2xbxfvq-ats.iot.ap-southeast-2.amazonaws.com.
8 319 [MQTTEcho] Sending command to MQTT task.
9 320 [MQTT] Received message 10000 from queue.
10 1320 [IP-task] Socket sending wakeup to MQTT task.
11 1820 [MQTT] About to close socket.
12 1820 [MQTT] Socket closed.
13 1820 [MQTT] Stack high watermark for MQTT task: 4688
14 1820 [MQTT] Notifying task.
15 1820 [MQTT] Received message 0 from queue.
16 1820 [MQTTEcho] Command sent to MQTT task failed.
17 1820 [MQTTEcho] ERROR:  MQTT echo failed to connect with error 1.
18 1821 [MQTTEcho] MQTT echo test could not connect to broker.
19 1821 [MQTTEcho] Sending command to MQTT task.
20 1822 [MQTT] Received message 20000 from queue.
21 1822 [MQTT] Notifying task.
22 1823 [MQTTEcho] Command sent to MQTT task passed.
23 1823 [MQTTEcho] MQTT echo demo finished.
24 1823 [MQTTEcho] ----Demo finished----
```

mradulan, 2019-05-31:
Can you check the policy attached to your certificate in your AWS account and see if `iot:Connect`, `iot:Publish`, `iot:Subscribe` and `iot:Receive` all are present. Go to IoT Core service, in that **Manage** ->**Things**-> <your thing>->**Security** -> <click on the certificate> -> **Policies**-><click on the policy document>

mradulan, 2019-05-31:
When the policy document is incorrect or not attached, I can reproduce the issue.

alexanderturner, 2019-05-31:
My thing and policy are both being created by the init script, the policy attached to the cert is as follows:

``{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": "iot:Connect",
      "Resource": "arn:aws:iot:ap-southeast-2:111:*"
    },
    {
      "Effect": "Allow",
      "Action": "iot:Publish",
      "Resource": "arn:aws:iot:ap-southeast-2:111:*"
    },
    {
      "Effect": "Allow",
      "Action": "iot:Subscribe",
      "Resource": "arn:aws:iot:ap-southeast-2:111:*"
    },
    {
      "Effect": "Allow",
      "Action": "iot:Receive",
      "Resource": "arn:aws:iot:ap-southeast-2:111:*"
    }
  ]
}```

mradulan, 2019-05-31:
can you try doing `make erase_flash` before `make flash monitor`. This will clear any unnecessary data in the flash which might be blocking the socket to be connected to the broker.

alexanderturner, 2019-06-01:
Still no luck. I then went ahead and ran `SetupAWS.py cleanup && SetupAWS.py setup` on the same device name config and ran into the same error above. I then modified configure.json and created a new device name and encountered the same issue. Again I'm building from v1.4.7-194-ga1e88df8

mradulan, 2019-06-03:
with v1.4.7-194-ga1e88df8, I am able to get a working MQTT demo on ESP32. I also followed the Getting started guide of ESP32. The successful demo log is as below:
`I (1376) WIFI: SYSTEM_EVENT_STA_CONNECTED
1 427 [IP-task] vDHCPProcess: offer c0a838b8ip
I (4396) event: sta ip: 192.168.56.184, mask: 255.255.224.0, gw: 192.168.32.1
I (4396) WIFI: SYSTEM_EVENT_STA_GOT_IP
2 428 [IP-task] vDHCPProcess: offer c0a838b8ip
3 428 [main] WiFi Connected to AP. Creating tasks which use network...
4 429 [main] Write certificate...
I (4406) PKCS11: Initializing NVS partition: "storage"
5 433 [main] Write device private key...
6 443 [main] Creating MQTT Echo Task...
7 444 [MQTTEcho] MQTT echo attempting to connect to aofjsf37kyrxb-ats.iot.ap-southeast-2.amazonaws.com.
8 1595 [MQTTEcho] MQTT echo connected.
9 1595 [MQTTEcho] MQTT echo test echoing task created.
10 1636 [MQTTEcho] MQTT Echo demo subscribed to freertos/demos/echo
11 1687 [MQTTEcho] Echo successfully published 'Hello World 0'
12 1718 [Echoing] Message returned with ACK: 'Hello World 0 ACK'
13 2240 [MQTTEcho] Echo successfully published 'Hello World 1'
14 2271 [Echoing] Message returned with ACK: 'Hello World 1 ACK'
15 2793 [MQTTEcho] Echo successfully published 'Hello World 2'
16 2824 [Echoing] Message returned with ACK: 'Hello World 2 ACK'
17 3346 [MQTTEcho] Echo successfully published 'Hello World 3'
18 3376 [Echoing] Message returned with ACK: 'Hello World 3 ACK'
19 3879 [MQTTEcho] Echo successfully published 'Hello World 4'
20 3930 [Echoing] Message returned with ACK: 'Hello World 4 ACK'
21 4421 [MQTTEcho] Echo successfully published 'Hello World 5'
22 4452 [Echoing] Message returned with ACK: 'Hello World 5 ACK'
23 4954 [MQTTEcho] Echo successfully published 'Hello World 6'
24 4984 [Echoing] Message returned with ACK: 'Hello World 6 ACK'
25 5496 [MQTTEcho] Echo successfully published 'Hello World 7'
26 5528 [Echoing] Message returned with ACK: 'Hello World 7 ACK'
27 6049 [MQTTEcho] Echo successfully published 'Hello World 8'
28 6080 [Echoing] Message returned with ACK: 'Hello World 8 ACK'
29 6602 [MQTTEcho] Echo successfully published 'Hello World 9'
30 6643 [Echoing] Message returned with ACK: 'Hello World 9 ACK'
31 7155 [MQTTEcho] Echo successfully published 'Hello World 10'
32 7197 [Echoing] Message returned with ACK: 'Hello World 10 ACK'`

Do you think you can verify if you can run any other demos like `vStartSimpleTCPServerTasks` or `vStartTCPEchoClientTasks_SeparateTasks`?

WRansohoff, 2019-06-04:
Could you also double-check that your Thing's certificate is active in AWS? Sometimes they can be created in a deactivated state. When you view a Thing's certificate in the IoT Core dashboard, the 'Actions' button in the upper-right lets you activate, deactivate, or revoke it.

You can also use the `describe-certificate` API, which should return `ACTIVE` in the `status` field for an active certificate:

https://docs.aws.amazon.com/cli/latest/reference/iot/describe-certificate.html

alexanderturner, 2019-06-06:
Thanks for this - feel like I've tried everything and not sure where its failing; a bit hard to determine with the lack of log verbosity on the failure. I've just deleted and recloned the freertos repo and run through the commands I'm running to execution. I've placed them in this gist: https://gist.github.com/alexanderturner/1f35dd4a8c62a290993a423c5a216bc6

WRansohoff, 2019-06-06:
Sorry that you're still having trouble, but thanks for bearing with us.

The steps that you listed to do look correct, and the demo ran correctly when I followed them in the ap-southeast-2 region on an ESP32-DevKit-C board. One more idea, the demo defaults to calling an 'ATS' endpoint, which uses certificates signed by Amazon's root CA. Could you try following the first step in the 'How to set up your Amazon Trust Services endpoint' at the end of this page, and then re-running the demo?

https://aws.amazon.com/blogs/iot/aws-iot-core-ats-endpoints/

It might be possible that your account doesn't have an ATS endpoint set up yet. You could also try removing '-ats' from the `clientcredentialMQTT_BROKER_ENDPOINT` URL in `demos/common/include/aws_clientcredential.h` to use the legacy endpoint, but be aware that it looks like ATS endpoints will be preferred in the future.

alexanderturner, 2019-06-07:
Thanks @WRansohoff 
SInce my last post I've now created an IAM user with both AmazonFreeRTOSFullAccess, AWSIoTFullAccess policies (instead of root access key), re-cloned the repo and have tried building with and without the ATS cert endpoint. I'm getting the same error with the ATS endpoint (including when confirming the endpoint with `describe-endpoint`. When I connect without the -ats, I'm logging a 404 error for no shadow device existing; when creating a shadow and adding it to my thing, I'm still seeing the error. For your reference, I'm using a NodeMCU ESP32-S, which is a ESP32-WROOM-32 with a UART-USB driver sitting in front of it. Something is starting to feel very broken here...

chegewara, 2019-06-07:
Maybe try to read thing shadow with http tool like postman or similar and see if it is reachable? 
This way you can at least confirm that everything is configured properly on AWS IoT side and you have issue with esp32/AFR.

WRansohoff, 2019-06-07:
If you're getting some sort of 'cannot connect' error when you call `describe-endpoint` from the command line, that definitely sounds like there is some sort of permissions issue on the AWS side. Just to make sure I'm not misunderstanding, you were able to call this command successfully when setting up the device:

    aws iot describe-thing --thing-name github_test

But you get an error when calling this command?:

    aws iot describe-endpoint --endpoint-type iot:Data-ATS

alexanderturner, 2019-06-07:
No issues with the AWS CLI.
``` ~  aws iot describe-thing --thing-name github_test                                                                                                                                                        488ms  Sat Jun  8 09:40:39 2019

{
    "version": 1,
    "thingName": "github_test",
    "defaultClientId": "github_test",
    "attributes": {},
    "thingArn": "arn:aws:iot:ap-southeast-2:222105418958:thing/github_test",
    "thingId": "124db0c2-66b3-4a28-a937-51f165e62598"
}
 ~  aws iot describe-endpoint --endpoint-type iot:Data-ATS                                                                                                                                                1459ms  Sat Jun  8 09:40:50 2019

{
    "endpointAddress": "a2vir6x2xbxfvq-ats.iot.ap-southeast-2.amazonaws.com"
}
 ~ ```

chegewara, 2019-06-08:
Sometimes, when im having problems with certificates and im not sure if i am using correct certificate or not, i am using this mosquitto command for test:
```
mosquitto_pub --cafile root.cert --cert deviceCertAndCACert.crt --key deviceCert.key -h <prefix>.iot.us-east-1.amazonaws.com -p 8883 -q 1 -t foo/bar -I anyclientID --tls-version tlsv1.2 -m "Hello" -d
```

gignicu, 2019-06-20:
I have an issue with a self signed root CA and device certificates, set correctly in:
keyCLIENT_CERTIFICATE_PEM
keyCLIENT_PRIVATE_KEY_PEM
keyJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM

I am using the latest version of FreeRTOS and the default mqtt demo with a STM32 L4 B-L475E-IOT01A2 board, (WiFi firmware version is: C3.5.2.5.STM, WiFi firmware is up-to-date.)
I have 2 scenarios:
1. Just in time provisioning - device certificate and thing don't exist in AWS IoT Core. Root CA already registered.
2. Device certificate and thing are active in AWS IoT Core, I used mosquitto_pub to do JITP, as explained in https://github.com/aws-samples/aws-iot-device-management-workshop/blob/master/AWS_IoT_Device_Management_Workshop.md#JITP
The only difference is that mosquitto requires rootca and device pem files bundled together.

For both error is: 
Details:"\r\n[TCP SSL] Initialize Identity failed\r\n[TCP SSL] Advanced context, failed to connect\r\nERROR: Unknown Error\r\nUsage: P6 <0 = Stop, 1 = Start> \r\n> \025", '\0' <repeats 1253 times>
In es_wifi.c, ES_WIFI_StartClientConnection, line 1608.

The demo works with certificates generated directly in AWS IoT Core.
Self signed certificates work with mosquitto_pub and the python sdk publisher subscriber demo.

There is also an error in iot_network_afr.c, pAfrCredentials->pRootCa is NULL, I had to set it manually in SOCKETS_SetSockOpt in _tlsSetup, line 286.

chegewara, 2019-06-20:
Try this 
https://aws.amazon.com/blogs/iot/setting-up-just-in-time-provisioning-with-aws-iot-core/
I also like to use those scripts to create certificates for IoT JITP:
https://github.com/aws-samples/aws-iot-jitp-sample-scripts

alexanderturner, 2019-06-20:
Fixed here: https://github.com/aws/amazon-freertos/pull/814

closed 2019-06-20 by alexanderturner

aggarg, 2019-06-20:
JITP requires that the device sends both the device certificate and the CA certificate when connecting. This is not supported by the Inventek WiFi module. Therefore you will not be able to use offload SSL and you will have to do SSL in software. Remove the project level define USE_OFFLOAD_SSL to do that:

* Right click on the project and select properties. 
* Go to C/C+++ Build --> Settings --> MCU GCC Compiler --> Preprocessor.
* Select USE_OFFLOAD_SSL and delete it. 

I followed the following blog for JITP: https://aws.amazon.com/blogs/iot/setting-up-just-in-time-provisioning-with-aws-iot-core/

Note that the provisioning template in above post only allows Connect and Publish operations in AWS IoT while our demo does Subscribe and Disconnect also. Therefore, I needed to update the provisioning template to allow all IoT operations:

```
{
    "templateBody":"{ \"Parameters\" : { \"AWS::IoT::Certificate::Country\" : { \"Type\" : \"String\" }, \"AWS::IoT::Certificate::Id\" : { \"Type\" : \"String\" } }, \"Resources\" : { \"thing\" : { \"Type\" : \"AWS::IoT::Thing\", \"Properties\" : { \"ThingName\" : {\"Ref\" : \"AWS::IoT::Certificate::Id\"}, \"AttributePayload\" : { \"version\" : \"v1\", \"country\" : {\"Ref\" : \"AWS::IoT::Certificate::Country\"}} } }, \"certificate\" : { \"Type\" : \"AWS::IoT::Certificate\", \"Properties\" : { \"CertificateId\": {\"Ref\" : \"AWS::IoT::Certificate::Id\"}, \"Status\" : \"ACTIVE\" } }, \"policy\" : {\"Type\" : \"AWS::IoT::Policy\", \"Properties\" : { \"PolicyDocument\" : \"{\\\"Version\\\": \\\"2012-10-17\\\",\\\"Statement\\\": [{\\\"Effect\\\":\\\"Allow\\\",\\\"Action\\\": [\\\"iot:*\\\"],\\\"Resource\\\" : [\\\"*\\\"]}]}\" } } } }",
    "roleArn":"arn:aws:iam::123456789012:role/JITPRole"
}
```
I also needed to make the following configuration updates:

* Update ```MQTT_TIMEOUT_MS``` to 60000 in ```demos/mqtt/iot_demo_mqtt.c```. This is needed because TLS handshake is slow when TLS is done in software.
* Increase ```democonfigDEMO_STACKSIZE``` to 2048 in ```vendors/st/boards/stm32l475_discovery/aws_demos/config_files/aws_demo_config.h```.
* Increase ```configTOTAL_HEAP_SIZE``` to ( 70 * 1024 ) in ```vendors/st/boards/stm32l475_discovery/aws_demos/config_files/FreeRTOSConfig.h```.

Hope that helps.

Thanks.

gignicu, 2019-06-21:
Thank you @aggarg for your detailed response.
Based on your indications, I managed to make my second scenario work, meaning using a self signed device certificate that was already active in the IoT Core.
The first scenario, JITP with a device certificate connecting for the first time to IoT Core didn't work resulting in timeout: (MQTT connection 0x20001548, CONNECT operation 0x200016d0) Wait complete with result TIMEOUT. 
I've even increased MQTT_TIMEOUT_MS to 300000, same result.
I don't know where to look for logs, if any.
Thanks again for your time.

aggarg, 2019-06-21:
Hi @gignicu,

Would you please explain what you did for the JITP case? Did you follow the same steps as mentioned in this blog post: https://aws.amazon.com/blogs/iot/setting-up-just-in-time-provisioning-with-aws-iot-core/

More specifically, you need to register your CA along with a template and role. When the device connects for the first time, the resources are provisioned as specified in the template:
* A thing and policy is created.
* The policy is attached to the certificate. 
* The certificate is attached to the thing.
* The certificate status is updated to ACTIVE.

Would you please check that after the connection attempt, the certificate exists in your account? Also, check the policy attached to the certificate. 

Thanks.

gignicu, 2019-06-24:
Hi @aggarg,

Yes, JITP with mosquitto_pub works. It goes through every stage, as I see in Cloud Trail:

- Register a certificate and set its status to PENDING_ACTIVE.
- Create one thing resource.
- Create one policy resource.
- Attach the policy to the certificate.
- Attach the certificate to the thing.

The last one doesn't happen.
- Update the certificate status to ACTIVE.
The certificate stays in PENDING_ACTIVE, I have to manually change it to ACTIVE using aws cli.

My problem in the previous post is that it doesn't even connect to mqtt, resulting in Timeout.

Thank you.

aggarg, 2019-06-24:
Hi @gignicu ,

Just for my understanding, are you saying:

* JITP works correctly when you use mosquito_pub.
* When you use Amazon FreeRTOS, the certificate status is not changed to ACTIVE and stays in PENDING_ACTIVE.

If so, are you able to connect to AWS IoT broker after you manually activate the certificate? Also, which AWS region are you using?

Thanks.

gignicu, 2019-06-24:
Hi @aggarg ,

No, JITP only works when I'm using mosquito_pub. 

When I'm using Amazon FreeRTOS it doesn't work, it times out when MQTT_TIMEOUT_MS expires.
I'm guessing that the self signed root CA isn't used like in the mosquito_pub scenario, when it's bundled together with the device certificate.

Amazon FreeRTOS works when the certificate is already active in IoT Core (after registering it with mosquito_pub).

Thanks

aggarg, 2019-06-24:
Would you please check the following:

* You have defined ```keyJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM``` to the root CA in ```aws_clientcredentials_keys.h```.
* You have removed ```USE_OFFLOAD_SSL``` from the project as I mentioned above.

Have you made any other change?

Thanks.

chegewara, 2019-06-25:
When i started to work with JITP i got exactly the same problems, my device certificates very often were registered with `PENDING_ACTIVE` status. But it had something to do with generating certificates.
Now when i have set of scripts to generate CA and device certificate i have no longer issue and im not changing AFR code responsible for handling certificates.
I know i am working with different device (esp32), but i thought i will share my experience since i had the same issue.

gignicu, 2019-06-25:
@aggarg, I changed what you recommended and I am using the following:

- Amazon FreeRTOS 201906.00 Major, Connect to AWS IoT - ST, for STM32L4 Discovery kit IoT node
- STM32 L4 B-L475E-IOT01A2 device, (WiFi firmware version is: C3.5.2.5.STM, WiFi firmware is up-to-date.)
- Removed USE_OFFLOAD_SSL
- Updated MQTT_TIMEOUT_MS to 60000
- Increased democonfigDEMO_STACKSIZE to 2048
- Increased configTOTAL_HEAP_SIZE to ( 70 * 1024 )
- keyCLIENT_CERTIFICATE_PEM - Device certificate pem file
- keyCLIENT_PRIVATE_KEY_PEM - Device certificate private key file
- keyJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM - self signed Root CA pem file

Logs:
9 34725 [iot_thread] [INFO ][MQTT][34725] Establishing new MQTT connection.
10 34732 [iot_thread] [INFO ][MQTT][34732] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRIC
11 34747 [iot_thread] [INFO ][MQTT][34747] (MQTT connection 0x20000f40, CONNECT operation 0x200010c8) Waiting for operation completion.
12 94760 [iot_thread] [INFO ][MQTT][94760] (MQTT connection 0x20000f40, CONNECT operation 0x200010c8) Wait complete with result TIMEOUT.
13 94772 [iot_thread] [ERROR][MQTT][94772] Failed to establish new MQTT connection, error TIMEOUT.
14 94781 [iot_thread] [INFO ][MQTT][94781] Network connection closed on error.
15 94790 [iot_thread] [INFO ][MQTT][94790] (MQTT connection 0x20000f40) Network connection destroyed.
16 94799 [iot_thread] [ERROR][DEMO][94799] MQTT CONNECT returned error TIMEOUT.
17 94806 [iot_thread] [INFO ][MQTT][94806] MQTT library cleanup done.
18 94813 [iot_thread] [ERROR][DEMO][94813] Error running demo.
19 94819 [iot_thread] [INFO ][INIT][94819] SDK cleanup done.

Please remember that the same setup works when:
- using mosquitto_pub
- using Amazon FreeRTOS and the certificate is already ACTIVE in IoT Core

@chegewara, you are right. When I used the provisioning template on the official AWS documentation page, I always had an error that the policy already exists and the certificate wasn't updated from PENDING_ACTIVE to ACTIVE.
I changed the template to use an existing policy and changed the PolicyDocument to PolicyName. 
Then it worked, certificate is ACTIVE, I can see the event history in Cloud Trail.
Remember, this scenario is carried out using mosquitto_pub, on my windows machine.
Maybe there is something wrong with my ST device.

Thank you.

aggarg, 2019-06-27:
Hi @gignicu,

At this point, I am a bit unsure about what might be going wrong. Following this blog with the changes I I suggested above worked for me: https://aws.amazon.com/blogs/iot/setting-up-just-in-time-provisioning-with-aws-iot-core/

Would you like to get on a joint debugging session to debug this issue?

Thanks.

gignicu, 2019-06-28:
Hi @aggarg,
Yes, a joint debugging session would be great.
Thank you.

aggarg, 2019-07-01:
Hi @gignicu,

Please use the **Business Contact** link on the following page to send your contact details: https://www.freertos.org/RTOS-contact-and-support.html

Please mention in your email that it is for aggarg.

Thanks.

gignicu, 2019-07-29:
Thank you so much @aggarg for your support during our debug session.
It was very helpful and it finally worked with your jitp template.
I modified it a bit to set the device name from the certificates CommonName.
The _Setting Up Just-in-Time Provisioning with AWS IoT Core_ blogpost works perfectly.
I have messed up some certificates in my previous tests.

Sorry for writing with such a delay.
Thank you again for your help.

aggarg, 2019-07-29:
I am glad that it worked for you. 

Thanks.

monkeytronics, 2019-11-12:
Hi @aggarg and @gignicu. It sounds like I'm in exactly the same place. I don't suppose you'd mind sharing your findings. I'm wokring with ESP32 and struggling to get my JITP to run. I've had a very good discussin with @chegewara on issue: https://github.com/aws/amazon-freertos/issues/1119. That describes in detail what is going on in my case. 

In a nutshell. JITP working in mosquitto. Generic MQTT demo working fine. But when I load the device key and cert in DER format directly into the NVS partition at 0x317000, and put the root.cert into the aws_client_credential_keys.h hardcoded, it doesn't work. I am getting the following. I added a few more Info logs to see how many times various things were called and the length of the certs. 

```
5 304 [IP-task] vDHCPProcess: offer c0a89647ip
6 304 [iot_thread] [INFO ][DEMO][3040] Successfully initialized the demo. Network type for the demo: 1
7 304 [iot_thread] [INFO ][MQTT][3040] MQTT library successfully initialized.
8 304 [iot_thread] [INFO ][DEMO][3040] MQTT demo client identifier is CO2_Testy4 (length 10).
I (3471) PKCS11: Initializing NVS partition: "storage"
I (3511) PKCS11: **** PKCS11_PAL_FindObject, pcLabel= Device Priv TLS Key, ulDatusLengthaSize = 19 bytes
I (3511) PKCS11: **** PKCS11_PAL_GetObjectValue
I (3511) PKCS11: pkcs11palFILE_NAME_KEY
I (3511) PKCS11: Reading file P11_Key
I (3521) PKCS11: Value: 0╝╗╔, length = 1194
I (3521) PKCS11: **** PKCS11_PAL_GetObjectValue
I (3531) PKCS11: pkcs11palFILE_NAME_KEY
I (3531) PKCS11: Reading file P11_Key
I (3541) PKCS11: Value: 0╝╗╔, length = 1194
I (3661) PKCS11: **** PKCS11_PAL_FindObject, pcLabel= Device Cert, ulDatusLengthaSize = 11 bytes
I (3661) PKCS11: **** PKCS11_PAL_GetObjectValue
I (3671) PKCS11: pkcs11palFILE_NAME_CLIENT_CERTIFICATE
I (3671) PKCS11: Reading file P11_Cert
I (3681) PKCS11: Value: 0╚0╔╗  , length = 784
I (3681) PKCS11: **** PKCS11_PAL_GetObjectValue
I (3691) PKCS11: pkcs11palFILE_NAME_CLIENT_CERTIFICATE
I (3691) PKCS11: Reading file P11_Cert
I (3701) PKCS11: Value: 0╚0╔╗  , length = 784
I (3701) PKCS11: **** PKCS11_PAL_GetObjectValue
I (3711) PKCS11: pkcs11palFILE_NAME_CLIENT_CERTIFICATE
I (3711) PKCS11: Reading file P11_Cert
I (3721) PKCS11: Value: 0╚0╔╗  , length = 784
I (6921) PKCS11: **** PKCS11_PAL_GetObjectValue
I (6921) PKCS11: pkcs11palFILE_NAME_KEY
I (6921) PKCS11: Reading file P11_Key
I (6921) PKCS11: Value: 0╝╗╔, length = 1194
9 887 [iot_thread] [INFO ][MQTT][8870] Establishing new MQTT connection.
10 887 [iot_thread] [INFO ][MQTT][8870] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
11 887 [iot_thread] [INFO ][MQTT][8870] (MQTT connection 0x3ffb3394, CONNECT operation 0x3ffb36f0) Waiting for operation completion.
12 1387 [iot_thread] [INFO ][MQTT][13870] (MQTT connection 0x3ffb3394, CONNECT operation 0x3ffb36f0) Wait complete with result TIMEOUT.
13 1387 [iot_thread] [ERROR][MQTT][13870] Failed to establish new MQTT connection, error TIMEOUT.
I (14001) wifi: state: run -> init (0)
I (14011) wifi: pm stop, total sleep time: 4290300 us / 11992268 us

I (14021) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (14021) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 8
14 1387 [iot_thread] [WARN ][NET][13870] Failed to close connection.
15 1387 [iot_thread] [INFO ][MQTT][13870] Network connection closed on error.
16 1387 [iot_thread] [INFO ][MQTT][13870] (MQTT connection 0x3ffb3394) Network connection destroyed.
17 1387 [iot_thread] [ERROR][DEMO][13870] MQTT CONNECT returned error TIMEOUT.
18 1387 [iot_thread] [INFO ][MQTT][13870] MQTT library cleanup done.
19 1387 [iot_thread] [ERROR][DEMO][13870] Error running demo.
20 1393 [iot_thread] [INFO ][INIT][13910] SDK cleanup done.
21 1393 [iot_thread] [INFO ][DEMO][13930] -------DEMO FINISHED-------

```

chegewara, 2019-11-12:
Your wifi shouldnt enter sleep mode:
```
I (14001) wifi: state: run -> init (0)
I (14011) wifi: pm stop, total sleep time: 4290300 us / 11992268 us
```

aggarg, 2019-11-19:
I see an update here https://github.com/aws/amazon-freertos/issues/1119#issuecomment-552784353. Is everything working for you?

Thanks.

monkeytronics, 2019-11-19:
@aggarg Yes. JITP is now working. Thanks. I was not understanding the certificates.