aws/amazon-freertos #689: OTA Agent : In Progress Notification Duplication

sbertrand-witekio, 2019-05-30:
**Describe the bug**
The OTA Agent will send in progress notification multiple times if duplicated packets are received when the number of received packets is multiple of `OTA_UPDATE_STATUS_FREQUENCY`.
This cause unnecessary packet being set.

**System information**
- ESP32
- AFR : v1.4.7-48-gc03bfdca

**Expected behavior**
The OTA Agent send a unique update message for a given number of received blocks.

**Console output**
https://gist.github.com/sbertrand-witekio/603a242dbc8e206643bb999516bece8b

Thank you!

pvyawaha, 2019-05-30:
Hi sbertrand-witekio,

Thank you for reporting this, I was able to reproduce this and will update the thread when this is fixed.

sbertrand-witekio, 2019-06-04:
Thanks, looking forward to the patch.

sbertrand-witekio, 2019-06-14:
Do you have any update ?

pvyawaha, 2019-06-19:
Hi sbertrand-witekio,

When we receive a duplicate block it is not considered progress and updating the job status is not required. We have added a filter to ignore the data blocks that are duplicate when sending the job status updates. This will reduce the number of MQTT messages sent for job status updates when performing OTA in congested networks. This fix is merged in mainline https://github.com/aws/amazon-freertos/commit/8be8e756d7f9d44a67a482f208b7a39c745d1a6e

pvyawaha, 2019-06-24:
Hi sbertrand-witekio, 

I am closing this ticket now as the fix is merged in mainline. Thank you again for pointing this out.

closed 2019-06-24 by pvyawaha