aws/amazon-freertos #691: Adding code to the STM32

escanzanocloudcath, 2019-05-30:
I have the demos for the STM32 Discovery working with the MQTT and Shadow libraries.

I now want to add code to the project which are custom routines to use MQTT and the Shadow.

I am using the recommended SystemWorkbench IDE and v1.4.8 of the RTOS attached to my Mac.

There are many virtual directories and linked files so I am wondering if you have a recommendation as to where I should add my own custom code. 

Thanks

mradulan, 2019-05-30:
Is it a custom demo that you are trying to add? If so, then demos will be a good place to add.

WRansohoff, 2019-06-11:
To clarify, the `main.c` file in each demo application is probably a good starting point for adding your own code to the projects. The other virtual directories that show up in the IDE are dependencies and libraries which you will need to include if you want to make a new project that uses the Amazon FreeRTOS connectivity libraries.

Are there any more specific points that we can clarify for you about how the demo projects and directories are structured?

sarenameas, 2019-07-08:
Hello @escanzanocloudcath 
Do you still need direction? Would you like to add demos in the same way we have added them into the project? I will be closing this for now. Please feel free to open if you have any updates.

closed 2019-07-08 by sarenameas