aws/amazon-freertos #727: [request] ESP32 esp-idf branch update

chegewara, 2019-06-06:
Hi,
is there plan to switch esp-idf to more recent release (release/3.1 at least). 
I would like to use esp_http_client component and it is incompatible with current tcpip_adapter component used in AWSFreeRTOS (bunch of redefinition, etc).
When i replace tcpip_adapter then smartconfig_ack is complaining.

WRansohoff, 2019-06-06:
Hello,

We currently use a tailored version of ESP-IDF, and while we have been accepting updates from Espressif on those core components, we have not yet been able to bring in some of the framework's newer components and features. Unfortunately, it looks like the HTTP client/server components are among those which we don't yet support.

I did bring this issue up internally; thank you for helping us prioritize our ongoing work. We are investigating ways to keep our dependencies more up-to-date across all of our supported platforms, but I'm sorry for the inconvenience in the meantime. Feel free to let us know if you have any further questions.

chegewara, 2019-06-06:
Hi,
basically what i need is option to call http(s) calls from IoT thing. My first thought is esp_http_client because i know it, but maybe there is some option in AWSFreeRTOS that i didnt find yet. Any help is appreciated, because its ongoing project.

WRansohoff, 2019-06-10:
Unfortunately, I don't think we have a built-in solution for HTTP functionality yet. HTTP[S] client connections are on our roadmap, but we don't have a timeline for those features yet.

When you say that the `esp_http_client` component is incompatible with the `tcpip_adapter` component that we use in our demos, do you mean that you get build errors when you copy the HTTP client component from a recent version of ESP-IDF into `lib/third_party/mcu_vendors/espressif/esp-idf/components/`? If so, could you share a log of some of those errors?

closed 2019-06-28 by chegewara