aws/amazon-freertos #734: Amazon FREERTOS sdk_sync branch sntp linker error

shijinjose, 2019-06-07:
**Describe the bug**
Am trying to enable the sntp  with afr and made the following changes to the component.mk file

1.) in COMPONENT_SRCDIRS := 
      
        $(AMAZON_FREERTOS_LIB_DIR)/third_party/lwip/src/api \
        $(AMAZON_FREERTOS_LIB_DIR)/third_party/lwip/src/apps/sntp \
        $(AMAZON_FREERTOS_LIB_DIR)/third_party/lwip/src/core \
        $(AMAZON_FREERTOS_LIB_DIR)/third_party/lwip/src/core/ipv4 \
        $(AMAZON_FREERTOS_LIB_DIR)/third_party/lwip/src/core/ipv6 \
        $(AMAZON_FREERTOS_LIB_DIR)/third_party/lwip/src/portable \
        $(AMAZON_FREERTOS_LIB_DIR)/third_party/lwip/src/netif \
        $(AMAZON_FREERTOS_LIB_DIR)/third_party/lwip/test/fuzz

2.) in COMPONENT_ADD_INCLUDEDIRS := 
                             $(AMAZON_FREERTOS_LIB_DIR)/third_party/lwip/src/include \
                             $(AMAZON_FREERTOS_LIB_DIR)/third_party/lwip/src/portable \
                             $(AMAZON_FREERTOS_LIB_DIR)/third_party/lwip/test/fuzz \
                             $(AMAZON_FREERTOS_LIB_DIR)/third_party/lwip/src/include/lwip

3.) CFLAGS += -Wno-address  # lots of LWIP source files evaluate macros that check address of stack variables
the copilation was successfull but showing following error in linking

changed sntp_opts.h file as follows
#if !defined SNTP_SERVER_DNS || defined __DOXYGEN__
#define SNTP_SERVER_DNS            1
#endif

LD build/afr_sspc.elf
/home/shijin/afr_test/afr/sspc/make/build/tcpip_adapter/libtcpip_adapter.a(tcpip_adapter.o):(.rodata.ethbroadcast+0x0): multiple definition of `ethbroadcast'
/home/shijin/afr_test/afr/sspc/make/build/amazon-freertos-common/libamazon-freertos-common.a(ethernet.o):(.rodata.ethbroadcast+0x0): first defined here
/home/shijin/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/shijin/afr_test/afr/sspc/make/build/afr_sspc.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/home/pentair/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 209784 bytes
/home/shijin/afr_test/afr/sspc/make/build/amazon-freertos-common/libamazon-freertos-common.a(timeouts.o):(.literal.sys_timeout+0x14): undefined reference to `sys_now'
/home/shijin/afr_test/afr/sspc/make/build/amazon-freertos-common/libamazon-freertos-common.a(timeouts.o): In function `sys_timeout':
/home/shijin/afr_test/afr/lib/third_party/lwip/src/core/timeouts.c:171: undefined reference to `sys_now'
collect2: error: ld returned 1 exit status
/home/shijin/afr_test/afr/sspc/make/../../lib/third_party/mcu_vendor/espressif/esp-idf/make/project.mk:406: recipe for target '/home/shijin/afr_test/afr/sspc/make/build/afr_sspc.elf' failed
make: *** [/home/shijin/afr_test/afr/sspc/make/build/afr_sspc.elf] Error 1

**System information**
- Hardware board : esp32
- afr freertos sdk_sync
- Operating System [|Linux|]
- Version of Amazon FreeRTOS (sdk_sync branch)
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]

**Additional context**
I have changed the component.mk file as per esp-idf version of sntp example.

It would be really great if anyone help with how to enable the sntp support for afr

Thank you!

chegewara, 2019-06-07:
```
/home/shijin/afr_test/afr/sspc/make/build/tcpip_adapter/libtcpip_adapter.a(tcpip_adapter.o):(.rodata.ethbroadcast+0x0): multiple definition of ethbroadcast' 
```
`section .dram0.bss will not fit in region dram0_0_seg`

`region dram0_0_seg overflowed by 209784 bytes`

WRansohoff, 2019-06-07:
It looks like your application is building the ESP-IDF lwip component; we use FreeRTOS+TCP as the TCP/IP stack in our ESP32 demos, so that might be causing some some of the build issues you're seeing such as `multiple definitions of 'ethbroadcast'` or `undefined reference to sys_now`.

The combined application might also be too large or use too much RAM; like chegewara mentioned, it looks like the linker can't fit everything it needs to into the `dram0_0_seg` memory section.

Since the ESP-IDF sntp example seems to rely on lwip for its sntp protocol, you would probably need to modify your application to use lwip instead of FreeRTOS+TCP, rather than building both of them into the project. I don't think that we have specific guidance on how to do that, though.

Sorry that I don't have a better solution; we are currently working on adding NTP functionality to the Amazon: FreeRTOS libraries, but unfortunately I don't have a timeframe for when that will be ready to use.

shijinjose, 2019-06-08:
@WRansohoff so is there any way to do the time sync in current afr repo?? May be through tls layer or mqtt..  Because my application requires to send the timestamp with the mqtt msg

qiutongs, 2019-07-11:
Sorry, we don't support time sync in current AFR repo. We are actively working on NTP and you can expect a "preview release" coming sometime in August, in a separate branch. We will let you know.

ravibhagavandas, 2019-07-22:
Hi @shijinjose,
Custom NTP application through MQTT/TLS is a feasible option. But please note it will involve some additional cost for sending MQTT messages periodically. 

I am closing this issue. Please feel free to reopen this if you still have questions.

closed 2019-07-22 by ravibhagavandas

BasemTarek, 2020-07-20:
Hello,

Is AFR still don't support time sync..?