aws/amazon-freertos #749: vTaskPrioritySet  may cause priority inversion

KiLuYa, 2019-06-08:
Try to think about this situation.

There are task1, task2, mutex1 in system.
Priority of task1 is 5, while priority of task2 is 4.
Task1 have taken mutex1, while task2 want to take mutex1.
Task2 will pend on the list of mutex1.
Priority inheritance didn't happen because task2's priority is lower than task1.
If user call vTaskSetPriority ( task1, 2 ),  then task1's priority will be lower then task2.
After vTaskSetPriority is called, if there is a task3 with priority of 3, 
then priority inversion may happen between task2 and task3.

RichardBarry, 2019-06-08:
Hi - a full priority inheritance implementation where priorities are stacked and there is a one to many relationship between mutexes and tasks can become very complex, and therefore require a lot of code that will add to the execution time to cater for scenarios that will hardly ever actually happen considering FreeRTOS is used in small MCU applications.  For example you can have a task take multiple different mutexes at the same time, causing multiple inheritances at the same time, give back some of the mutexes in a different order resulting in priorities to drop again, only to have yet more tasks try to take mutexes that are still held from yet other priorities.  Therefore we opted to implement a simplified scheme whereby a task maintains whatever was the highest priority it inherited while it held multiple mutexes until such time that is releases all the mutexes.

As I said this is rarely an issue in small simple statically linked systems, however if it does become an issue you might consider designs that don't require a mutex.  For example, having only one task that is actually allowed to access the resource, and with other tasks proxying their accesses through that one task.  This is how our TCP/IP stack works.  One task controls the stack itself, socket API calls work through queues to communicate with that task (with event groups being used to communicate events the other way).

KiLuYa, 2019-06-09:
Thanks for your explanation about mutex.
Your advice is inspiring. It sounds similar to how timer works and how lwIP works.

closed 2019-06-09 by KiLuYa