aws/amazon-freertos #751: Cannot run ESP32 Demo : aws_mqtt_agent.c:1907:25: error: implicit declaration of function 'xQueueCreateStatic'

GeorgeGkinis, 2019-06-09:
**Describe the bug**
I am trying to compile the ESP32 Demo.
The IDF_PATH has been cleared.

When I run make flash under ~/esp/afreertos/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make I get the following error :  

````
/home/osboxes/esp/afreertos/amazon-freertos/lib/mqtt/aws_mqtt_agent.c: In function 'MQTT_AGENT_Init':
/home/osboxes/esp/afreertos/amazon-freertos/lib/mqtt/aws_mqtt_agent.c:1907:25: error: implicit declaration of function 'xQueueCreateStatic' [-Werror=implicit-function-declaration]
         xCommandQueue = xQueueCreateStatic( mqttCOMMAND_QUEUE_LENGTH, sizeof( MQTTEventData_t ), ucQueueStorageArea, &xStaticQueue );
````
**System information**
- Which hardware board or part numbers?
**ESP32 Lolin D32 Pro V2**
- IDE used
**CLion, but command runs from terminal**
- Operating System [Windows|Linux|MacOS]
**Linux Ubuntu VM**
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)
**v1.4.7-214-gbc86e638**
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
**MQTT**
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
A clear and concise description of what you expected to happen.
I Excpect the demo to compile

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 

````
/home/osboxes/esp/afreertos/amazon-freertos/lib/mqtt/aws_mqtt_agent.c: In function 'MQTT_AGENT_Init':
/home/osboxes/esp/afreertos/amazon-freertos/lib/mqtt/aws_mqtt_agent.c:1907:25: error: implicit declaration of function 'xQueueCreateStatic' [-Werror=implicit-function-declaration]
         xCommandQueue = xQueueCreateStatic( mqttCOMMAND_QUEUE_LENGTH, sizeof( MQTTEventData_t ), ucQueueStorageArea, &xStaticQueue );
                         ^
/home/osboxes/esp/afreertos/amazon-freertos/lib/mqtt/aws_mqtt_agent.c:1907:23: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
         xCommandQueue = xQueueCreateStatic( mqttCOMMAND_QUEUE_LENGTH, sizeof( MQTTEventData_t ), ucQueueStorageArea, &xStaticQueue );
                       ^
/home/osboxes/esp/afreertos/amazon-freertos/lib/mqtt/aws_mqtt_agent.c:1910:27: error: implicit declaration of function 'xTaskCreateStatic' [-Werror=implicit-function-declaration]
         xMQTTTaskHandle = xTaskCreateStatic( prvMQTTTask, "MQTT", mqttconfigMQTT_TASK_STACK_DEPTH, NULL, mqttconfigMQTT_TASK_PRIORITY, xStack, &xStaticTask );
                           ^
/home/osboxes/esp/afreertos/amazon-freertos/lib/mqtt/aws_mqtt_agent.c:1910:25: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
         xMQTTTaskHandle = xTaskCreateStatic( prvMQTTTask, "MQTT", mqttconfigMQTT_TASK_STACK_DEPTH, NULL, mqttconfigMQTT_TASK_PRIORITY, xStack, &xStaticTask );
                         ^
cc1: some warnings being treated as errors
make[1]: *** [/home/osboxes/esp/afreertos/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/make/component_wrapper.mk:286: lib/mqtt/aws_mqtt_agent.o] Error 1
make: *** [/home/osboxes/esp/afreertos/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make/../../../../lib/third_party/mcu_vendor/espressif/esp-idf/make/project.mk:477: component-amazon-freertos-common-build] Error 2
````

Seems like I am missing some headers...

Can anyone help?

Thanks!

WRansohoff, 2019-06-10:
Hi there, sorry to hear that you're having trouble. Can you double-check that the ESP-IDF FreeRTOS component is configured to enable static allocation?

Run `make menuconfig` in the `demos/espressif/esp32_devkitc_esp_wrover_kit/make` directory, navigate to `Component Config -> FreeRTOS`, and verify that the `Enable FreeRTOS static allocation API` option is enabled.

ravitomar7, 2019-06-15:
I have enabled it, still same issue.

Users/ravitomar/Documents/esp/amazon-freertos/lib/mqtt/aws_mqtt_agent.c:1907:25: error: implicit declaration of function 'xQueueCreateStatic'; did you mean 'xQueueCreateSet'? [-Werror=implicit-function-declaration]
         xCommandQueue = xQueueCreateStatic( mqttCOMMAND_QUEUE_LENGTH, sizeof( MQTTEventData_t ), ucQueueStorageArea, &xStaticQueue );
                         ^~~~~~~~~~~~~~~~~~
                         xQueueCreateSet
/Users/ravitomar/Documents/esp/amazon-freertos/lib/mqtt/aws_mqtt_agent.c:1907:23: warning: assignment to 'QueueHandle_t' {aka 'struct QueueDefinition *'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
         xCommandQueue = xQueueCreateStatic( mqttCOMMAND_QUEUE_LENGTH, sizeof( MQTTEventData_t ), ucQueueStorageArea, &xStaticQueue );
                       ^
/Users/ravitomar/Documents/esp/amazon-freertos/lib/mqtt/aws_mqtt_agent.c:1910:27: error: implicit declaration of function 'xTaskCreateStatic'; did you mean 'xTaskCreate'? [-Werror=implicit-function-declaration]
         xMQTTTaskHandle = xTaskCreateStatic( prvMQTTTask, "MQTT", mqttconfigMQTT_TASK_STACK_DEPTH, NULL, mqttconfigMQTT_TASK_PRIORITY, xStack, &xStaticTask );
                           ^~~~~~~~~~~~~~~~~
                           xTaskCreate

ravitomar7, 2019-06-15:
> Hi there, sorry to hear that you're having trouble. Can you double-check that the ESP-IDF FreeRTOS component is configured to enable static allocation?
>
> Run `make menuconfig` in the [...]

I tried enabling it, but it doesn't gets enabled..

if i try sudo then everything changes..

PLs help

monkeytronics, 2019-06-18:
@WRansohoff By my reckoning, you are 100% right. I had the exact same problem, and your suggestion fixed it. Clearly, @ravitomar7 is experiencing a separate issue. Awesome work. Thanks. 

[image]

dryprogrammer, 2019-06-20:
@ravitomar7 I have the same problem - Enable FreeRTOS static allocation API cannot be enabled.
**macOS Mojave 10.14.5**
**aws/amazon-freertos @v1.4.8**

I enabled it in make menuconfig, but GENCONFIG seems to be rewriting my sdkconfig. I tried manually setting CONFIG_FREERTOS_SUPPORT_STATIC_ALLOCATION=y and CONFIG_SUPPORT_STATIC_ALLOCATION=y but the keep getting overwritten.

I switched to Ubuntu VM and the CONFIG_SUPPORT_STATIC_ALLOCATION menuconfig problem went away but then ran into other problems so came back to mac.

make flash output:
```
(base) davidson1@mbp:make$ make flash
Toolchain path: /Users/davidson1/dev/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/davidson1/dev/esp/esp-idf/requirements.txt are satisfied.
GENCONFIG
/Users/davidson1/dev/esp/aws/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make/sdkconfig:494 CONFIG_SUPPORT_ST
ATIC_ALLOCATION was replaced with CONFIG_FREERTOS_SUPPORT_STATIC_ALLOCATION
/Users/davidson1/dev/esp/aws/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make/sdkconfig:495 CONFIG_TIMER_TASK
_PRIORITY was replaced with CONFIG_FREERTOS_TIMER_TASK_PRIORITY
/Users/davidson1/dev/esp/aws/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make/sdkconfig:496 CONFIG_TIMER_TASK
_STACK_DEPTH was replaced with CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH
/Users/davidson1/dev/esp/aws/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make/sdkconfig:497 CONFIG_TIMER_QUEU
E_LENGTH was replaced with CONFIG_FREERTOS_TIMER_QUEUE_LENGTH
warning: the value 'FREERTOS_NO_AFFINITY' is invalid for LWIP_TCPIP_TASK_AFFINITY, which has type hex
App "aws_demos" version: v1.4.7-219-g0c2f0684-dirty
CC build/bootloader/bootloader_support/src/bootloader_clock.o
```

ravitomar7, 2019-06-20:
@dryprogrammer just restart your terminal, it will work.. mine worked in this case

Mac mojave

dryprogrammer, 2019-06-20:
@ravibhagavandas 
Can you tell me which commit you are using? It seems the Amazon guide has been updated yesterday to use CMake, which doensn't work with release v1.4.8.

Did you need to update to the GNU versions of the flex, bison, and gcc? Or did you copy in the compiled zconf.lex.c from https://github.com/espressif/esp-idf/issues/1281#issuecomment-345216108 ?

I had too many issues with Ubuntu 18.04 #810 so I'm trying macOS again and your reply gives me hope.

After rebooting I'm facing my old issues again, namely use of undeclared identifier 'PATH_MAX', so I will try copying in the binary again:
```
(base) davidson1@mbp:make$ CFLAGS=-w make flash
cc -c  -DCURSES_LOC="<curses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD  /Users/davidson1/dev/esp/aws/amazon-freertos/lib/third_party/mcu_vendor/
espressif/esp-idf/tools/kconfig/zconf.tab.c -o zconf.tab.o
In file included from /Users/davidson1/dev/esp/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/tools/kconfig/zconf.tab.c:2332:
/Users/davidson1/dev/esp/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/tools/kconfig/confdata.c:113:23: error: use of undeclared
      identifier 'PATH_MAX'
        static char fullname[PATH_MAX+1];
```

Trying a bit more before I resort to Mongoose OS or Zerynth.

dryprogrammer, 2019-06-20:
Ok, release-1.5 build seems to be more predictable on macOS Mojave 10.14.5
`cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -S . -B ./demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/espressif_code/freertos`

However, the `./vendors/espressif/esp-idf/tools/idf.py` is going to the wrong serial port, so trying to find out how to specify it.

@ravibhagavandas I guess you were able to deploy v1.4.8, so if you could share your setup that worked, I'd be very grateful, because I have to demo this solution soon.

I switched over from the Google Cloud IoT Device SDK because I thought the Amazon FreeRTOS would be quicker to run with. Google SDK was very stable and predictable but didn't have full end-to-end examples. Arduino has been the easiest so far.

dryprogrammer, 2019-06-20:
idf.py port can be specified like `-p /dev/ttyUSB1` as shown in the documentation

dryprogrammer, 2019-06-20:
I was able to successfully build with release-1.5 but it lead to _Brownout detector was triggered_ error, but swapping the USB cable fixed the issue
`./vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p /dev/tty.SLAB_USBtoUART -B ./demos/espressif/esp32_devkitc_esp_wrover_kit/make/build/freertos/demos/mqtt`

Hopefully somebody can share the environment in which they were able to build v1.4.8 as I do have mac, Windows, and Ubuntu ready to use. Thanks.

More info on Brownout detector:
https://www.esp32.com/viewtopic.php?t=2462
https://github.com/nkolban/esp32-snippets/issues/168

```
Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6276
load:0x40078000,len:11276
load:0x40080000,len:6084
0x40080000: _WindowOverflow4 at /Users/davidson1/dev/esp/aws/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1685

entry 0x4008032c
0x4008032c: _KernelExceptionVector at ??:?

I (29) boot: ESP-IDF v3.1.3-25-g36733db8d 2nd stage bootloader
I (29) boot: compile time 00:48:45
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (65) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (73) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (80) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (88) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (95) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (103) boot: End of partition table
I (107) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x37da4 (228772) map
I (196) esp_image: segment 1: paddr=0x00057dcc vaddr=0x3ffc0000 size=0x03008 ( 12296) load
I (201) esp_image: segment 2: paddr=0x0005addc vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/davidson1/dev/esp/aws/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1685

I (203) esp_image: segment 3: paddr=0x0005b1e4 vaddr=0x40080400 size=0x04e2c ( 20012) load
I (220) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xe9f04 (958212) map
0x400d0018: _flash_cache_start at ??:?

I (556) esp_image: segment 5: paddr=0x00149f24 vaddr=0x4008522c size=0x11700 ( 71424) load
0x4008522c: phy_dis_hw_set_freq at ??:?

I (599) boot: Loaded app from partition at offset 0x20000
I (599) boot: ota rollback check done
I (599) boot: Disabling RNG early entropy source...
I (604) cpu_start: Pro cpu up.
I (608) cpu_start: Single core mode
I (612) heap_init: Initializing. RAM available for dynamic allocation:
I (619) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (625) heap_init: At 3FFD26A0 len 0000D960 (54 KiB): DRAM
I (631) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (637) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (644) heap_init: At 4009692C len 000096D4 (37 KiB): IRAM
I (650) cpu_start: Pro cpu start user code
I (332) cpu_start: Starting scheduler on PRO CPU.
I (359) uart: queue free spaces: 10
0 2 [main] Write certificate...
I (369) PKCS11: Initializing NVS partition: "storage"
1 4 [main] Write device private key...
I (639) BTDM_INIT: BT controller compile version [3828e65]

I (639) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
W (649) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration

Brownout detector was triggered
```

sarenameas, 2019-07-03:
Hello, it is great that you were able to build and flash the ESP32 code with Amazon FreeRTOS. Running the MQTT demo with no other configurations outside of the instructions in the getting started guide (https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html) I do not run into this brownout error. The links you gave above show that it is a hardware error, rather than a software one. Are there any other issues that you need resolved regarding building the project? Otherwise I am going to close this issue in a few days.

closed 2019-07-06 by sarenameas