aws/amazon-freertos #752: MQTT demo with Amazon FreeRTOS and Android as Mqtt proxy

Amijic, 2019-06-09:
Greetings,

Currently I am porting Amazon FreeRTOS on my nRF52840 DK and trying to connect to the cloud with mqtt proxy on my android device. After configuring the demo and downloading the android app on my phone, I couldnt find some of the constants that needed to be defined. I could not find amazon-freertos-ble-android-sdk/app/src/main/java/com/amazon/aws/freertosandroid/MainActivity.java.

In the end I configured the json file and DemoConstants.java in wich I wrote:
AWS_IOT_POLICY_NAME:
AWS_IOT_REGION:

In json file I configured:
Under CognitoIdentity, redefine the following variables:
PoolId: Your Amazon Cognito identity pool ID.
Region: Your AWS Region.

Under CognitoUserPool, redefine the following variables:
PoolId: Your Amazon Cognito user pool ID.
AppClientId: Your app client ID.
AppClientSecret: Your app client secret.
Region: Your AWS Region.

Then I installed the app on my android device and started a BLE demo on my board.
First problem is related to my android app, I can not click "More.." option on my device.

The RAM must start at 200046f8, it's size must be 3b908
Common libraries successfully initialized.
No networks connected. Waiting for a network connection. 
2 989 https://forums.aws.amazon.com/ Requesting to update ATT MTU to 247 bytes on connection 0x0.3 989 https://forums.aws.amazon.com/ sd_ble_gattc_exchange_mtu_request() on connection 0x0 returned busy, will retry.4 989 https://forums.aws.amazon.com/ BLE Connected to remote device, connId = 0

Successfully initialized the demo. Network type for the demo6 989 https://forums.aws.amazon.com/ https://forums.aws.amazon.com/https://forums.aws.amazon.com/https://forums.aws.amazon.com/ MQTT library successfully initialized.
MQTT demo client identifier is XXXXXX (length X).
Numeric comparison:"some numbers"
Press 'y' to confirm
UNKNOWN BLE EVENT 35

Is it possible that there are some steps I missed that are not written in https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_nordic.html.

Best regards :)

WRansohoff, 2019-06-10:
Does the `UNKNOWN BLE EVENT 35` error happen after you press 'y' to confirm the pairing?

One possible cause of those errors seems to be that mobile devices often cache Bluetooth connection data, and that cached data can sometimes become stale or unreliable. In your phone's Bluetooth settings, can you see the nRF52840 board under the list of paired devices? If it does show up, could you try deleting the connection, restarting the phone, and then re-pairing the devices?

Amijic, 2019-06-11:
Hello,

Thank you for answering. 

The UNKNOWN BLE EVENT 35 error shows up before I press 'y' to confirm pairing. It doesnt give me a chance to press 'y' it immediately throws an error. In my mobile app I can detect my nRF52840 but it wont connect. However, is it possible that my phone cant accses the board because i didnt put MAC adress of a device and I dont have MainActivity.java. When I downloaded Android SDK for Amazon FreeRTOS the MainActivity.java file was not there. 
Nevertheless, I configured all other parameters like Amazon Cognito, AWS IoT policy, region in my JSON file and DemoConstants.java.  

Best regards :)

bxpan, 2019-06-14:
Hi Amijic,

There is no MainActivity.java in the demo app. The documentation is out-of-date. You need to modify DemoConstants.java and the awsconfiguration.json file, which you did. The fact that "more" button is not enabled suggests the connection is not fully ready. Could you please provide following information:

Build info
What is the branch and commit on the device side?
What is the branch and tag the SDK side?
Does pulling the latest code help resolve the issue?

System info
What is the Android version?
What is the bluetooth version?

To Reproduce
What is the frequency of this issue?
Steps to reproduce the behavior:

Log
Could you provide log on both the device side and SDK side?

Additional context
Add any other context about the problem here.

Amijic, 2019-06-14:
Hello,

Build info:
Regarding my device I have last version of Amazon FreeRTOS, specificly commit  4fa09aa  on 18 Apr. I downloaded the latest version of SDK commit b073707  23 hours ago. 

Again it shows the same thing. It does not let me press 'y' on my segger terminal connection immediately chrashes

System info:
Regarding my system info, my phone has Bluetooth 5 and it is Android 9 version.

To reproduce:

I followed all the steps in Getting started tutorial on nRF52840 DK. Firstly I set my AWS IoT thing and policy. After that I got my Amazon Cognito identity and user pool. 
In next step I downloaded amazon freertos and configured the demo for simple hello with BLE and MQTT. Afterwards I downloaded the SDK and installed it on my mobile device. In Android Studio I configured JSON file and DemoConstants.java as it was said in the tutorial. Then I powered up both the board and the app and UNKNOW BLE EVENT 35 error occurs.

Log: 

[image]

This is my output in Segger. And in my app I just get a pairing request from the board and when I click OK it just disconnects. 

Additional context:

Do you think that missing MAC adress can be the problem. In the tutorial for this demo it says that it is needed to put MAC adress of a device in MainActivity.java.

This is my IAM policy;
{
   "Version":"2012-10-17",
   "Statement":[
      {
         "Effect":"Allow",
         "Action":[
            "iot:AttachPolicy",
            "iot:AttachPrincipalPolicy",
            "iot:Connect",
            "iot:Publish",
            "iot:Subscribe",
            "iot:Receive",
            "iot:GetThingShadow",
            "iot:UpdateThingShadow",
            "iot:DeleteThingShadow"
         ],
         "Resource":[
            "*"
         ]
      }
   ]
}

It is the same as my AWS IoT policy in my AWS console.

Considering that I used mobile device as proxy I did not create certificate for my AWS IoT thing.
In my Amazon FreeRTOS I configured the demo by uncommenting certain functions and hardcoding my endpoint and my IoT thing.

Kind regards :)

bxpan, 2019-06-14:
Cloud you please share adb log from android studio? 

You do not need to to add mac address in the demo app as this step is no longer needed. When you open the demo app, you should see the mac address of the device, and you can toggle the switch to connect to it. To debug why the "More" button is not enabled, please share the adb log from your demo app.

Amijic, 2019-06-15:
Hello,

This is output of my info log on ADB is

2019-06-15 15:42:08.674 31366-31366/software.amazon.freertos.demo I/DeviceScanFragment: scan button clicked.
2019-06-15 15:42:08.677 31366-31366/software.amazon.freertos.demo I/AmazonFreeRTOSManager: Starting ble device scan
2019-06-15 15:42:08.891 31366-31366/software.amazon.freertos.demo I/chatty: uid=10239(software.amazon.freertos.demo) identical 1 line
2019-06-15 15:42:09.772 31366-31366/software.amazon.freertos.demo I/DeviceScanFragment: connect switch isChecked: ON
2019-06-15 15:42:09.907 31366-31382/software.amazon.freertos.demo I/FRD: BLE connection state changed: 0; new state: BLE_CONNECTED
2019-06-15 15:42:09.909 31366-31382/software.amazon.freertos.demo I/FRD: Connected to GATT server.
2019-06-15 15:42:09.913 31366-31382/software.amazon.freertos.demo I/BluetoothDevice: createBond() for device F4:B6:6C:58:32:F1 called by pid: 31366 tid: 31382
2019-06-15 15:42:10.891 31366-31382/software.amazon.freertos.demo I/FRD: Discovered Ble gatt services successfully. Bonding state: 11
2019-06-15 15:42:18.689 31366-31464/software.amazon.freertos.demo I/AmazonFreeRTOSManager: Stopping ble device scan
2019-06-15 15:42:40.488 31366-31382/software.amazon.freertos.demo I/FRD: BLE connection state changed: 19; new state: BLE_DISCONNECTED
2019-06-15 15:42:40.488 31366-31382/software.amazon.freertos.demo I/FRD: Disconnected from GATT server.
2019-06-15 15:42:40.488 31366-31382/software.amazon.freertos.demo I/DeviceScanFragment: BLE connection status changed to: BLE_DISCONNECTED
2019-06-15 15:42:40.497 31366-31366/software.amazon.freertos.demo I/DeviceScanFragment: connect switch isChecked: OFF
2019-06-15 15:42:42.188 31366-31366/software.amazon.freertos.demo I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus
2019-06-15 15:42:46.263 31366-31366/software.amazon.freertos.demo I/InputMethodManager: startInputInner - mService.startInputOrWindowGainedFocus

bxpan, 2019-06-17:
There seems to be some issue during BLE bonding. It is not fully paired and bonded, and it got disconnected. Could you share the BLE configuration below on the device side: 

#define IOT_BLE_ENABLE_NUMERIC_COMPARISON        ( 1 )
#define IOT_BLE_ENABLE_SECURE_CONNECTION         ( 1 )
#define IOT_BLE_INPUT_OUTPUT (eBTIODisplayYesNo)

> 2019-06-15 15:42:10.891 31366-31382/software.amazon.freertos.demo I/FRD: Discovered Ble gatt services successfully. Bonding state: 11
2019-06-15 15:42:18.689 31366-31464/software.amazon.freertos.demo I/AmazonFreeRTOSManager: Stopping ble device scan
2019-06-15 15:42:40.488 31366-31382/software.amazon.freertos.demo I/FRD: BLE connection state changed: 19; new state: BLE_DISCONNECTED

Amijic, 2019-06-17:
#ifndef IOT_BLE_ENABLE_NUMERIC_COMPARISON
#define IOT_BLE_ENABLE_NUMERIC_COMPARISON        ( 1 )
#endif

#ifndef IOT_BLE_ENABLE_SECURE_CONNECTION
#define IOT_BLE_ENABLE_SECURE_CONNECTION         ( 1 )
#endif

#ifndef IOT_BLE_INPUT_OUTPUT
#define IOT_BLE_INPUT_OUTPUT eBTIODisplayYesNo
#endif

This is what I got in my iot_ble_config_defaults.h file.

bxpan, 2019-06-18:
These look correct .

> #ifndef IOT_BLE_ENABLE_NUMERIC_COMPARISON
#define IOT_BLE_ENABLE_NUMERIC_COMPARISON ( 1 )
#endif
#ifndef IOT_BLE_ENABLE_SECURE_CONNECTION
#define IOT_BLE_ENABLE_SECURE_CONNECTION ( 1 )
#endif
#ifndef IOT_BLE_INPUT_OUTPUT
#define IOT_BLE_INPUT_OUTPUT eBTIODisplayYesNo
#endif

Error 19 indicates that it was disconnected from the device side. 

> 2019-06-15 15:42:40.488 31366-31382/software.amazon.freertos.demo I/FRD: BLE connection state changed: 19; new state: BLE_DISCONNECTED

Amijic, 2019-06-18:
Can you give me few tips on where the mistake could be? 

I did not hardcode my credidentials in my device code because it says in the tutorial that it is not needed when I am using mobile device as MQTT proxy. 

I enabled the demo by uncommenting and entered my endpoint and IoT thing name. I really dont know what to try anymore :D

bxpan, 2019-06-18:
Did you see the pair notification from your app? Were you able to press the pair button successfully? 

Another option to try is that without using the demo app, go to the Bluetooth settings on your phone, and see if the device shows up and try to connect and pair to the device. Please let us know if this works?

Amijic, 2019-06-26:
Hello,

I followed you instruction and my phone does see the device using Bluetooth settings. When I tried to pair ( I can press the pair button successfully) same error occurs UNKNOW BLE EVENT 35.

[image]

This is my Segger terminal output.

huguesBouvier, 2019-06-27:
Hello,

The UNKNOW BLE EVENT 35 is an un-handled message. It is not harmful.
We are working on supressing it cleaning but it won't give you any problem.

bxpan, 2019-06-27:
Just to clarify, "The UNKNOW BLE EVENT 35" is not the cause of the disconnection. This is simply an unhandled message type. There's some other issues that caused the disconnection. We still need to root cause this.

Amijic, 2019-06-28:
Hello,

Not much is changed on my device code. I just uncommented the demo I wanted to use and hardcoded my endopoint and my AWS IoT Thing name. 
Afterwards, I defined #define IOT_BLE_ADD_CUSTOM_SERVICES ( 1 ) in my iot_ble_config.h file

I did not find this file in my path.

[image]

This is my file tree. 

[image]

Is it possible that lack of this definition can cause the device to disconnect?

bxpan, 2019-06-28:
> Is it possible that the region has anything to do with connection? I am in Europe and my AWS default region is US West(Oregon).

This is not an issue. The disconnection is between the device and the phone over BLE, it is not disconnected between phone and the cloud.

huguesBouvier, 2019-06-28:
Could you repport the error statuses? It should be displayed at this line:
https://github.com/aws/amazon-freertos/blob/b76006df130102971b5f3575b7a85356c5439394/vendors/nordic/boards/nrf52840-dk/ports/ble/aws_ble_hal_common_gap.c#L1556
There another possibility, we support only BLE 4.2 and above. Could you check the version of the BT stack of your phone?

Amijic, 2019-06-30:
The RAM must start at 200046f8, it's size must be 3b908
0 0 [iot_thread] [INFO ][COMMON][0] Common libraries successfully initialized.
1 2 [iot_thread] [INFO ][DEMO][20] No networks connected. Waiting for a network connection. 
2 2821 [BLE] Requesting to update ATT MTU to 247 bytes on connection 0x0.3 2821 [BLE] sd_ble_gattc_exchange_mtu_request() on connection 0x0 returned busy, will retry.4 2821 [BLE] [INFO ][DEMO][28210] BLE Connected to remote device, connId = 0

5 2821 [iot_thread] [INFO ][DEMO][28210] Successfully initialized the demo. Network type for the de6 2821 [iot_thread] [INFO ][MQTT][28210] MQTT library successfully initialized.
7 2821 [iot_thread] [INFO ][DEMO][28210] MQTT demo client identifier is BLEthing (length 8).
8 2857 [InputTask] Numeric comparison:388263
9 2857 [InputTask] Press 'y' to confirm
10 2903 [BLE] UNKNOWN BLE EVENT 35
11 5854 [BLE] Failed to secure connection. Disconnecting.12 5854 [BLE] BLE_GAP_EVT_AUTH_STATUS
13 5855 [BLE] [INFO ][DEMO][58550] BLE disconnected with remote device, connId = 0 

14 5855 [BLE] Disconnect received for MQTT service instance 0

This is my Debugg Terminal.

huguesBouvier, 2019-07-01:
You don't need to download the C SDK.
The parameter is located here: https://github.com/aws/amazon-freertos/blob/6144acbe61db1928a9a925f00fdf85ec239e012b/vendors/nordic/boards/nrf52840-dk/aws_demos/config_files/aws_demo_config.h#L40
Could you report the detailed error status at the following location?
Could you repport the error statuses? It should be displayed at this line:
https://github.com/aws/amazon-freertos/blob/b76006df130102971b5f3575b7a85356c5439394/vendors/nordic/boards/nrf52840-dk/ports/ble/aws_ble_hal_common_gap.c#L1556
Thanks!

Amijic, 2019-07-01:
Can you tell me what do you mean by report the error statuses? Isnt that just the debug terminal messages?

huguesBouvier, 2019-07-01:
No, that NRF debug log is not reported. I can't read the string "PM_EVT_CONN_SEC_FAILED src:" from your terminal

Amijic, 2019-07-07:
Greetings, 

My build wont run on my board when I enable NRF_LOG in my sdk_config.h
[image]

When I try to debugg it just stops in unknow function.

huguesBouvier, 2019-07-08:
I have had issues with step by step too. But you could:
1.  Add a breakpoint there and check the statuses (do not step inside, just stop the program where you want)
2. Use FreeRTOS configPRINTF to output the message. Make sure that the totality of the message is displayed.

closed 2019-08-16 by dcgaws