aws/amazon-freertos #757: FreeRTOS+TCP improve FreeRTOSIPConfigDefaults

hs2gh, 2019-06-11:
**Allow making use of the default MTU size in FreeRTOSIPConfigDefaults.h**  
Making use of the default MTU size defined in FreeRTOSIPConfigDefaults.h currently errors out due to premature min. size check before defining the default.

**Describe the solution you would like.** 
Move the min. size check down to the default section e.g. like this:
```
#ifndef ipconfigNETWORK_MTU
	#define ipconfigNETWORK_MTU		1500
#elif ( ipconfigNETWORK_MTU < 46 )
	#error ipconfigNETWORK_MTU must be at least 46.
#endif
```
Thank you!

htibosch, 2019-06-11:
Hi, and thanks for your suggestion.

You are proposing a minimum value of 46 bytes. Why 46? With that size of MTU it will not be possible to use the UDP or TCP protocol.
At the other hand, I would not know any other MTU size that does make sense. A hundred bytes at least, I would say.

hs2gh, 2019-06-11:
The minimum value is just the one in FreeRTOSIPConfigDefaults.h in the repo. Not my idea ;)
To be honest I don‘t really care about the minimum b/c I‘ll stick to the most common std. size 1500.

aggarg, 2019-06-12:
@htibosch - The check ```#if ( ipconfigNETWORK_MTU < 46 )``` is before the default value is defined. As a result, ```ipconfigNETWORK_MTU``` needs to be defined in ```FreeRTOSIPConfig.h```. Moving the check afterwards will solve this problem. 

Does this look good to you?

Thanks.

htibosch, 2019-06-13:
Syntactically, this code looks OK to me, and needs no change:

	#ifndef ipconfigNETWORK_MTU
		#define ipconfigNETWORK_MTU		1500
	#elif ( ipconfigNETWORK_MTU < 46 )
		#error ipconfigNETWORK_MTU must be at least 46.
	#endif

The comparison `( ipconfigNETWORK_MTU < 46 )` is only evaluated, in case the macro has been defined earlier ( in `FreeRTOSIPConfig.h` ).

Another way to write it would be:

	#ifndef ipconfigNETWORK_MTU
		#define ipconfigNETWORK_MTU		1500
	#else
		/* Test if the user-provided value of MUT is at least 46. */
		#if ( ipconfigNETWORK_MTU < 46 )
			#error ipconfigNETWORK_MTU must be at least 46.
		#endif
	#endif

So yes, I'm OK with this extra test.

hs2gh, 2019-06-13:
@htibosch It's nitpicking, but I'm not a big fan of redundancy but prefer compact code.
So I'd propose to omit the comment more or less repeating the error text below regarding the 46 byte limit.
Just my final 2ct :)

htibosch, 2019-06-13:
It was not my intention to add a redundant comment, I wanted to make sure that Gaurav and I were talking about the same thing. His comment left me a bit puzzled

aggarg, 2019-06-13:
@htibosch  Apologies. I just wanted to clarify that you agree with the change. It looks okay to me as well.

Thanks.

closed 2019-08-16 by dcgaws