aws/amazon-freertos #770: About Object delete for FreeRTOS kernal

KiLuYa, 2019-06-12:
QueueDelete only free the resource of Queue_t.
EventGroupDelete will ready all the tasks on its list, then free the resource of EventGroup_t.
Should the delete functions' behavior be uniformed?
Why are they designed so different?

aggarg, 2019-06-17:
You are right that behavior of Queue delete and Event Group delete is a bit different and we unblock all the blocked tasks before deleting event group. However, an application developer must ensure that no task is blocked before deleting a Queue or an Event Group because the blocked tasks which get unblocked will have no way to determine that the object (Queue or Event Group) has been deleted.

Thank you for pointing it out.

sarenameas, 2019-07-08:
If there is nothing else to add I will be closing this issue.

closed 2019-07-08 by sarenameas