aws/amazon-freertos #784: generateDeviceIdentifierAndMetrics - assert failed!

chegewara, 2019-06-14:
Hi, 
in release-1.5 branch function `generateDeviceIdentifierAndMetrics` requires certificate and key to be embedded in code, which completely brakes production system when i want to flash certificate and key in NVS partition.

```
F:/aws-freertos-1.5/demos/demo_runner/iot_demo_afr.c:596 (generateDeviceIdentifierAndMetrics)- assert failed!
abort() was called a0 43 [iot_thread] ERROR: Generatingt PC 0x400e2e83 on core 0
0x400e2e83: generateDeviceIdentifierAndMetrics at F:/aws-freertos-1.5/demos/demo_runner/iot_demo_afr.c:620

Backtrace: 0x40094370:0x3ffdb990 0x4009453f:0x3ffdb9b0 0x400e2e83:0x3ffdb9d0 0x400e2e8b:0x3ffdb9f0 0x400e2e95:0x3ffdba10 0x400ec901:0x3ffdba30
0x40094370: invoke_abort at F:/aws-freertos-1.5/vendors/espressif/esp-idf/components/esp32/panic.c:676

0x4009453f: abort at F:/aws-freertos-1.5/vendors/espressif/esp-idf/components/esp32/panic.c:676

0x400e2e83: generateDeviceIdentifierAndMetrics at F:/aws-freertos-1.5/demos/demo_runner/iot_demo_afr.c:620

0x400e2e8b: _initialize at F:/aws-freertos-1.5/demos/demo_runner/iot_demo_afr.c:620

0x400e2e95: runDemoTask at F:/aws-freertos-1.5/demos/demo_runner/iot_demo_afr.c:620

0x400ec901: _threadRoutineWrapper at F:/aws-freertos-1.5/libraries/abstractions/platform/freertos/iot_threads_afr.c:318
```

Providing some certificate is fixing that issue, but is that the point?

aggarg, 2019-06-17:
As you correctly pointed out, the demo code requires certificate and key to be embedded in the code which is the reason for assert. Given that you are flashing credentials in Flash, you do not need that assert and please feel free to update the demo code as per your requirements.

Thanks.

chegewara, 2019-06-17:
Ok, fair enough. Thanks.

Just one suggestion, could it be surrounded with:
`#if  (AWS_IOT_MQTT_ENABLE_METRICS == 1)`

closed 2019-06-17 by chegewara

aggarg, 2019-06-17:
This function populates ```deviceIdentifier``` as well which is used in places not guarded by ```#if (AWS_IOT_MQTT_ENABLE_METRICS == 1)```. Therefore it is not possible to guard this function. 

However, the part which populates ```deviceMetrics``` and the corresponding function can be guarded.

Thanks.