aws/amazon-freertos #802: compiler error

sheldon123z, 2019-06-19:
Hi, 
I saw there was a change in the CMakelist file 2 days ago. I successfully ran the demos like one week ago, however, today when I try to do the same thing, I got the error like this.

CMake Error at tools/cmake/toolchains/find_compiler.cmake:23 (message):
  Compiler not found, you can specify search path with "AFR_TOOLCHAIN_PATH".
Call Stack (most recent call first):
  tools/cmake/toolchains/xtensa-esp32.cmake:6 (afr_find_compiler)
  C:/Program Files/CMake/share/cmake-3.14/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:8 (project)

I don't know how to handle this problem, because I followed the getting started instruction exactly step-by-step.

sheldon123z, 2019-06-19:
Also, in some of the demos, the errors appeared:

CMake Error at CMakeLists.txt:3 (afr_demo_module):
  Unknown CMake command "afr_demo_module".

tgsong, 2019-06-19:
Hi, can you post the exact command that you ran? On ESP32, you need to make sure your compiler is in your PATH environment variable. Also make sure you delete the CMake build folder first if you had one since many things changed from last release, that may help with the CMake command issue.

sheldon123z, 2019-06-20:
Yes, here is the commands. I created a folder called built for the building files.
---------------------------------cmd output-------------------------------------------

-- Configuring incomplete, errors occurred!
See also "C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/demos/shadow/built/CMakeFiles/CMakeOutput.log".

c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master\demos\shadow>cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B .\built
-- Building for: Visual Studio 16 2019
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.17134.
-- The C compiler identification is MSVC 19.21.27702.2
-- The CXX compiler identification is MSVC 19.21.27702.2
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.21.27702/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.21.27702/bin/Hostx64/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.21.27702/bin/Hostx64/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.21.27702/bin/Hostx64/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:3 (afr_demo_module):
  Unknown CMake command "afr_demo_module".

-- Configuring incomplete, errors occurred!
See also "C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/demos/shadow/built/CMakeFiles/CMakeOutput.log"

sheldon123z, 2019-06-20:
Also, here is the error output when I try to cmake the root directory of the freeRTOS folder:
----------------------------------------------------------------------------------------------------------
c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master>cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B built
CMake Error at tools/cmake/toolchains/find_compiler.cmake:23 (message):
  Compiler not found, you can specify search path with "AFR_TOOLCHAIN_PATH".
Call Stack (most recent call first):
  tools/cmake/toolchains/xtensa-esp32.cmake:6 (afr_find_compiler)
  C:/Program Files (x86)/CMakex86/share/cmake-3.14/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
----------------------------------------------------------------------------------------------------------------
for me, I checked the gcc compiler's location in msys32:

shzheng@NB0013671 MINGW32 ~
$ gcc   -print-search-dirs
install: /usr/lib/gcc/i686-pc-msys/7.3.0/

and I added it to the Path, but I still got the same result

sheldon123z, 2019-06-20:
**Issue update:**
Hi, I have figured out the compiler issue, but I think there are still some issues about the CMakeList.txt file.
I tried to use the Mingw32 to cmake the demo "mqtt", then I got the the errors about the CMakeList file.

shzheng@NB0013671 MINGW32 /c/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/demos/mqtt
$ cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B built
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.17134.
CMake Error at CMakeLists.txt:5 (afr_demo_module):
  Unknown CMake command "afr_demo_module".

-- Configuring incomplete, errors occurred!
See also "C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/demos/mqtt/built/CMakeFiles/CMakeOutput.log". 
------------------------------------------issue 2------------------------------------------------------
When I tried to cmake in the root directory, the errors shows like this:shzheng@NB0013671 MINGW32 /c/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master
$ cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B built
-- Building for: Visual Studio 16 2019
-- The C compiler identification is MSVC 19.21.27702.2
-- The CXX compiler identification is MSVC 19.21.27702.2
-- The ASM compiler identification is GNU
-- Found assembler: C:/msys32/opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Found Git: C:/msys32/usr/bin/git.exe (found version "2.19.0")
-- Checking Python dependencies...
Python requirements from C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/requirements.txt are satisfied.
-- Building for target esp32
-- Component names: soc log heap freertos vfs newlib driver tcpip_adapter xtensa-debug-module app_trace partition_table app_update spi_flash mbedtls micro-ecc bootloader_support ethernet nvs_flash pthread smartconfig_ack wpa_supplicant esp32 cxx amazon-freertos-common bootloader bt console esp_adc_cal esptool_py expat wear_levelling sdmmc fatfs nghttp openssl spiffs ulp
-- Component paths: C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/soc;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/log;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/heap;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/freertos;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/vfs;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/newlib;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/driver;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/tcpip_adapter;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/xtensa-debug-module;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/app_trace;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/partition_table;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/app_update;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/spi_flash;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/mbedtls;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/micro-ecc;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/bootloader_support;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/ethernet;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/nvs_flash;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/pthread;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/smartconfig_ack;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/wpa_supplicant;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/esp32;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/cxx;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/amazon-freertos-common;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/bootloader;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/bt;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/console;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/esp_adc_cal;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/esptool_py [... cut ...]
Creating config file C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/built/sdkconfig...
CMake Error at vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:202 (message):
  Internal error, toolchain has not been set correctly by project (or an
  invalid CMakeCache.txt file has been generated somehow)
Call Stack (most recent call first):
  vendors/espressif/esp-idf/CMakeLists.txt:68 (idf_verify_environment)

-- Configuring incomplete, errors occurred!
See also "C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/built/CMakeFiles/CMakeOutput.log".

tgsong, 2019-06-20:
Looks like you're executing the CMake command from the demo directory. You need to make sure to always run the CMake command from Amazon FreeRTOS root directory because CMake needs to start the process from the `CMakeLists.txt` file under the root directory.

Another thing I noticed is the MSVC compiler is picked up instead of the Espressif GCC. This is documented in our [getting started guide](https://docs.aws.amazon.com/freertos/latest/userguide/building-cmake.html). Under the CMake commandline instructure, there's a note saying "If you are using Windows, you must specify the native build system because CMake uses Visual Studio by default."

So to fix your issue,
1. Make sure the ESP32 compiler `xtensa-esp32-elf-gcc` is in environment variable `PATH`.
2. Execute the CMake command from the Amazon FreeRTOS root directory instead of under demo.
3. Specify a build system with `-G`. For example,
```sh
cd c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-maste
# or you can use -G"MinGW Makefile" instead of Ninja if you want.
cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B built -GNinja
```

sheldon123z, 2019-06-21:
Thank you very much for you reply. If I use the -G"MinGW Makefile" does that mean I have to write a make file myself ? I am pretty much sure the `xtensa-esp32-elf-gcc` is in the PATH environment variable, and I executed the from the Amazon FreeRTOS root directory, also, I added the -G"MinGW Makefile" into the command, but it still gives me an error like this:
----------------------------------------errors-----------------------------------------------------------------
shzheng@NB0013671 MINGW32 /c/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master
$ cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B built -G"MinGW Makefiles"
CMake Error at C:/Users/shzheng/Desktop/cmake-3.15.0-rc1-win64-x64/share/cmake-3.15/Modules/CMakeMinGWFindMake.cmake:12 (message):
  sh.exe was found in your PATH, here:

  C:/msys32/usr/bin/sh.exe

  For MinGW make to work correctly sh.exe must NOT be in your path.

  Run cmake from a shell that does not have sh.exe in your PATH.

  If you want to use a UNIX shell, then use MSYS Makefiles.

Call Stack (most recent call first):
  CMakeLists.txt:8 (project)

CMake Error: CMake was unable to find a build program corresponding to "MinGW Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMake was unable to find a build program corresponding to "MinGW Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
-------------------------------------------------end----------------------------------------------------------

I checked my environment variable PATH, the   `C:/msys32/usr/bin/sh.exe ` wasn't there.  I have ran where sh.exe in windows cmd, and get the response `INFO: Could not find files for the given pattern(s)`

However, I used MSYS Makefiles, things seemed to work. Files were generated in the folder. However, when I use `make flash` command to flash it in the boards, the IoT console has not received any MQTT messages. I think the default demo should be the MQTT demo right?

sheldon123z, 2019-06-21:
Hi, I also tested the command `cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B built -G"MinGW Makefiles"` under windows cmd, this is the error it gives. The "sh.exe ...." error was gone, but another error appears....

can I get the docs of the previous version of FreeRTOS? I had successfully ran the demos within 10 minutes before this release(6/11/2019), but this one gives me too much trouble. I actually just need FreeRTOS to do a small project with esp32.

--------------------------------------------------error-----------------------------------------------------------

c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master>cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B build2 -G"MinGW Makefiles"
-- Checking Python dependencies...
Python requirements from C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/requirements.txt are satisfied.
-- Building for target esp32
-- Component names: soc log heap freertos vfs newlib driver tcpip_adapter xtensa-debug-module app_trace partition_table app_update spi_flash mbedtls micro-ecc bootloader_support ethernet nvs_flash pthread smartconfig_ack wpa_supplicant esp32 cxx amazon-freertos-common bootloader bt console esp_adc_cal esptool_py expat wear_levelling sdmmc fatfs nghttp openssl spiffs ulp
-- Component paths: C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/soc;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/log;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/heap;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/freertos;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/vfs;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/newlib;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/driver;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/tcpip_adapter;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/xtensa-debug-module;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/app_trace;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/partition_table;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/app_update;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/spi_flash;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/mbedtls;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/micro-ecc;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/bootloader_support;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/ethernet;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/nvs_flash;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/pthread;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/smartconfig_ack;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/wpa_supplicant;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/esp32;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/cxx;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/amazon-freertos-common;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/bootloader;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/bt;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/console;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/esp_adc_cal;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/esptool_py;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/expat;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/wear_levelling;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/sdmmc;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/fatfs;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/nghttp;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/openssl;C:/Users/shzheng/D [... cut ...]
  C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/esptool_py/Kconfig.projbuild;--env;IDF_CMAKE=y).
  Error 1
Call Stack (most recent call first):
  vendors/espressif/esp-idf/CMakeLists.txt:62 (kconfig_process_config)

-- Configuring incomplete, errors occurred!
See also "C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/build2/CMakeFiles/CMakeOutput.log".
See also "C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/build2/CMakeFiles/CMakeError.log".

ulfalizer, 2019-06-21:
Kconfiglib guy here. The culprit in the last error log is this line:

    UnicodeDecodeError: 'gbk' codec can't decode byte 0x80 in position 7875: illegal multibyte sequence

Your machine is configured to use the GBK encoding, and some Kconfig file has data in it that isn't valid GBK.

Espressif uses an old (January 2018) patched version of Kconfiglib. More recent versions default to UTF-8 when reading files rather than the encoding specified in the environment (via `LC_CTYPE`). That fixes issues like this. Recent versions also catch decoding errors and tell you in what file (and approx. where in it) the error is.

If the Espressif guys want to stick with the old patched Kconfiglib, they ought to add `encoding='utf-8'` o the `open()` calls in `kconfiglib.py`. It'll solve issues like this at least. :)

On Linux, a workaround that usually works is to set `LC_CTYPE=C.UTF-8` in the environment. Won't work on Windows though. Not sure what's recommended there.

sheldon123z, 2019-06-21:
> Kconfiglib guy here. The culprit in the last error log is this line:
> 
> ```
> UnicodeDecodeError: 'gbk' codec can't decode byte 0x80 in position 7875: illegal multibyte sequence
> ```
> 
> Your machine is configured to use the GBK encoding, and some Kconfig file has data in it that isn't valid GBK.
> 
> Espressif uses an old (January 2018) patched version of Kconfiglib. More recent versions default to UTF-8 when reading files rather than the encoding specified in the environment (via `LC_CTYPE`). That fixes issues like this. Recent versions also catch decoding errors and tell you in what file (and approx. where in it) the error is.
> 
> If the Espressif guys want to stick with the old patched Kconfiglib, they ought to add `encoding='utf-8'` to the `open()` calls in `kconfiglib.py`. It'll solve issues like this at least. :)
> 
> On Linux, a workaround that usually works is to set `LC_CTYPE=C.UTF-8` in the environment. Won't work on Windows though. Not sure what's recommended there.

Thank you for your comments, I noticed that error but I didn't know how to fix it. :(

tgsong, 2019-06-21:
Hi, I'm sorry for the trouble, I understand it is frustrating to see something worked before but broken now. I don't have a link to our previous getting started guide but I'll bring this up to our doc team and see if we can provide versioned getting started doc.

If you still would like to try the current release, please use the [espressif CMake getting started guide](https://docs.espressif.com/projects/esp-idf/en/v3.1.3/get-started-cmake/windows-setup.html) to download the ESP32 toolchain instead of the regular one. We will update our document to correct the link.

The problem you were experiencing about "sh.exe" is because ESP32 used to build in MSYS with Makefile under Windows, and MSYS is a simulated Linux environment under Windows. Since our last release we have migrated the build system to CMake which allows you to build natively under Windows (you don't need any simulated Linux environment anymore) and also a lot faster, but this usually won't work if you try to run those commands under simulated Linux environment, like MSYS, CYGWIN, WSL or git bash.

Another point about the "build generator" in CMake, you don't need to know or write any underlying build files (whether a makefile or ninja file or something else) as that is all handled by CMake.

Below is the combination that I tried on a fresh Windows machine that works:
1. Install [ESP-IDF tools](https://dl.espressif.com/dl/esp-idf-tools-setup-1.1.exe) from [Espressif CMake get started guide](https://docs.espressif.com/projects/esp-idf/en/v3.1.3/get-started-cmake/windows-setup.html). Make sure you put the installation path in PATH environment variable. This should gives you ESP32 toolchain and CMake + Ninja + python2.7 (you can opt to not install those if you already have them). This should be 1GB less than the MSYS package they provided before for Makefile approach.
1. In the Amazon FreeRTOS root directory, run the CMake command under cmd or powershell, `cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B build -GNinja`
1. Invoke the build command, `cmake --build build`.

Sorry again for the trouble, I'll take your feeback and work with our doc team to address those problems.

sheldon123z, 2019-06-23:
Hi,
Thank you very much for your reply ! That has worked! After I downloaded the tools and ran the commands you provided, the build files has been generated successfully and the demo application has been built. However, now I am not able to flash the application to the board. The `make flash` command is not working, the cmd prompts `c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master\build>make flash
make: *** No rule to make target 'flash'`.

I also tried to use IDF script to flash it with the commands in the getting started guide `./vendors/espressif/esp-idf/tools/idf.py flash -B build`, and got the result:

c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master\build>python ../vendors/espressif/esp-idf/tools/idf.py flash -B build
Setting IDF_PATH environment variable: c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master\vendors\espressif\esp-idf
CMakeLists.txt not found in project directory c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master\build

tgsong, 2019-06-24:
Hi, the instruction I provided above use Ninja instead of makefile, you need use `ninja flash` instead of `make flash`. Another way is to use the CMake interface as it is generic to the underlying build system it generates, `cmake --build build --target flash`.

sheldon123z, 2019-06-27:
Hi tgsong,

Thanks for your help! I have tried the commands, and the application was seemed to flashed in. But I still didn't get anything on my IoT console, under the topic freertos/demo/echo.

here is the full record from generating build files to flashing the application to the board.
-----------------------------------------------------records------------------------------------------------------
Microsoft Windows [Version 10.0.17134.829]
(c) 2018 Microsoft Corporation. All rights reserved.

c:\Users\shzheng\Desktop>cd FreeRTOSWindows\amazon-freertos-master

c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master>cmake --build build --target flash
[0/2] Re-checking globbed directories...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D C:\Users\shzheng\Desktop\FreeRTOSW...indows/amazon-freertos-master/build" -P run_esptool.cmake"
Note: esptool.py will search for a serial port. To specify a port, set the ESPPORT environment variable.
Note: Using default baud rate 460800. To modify, set ESPBAUD environment variable.

c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master>cmake --build build4 --target flash
[0/2] Re-checking globbed directories...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cmd.exe /C "cd /D C:\Users\shzheng\Desktop\FreeRTOSW...ndows/amazon-freertos-master/build4" -P run_esptool.cmake"
Note: esptool.py will search for a serial port. To specify a port, set the ESPPORT environment variable.
Note: Using default baud rate 460800. To modify, set ESPBAUD environment variable.

c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master>aws --version
aws-cli/1.16.184 Python/3.6.0 Windows/10 botocore/1.12.174

c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master>cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -GNinja -S . -B build4
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/Espressif/ESP-IDF Tools/toolchain/bin/xtensa-esp32-elf-gcc.exe
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1")
-- Checking Python dependencies...
Python requirements from C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/requirements.txt are satisfied.
-- Building for target esp32
-- Component names: soc log heap freertos vfs newlib driver tcpip_adapter xtensa-debug-module app_trace partition_table app_update spi_flash mbedtls micro-ecc bootloader_support ethernet nvs_flash pthread smartconfig_ack wpa_supplicant esp32 cxx amazon-freertos-common bootloader bt console esp_adc_cal esptool_py expat wear_levelling sdmmc fatfs nghttp openssl spiffs ulp
-- Component paths: C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/soc;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/log;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/heap;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/freertos;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/vfs;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/newlib;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/driver;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/tcpip_adapter;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/xtensa-debug-module;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/app_trace;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/partition_table;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/app_update;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/spi_flash;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/mbedtls;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/micro-ecc;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/bootloader_support;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/ethernet;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/nvs_flash;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/pthread;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/smartconfig_ack;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/wpa_supplicant;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freertos-master/vendors/espressif/esp-idf/components/esp32;C:/Users/shzheng/Desktop/FreeRTOSWindows/amazon-freer [... cut ...]
Note: esptool.py will search for a serial port. To specify a port, set the ESPPORT environment variable.
Note: Using default baud rate 460800. To modify, set ESPBAUD environment variable.

c:\Users\shzheng\Desktop\FreeRTOSWindows\amazon-freertos-master>

---------------------------------------------finished---------------------------------------------------------

I think the ninja prompt "no work to do" is because I have flashed the application to the board once before. But there is nothing show up on the console.

Regards,
Sheldon

sheldon123z, 2019-06-28:
Hi,
I tried the command "ninja flash ESPPORT=COM16" but it doesn't work 
`c:\Users\shzheng\Desktop\amazon-freertos-master\buildnew>ninja flash ESPPORT=16
[0/2] Re-checking globbed directories...
ninja: error: unknown target 'ESPPORT=16'`

sarenameas, 2019-07-08:
Hello Sheldon, 
Are you still having issues?
The topic is no longer called "freertos/demo/echo". You must subscribe to: "iotdemo/#" on the AWS IoT Console test page. 

Please run the commands in this getting started guide to build and flash:
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#build-and-run-example-espressif

In the amazon-freertos root I run the following commands to build then flash and monitor:
`make -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -GNinja -S . -B build`
`cd build`
`ninja`
`cd ..`
`python ./vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p COM72 -B build`

closed 2019-08-05 by sheldon123z