aws/amazon-freertos #803: Nordic demo no longer runs without defining client certificates

ljaynes87, 2019-06-19:
Nordic demo now hard faults when leaving keyCLIENT_CERTIFICATE_PEM as an empty string in aws_clientcredential_keys.h.

**System information**
- Nordic NRF52840DK
- SEGGER Embedded Studio v4.12
- Amazon FreeRTOS release 201906.00_Major
- MQTT Demo

**Behavior**
When running the default MQTT Demo, the application will hard fault in file iot_demo_afr.c on line 596 when leaving keyCLIENT_CERTIFICATE_PEM defined as "" in the aws_clientcredential_keys.h.

ravibhagavandas, 2019-06-20:
Hi @ljaynes87,

This is because you are hitting an assert statement where a valid certificate is required for generating device metrics data. Metrics are generated per device to collect meaningful statistical data about the devices. To identify each device, it uses a unique identifier generated for the device using the device certificate. While we recommend devices connecting using BLE as well to be registered in AWS IoT cloud with a valid thing name and certificate, we don't enforce it. 
Having said that, you can still disable the metrics generation for the device by commenting out line 189 in iot_demo_afr.c. We will soon release a patch for not enforcing this.

ravibhagavandas, 2019-06-25:
The changes for not enforcing a valid certificate for metrics has been made as part of the PR: https://github.com/aws/amazon-freertos/pull/839
Please feel free to reopen if you have issues.

closed 2019-06-25 by ravibhagavandas