aws/amazon-freertos #810: Cannot run ESP32 Demo on Ubuntu 18.04.2 LTS due to cc1: some warnings being treated as errors

dryprogrammer, 2019-06-20:
**Describe the bug**
I encountered build error on `make flash` on Ubuntu 18.04.2 LTS due to cc1: some warnings being treated as errors.
I was following the Getting Started guide on Amazon FreeRTOS: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

IDF_PATH is not set.

**System information**
- Operating System: Ubuntu 18.04.2 LTS
- Version of Amazon FreeRTOS: v1.4.8-1-g48ba6402
- Project: ESP32 Demo

**Expected behavior**
Build successfully without error, and start flashing ESP32.

**Screenshots or console output**
```
dry@dry-ubuntu:~/aws/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make$ make flash   
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
GENCONFIG                                                                                                         
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu 
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o                                                
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/bootloader_init.o                                               
CC build/bootloader/bootloader_support/src/bootloader_common.o
/home/dry/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/bootloader_support/src/bootloader_common.c: In function 'bootloader_common_erase
_part_type_data':          
/home/dry/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/bootloader_support/src/bootloader_common.c:129:65: error: argument to 'sizeof' $
n 'strncpy' call is the same expression as the source; did you mean to use the size of the destination? [-Werror=sizeof-pointer-memaccess]
                 strncpy(label, (char *)&partition->label, sizeof(partition->label));      
                                                                 ^                     
cc1: some warnings being treated as errors
/home/dry/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/make/component_wrapper.mk:285: recipe for target 'src/bootloader_common.o' failed
make[2]: *** [src/bootloader_common.o] Error 1
/home/dry/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/make/project.mk:477: recipe for target 'component-bootloader_support-build' failed
make[1]: *** [component-bootloader_support-build] Error 2
/home/dry/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/bootloader/Makefile.projbuild:41: recipe for target '/home/dry/aws/amazon-freert
os/demos/espressif/esp32_devkitc_esp_wrover_kit/make/build/bootloader/bootloader.bin' failed
make: *** [/home/dry/aws/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make/build/bootloader/bootloader.bin] Error 2
```

**To reproduce**
Steps to reproduce the behavior:

1. Check out v1.4.8-1-g48ba6402
2. cd ./demos/espressif/esp32_devkitc_esp_wrover_kit/make
3. make flash

dryprogrammer, 2019-06-20:
After forcing CFLAGS -w -Wno-error=all by editing COMMON_WARNING_FLAGS in ~/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/make/project.mk, I get other build errors where the headers don't seem to get included properly:

```
AR build/bt/libbt.a                                                        
CC build/console/linenoise/linenoise.o                                                                                            
CC build/console/argtable3/argtable3.o                                                                                                                      
CC build/console/split_argv.o                                                        
CC build/console/commands.o                                                                                                                                           
AR build/console/libconsole.a                                                                                                                               
CXX build/cxx/cxx_guards.o                                                                                                                                            
CXX build/cxx/cxx_exception_stubs.o                                                  
In file included from /home/dry/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:18,                                                                   
                 from /home/dry/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdlib:75,                                                                   
                 from /home/dry/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/cxx/cxx_exception_stubs.cpp:1:
/home/dry/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/newlib/include/sys/reent.h:166:8: error: '_VOID' does not name a type         
 extern _VOID   _EXFUN(__sinit,(struct _reent *));                              
        ^~~~~                                                         
/home/dry/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/newlib/include/sys/reent.h:193:3: error: '_PTR' does not name a type           
   _PTR _cookie; /* cookie passed to io functions */                 
   ^~~~                                                                                       
/home/dry/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/newlib/include/sys/reent.h:195:36: error: '_read' has not been declared        
   _READ_WRITE_RETURN_TYPE _EXFNPTR(_read, (struct _reent *, _PTR,                                                                                                    
                                    ^~~~~                                        
/home/dry/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/newlib/include/sys/reent.h:195:43: error: expected identifier before '(' token
   _READ_WRITE_RETURN_TYPE _EXFNPTR(_read, (struct _reent *, _PTR,                                                                                                                                               ^                                                        
/home/dry/aws/amazon-freertos/lib/third_party/mcu_vendor/espressif/esp-idf/components/newlib/include/sys/reent.h:195:61: error: '_PTR' has not been declared
   _READ_WRITE_RETURN_TYPE _EXFNPTR(_read, (struct _reent *, _PTR,                   
                                                             ^~~~    
```

FYI, I'm my gcc is 7.4.0.
```
dry@dry-ubuntu:~/aws/amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make$ gcc --version
gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
111

tgsong, 2019-06-20:
Hi, Looks like you're using Espressif's latest compiler, to check the exact version, you can run `xtensa-esp32-elf-gcc --version`. (`gcc --version` is the regular compiler targeting your host machine, not the cross compiler used by ESP32)

The problem is Espressif latest compiler version 8.2 does not work with ESP-IDF 3.13 used in Amazon FreeRTOS, it will turn some warnings into errors, we also noticed this just recently. Please download their 5.2 compiler first and see if the problem still exist, you can find it in their old document, https://docs.espressif.com/projects/esp-idf/en/v3.1.3/get-started/linux-setup.html

And thanks for reporting the issue, in the meantime we'll update our getting started document to point this out.

dhwalters423, 2019-06-25:
FYI I ran into this exact issue as well using v. 8.2 of xtensa-esp32-elf-gcc on Mac OS 10.13.16.
Downgrading to v. 5.2 solved the issue for me.

sarenameas, 2019-07-08:
Hello @dryprogrammer has this resolved the issue for you? I will be closing this in the meantime.

closed 2019-07-08 by sarenameas