aws/amazon-freertos #812: Issues with cmake generated Makefile - Esp32 build

gregsochanik, 2019-06-20:
**Describe the bug**
Following the instructions here: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

I get the following error when trying to run `make all -j4` from MING32.

```
Greg@DESKTOP-AIBEA1B MINGW32 ~/amazon-freertos-build
$ make all
Makefile:35: *** missing separator.  Stop.
```

**System information**
- Which hardware board or part numbers?
ESP32_LyraT
- IDE used none
- Operating System Windows
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)
201906.00_Major-11-g2bb31547
- Project [MQTT]
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
According to the docs I should be able to do the following after running cmake from the amazon-freertos root dir:

To build the application with Make

Change directories to the build directory.

Use the following command to build the application with Make:

`make all -j4`
Note

You must generate the build files with the cmake command every time you switch between the aws_demos project and the aws_tests project.

**Screenshots or console output**

```
Greg@DESKTOP-AIBEA1B MINGW32 ~/amazon-freertos-build
$ `make all -j4`
Makefile:35: *** missing separator.  Stop.
```

**To reproduce**
Steps to reproduce the behavior:

1. Pull down the latest tag
2. Follow the "Build, Flash, and Run the Amazon FreeRTOS Demo Project" instructions from the https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

**Additional context**

It looks like the docs have been updated recently as I had been following them successfully, using make [all|app|fash|monitor] etc with an older version of AmazonFreeRTOS,

Thank you!

tgsong, 2019-06-20:
Hi, can you post the exact CMake command that you ran? Our doc did not clearly say that if you're on Windows, you need to specify the build system to use, otherwise it will pick up Visual Studio by default. This is documented in our general [getting started guide](https://docs.aws.amazon.com/freertos/latest/userguide/building-cmake.html). Under the CMake commandline instructure, there's a note saying "If you are using Windows, you must specify the native build system because CMake uses Visual Studio by default."

gregsochanik, 2019-06-21:
Thanks for the response!

 Sure:
'cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B ../amazon-freertos-build'

I'm not using Visual Studio - I don't have in installed.

I'll post a gist of the generated Makefile and link from here

gregsochanik, 2019-06-21:
Quick update - I have taken a look at the getting started guide and can see the following line:

> Note
>
> If you are using Windows, you must specify the native build system because CMake uses Visual Studio by default. For example:
>
> `cmake -DVENDOR=ti -DBOARD=cc3220_launchpad -DCOMPILER=arm-ti -S . -B build -G "MinGW Makefiles"`

I'm using MINGW32 and a quick run of `cmake --help` shows that it's using `NMake Makefiles` as default.

So - I tried explicitly setting MinGW Makefiles as outlined in the docs and got the following error:

```
$ cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B ../amazon-freertos-201906.00_Major.build -G "MinGW Makefiles"
CMake Error at C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeMinGWFindMake.cmake:12 (message):
  sh.exe was found in your PATH, here:

  C:/msys32/usr/bin/sh.exe

  For MinGW make to work correctly sh.exe must NOT be in your path.

  Run cmake from a shell that does not have sh.exe in your PATH.

  If you want to use a UNIX shell, then use MSYS Makefiles.

Call Stack (most recent call first):
  CMakeLists.txt:8 (project)

```

so I switched to using MSYS Makefiles as the error suggested and it's all working 👍

tgsong, 2019-06-21:
Glad you got it working. In case you face any issue in the future, one thing you might want to check is the [Espressif CMake get started guide](https://docs.espressif.com/projects/esp-idf/en/v3.1.3/get-started-cmake/windows-setup.html). Since we're switching to CMake based build, actually you don't need the whole MSYS simulated Linux environment anymore because CMake offers you the option to build natively on Windows. We have another post experiencing some similar problems, you can check my comment there, https://github.com/aws/amazon-freertos/issues/802#issuecomment-504587466. Basically the recommended approach for running ESP32 in the latest release on Windows is "ESP32 compiler + CMake + Ninja + cmd/powershell".

sheldon123z, 2019-06-23:
@gregsochanik  Hi man, can you receive any mqtt message on IoT console in this way? I have done it in this way before using the `MSYS Makefiles`, it indeed produces some building files, but after I used the `make flash` command, nothing happened to my IoT console. I am pretty much sure that my IoT settings are correct. 
Also, the `make menuconfig` commands produced unclear setting interface under this way, and I think it is different with the one of the last release version that I can easily set the COM port number.

tgsong, 2019-06-24:
@sheldon123z Hi, regarding to the MQTT message issue, are you using the quick start script in `tools/aws_config_quick_start/SetupAWS.py`? We identified a bug in this script after our last release, the credential template file was not updated, this is fixed in this commit: f9c987a. If that's the case, you can pull the latest code on master branch.

For serial port you can export an environment variable ESPPORT, e.g., if you are using bash in a Linux environment, `export ESPPORT=/dev/ttyUSB1`.

qiutongs, 2019-07-16:
Free free to reopen if you still have question.

closed 2019-07-16 by qiutongs