aws/amazon-freertos #815: [question] how to handle MQTT CONNECT returned error NETWORK ERROR

chegewara, 2019-06-20:
Hi,
i am testing code against different network situations and i had very unusual situation. I am connecting esp32 to smartphone AP and one time it happened to me that assigned STA IP was in different subnet than GW (IP from my home network, GW from smartphone). 
```
I (2466) WIFI: SYSTEM_EVENT_STA_CONNECTED
2 331 [IP-task] vDHCPProcess: offer c0a82bd8ip
I (3796) event: sta ip: 192.168.0.105, mask: 255.255.255.0, gw: 192.168.43.1
I (3796) WIFI: SYSTEM_EVENT_STA_GOT_IP
3 354 [IP-task] vDHCPProcess: offer c0a82bd8ip
I (3806) my_connectedCallback: Connected callback
4 355 [iot_thread] [INFO ][DEMO][3550] Successfully initialized the demo. Network type for the demo: 1
5 380 [conT] [INFO ][DEMO][3800] Shadow Thing Name is iot-test (length 8).
6 382 [conT] [ERROR][NET][3820] Failed to resolve xxxxxxxxx-ats.iot.eu-central-1.amazonaws.com.
7 382 [conT] [ERROR][MQTT][3820] Failed to establish new MQTT connection, error NETWORK ERROR.
8 382 [conT] [ERROR][DEMO][3820] MQTT CONNECT returned error NETWORK ERROR.
```
Then i thought i will turn off and turn on router in my smartphone, so esp32 should disconnect and connect again, but in esp32 i didnt get any on disconnect event (which would suggest that esp32 is connected to home network, but i have very different SSIDs on both networks).

Im thinking its some glitch in flash, because before i provisioned wifi credentials with BLE i did erase flash.

My question is, how to handle `MQTT CONNECT returned error NETWORK ERROR`.

closed 2019-06-21 by chegewara

sheldon123z, 2019-07-02:
Hi,
Did you solve this problem? I encountered the same problem as you did.

chegewara, 2019-07-02:
This is issue that is unlikely to happen after developing stage (different subnet for IP and GW), so i decided to ignore it.