aws/amazon-freertos #816: OTA_AgentShutdown issue

chegewara, 2019-06-20:
I have code where i am calling `OTA_AgentShutdown` in loop as long as wont get `status != eOTA_AgentState_NotReady` (now i know i need to await `eOTA_AgentState_ShuttingDown`). 
With long timeout like 10000ms (1000 ticks) or so im getting always status 2 (eOTA_AgentState_Active):
```
E (146226) _disconnectCallback: OTA shutdown                 // before calling 
E (176226) _disconnectCallback: OTA shutdown => 2        // after return from 
E (176226) _disconnectCallback: OTA shutdown
E (206226) _disconnectCallback: OTA shutdown => 2
E (206226) _disconnectCallback: OTA shutdown
E (236226) _disconnectCallback: OTA shutdown => 2
E (236226) _disconnectCallback: OTA shutdown
E (266226) _disconnectCallback: OTA shutdown => 2
```
and exactly the same code but with short timeout (let say about 300 ticks):
```
E (124076) _disconnectCallback: OTA shutdown
E (124076) _disconnectCallback: OTA shutdown => 3
E (124076) _disconnectCallback: OTA shutdown
E (124076) _disconnectCallback: OTA shutdown => 3
```

I am working with branch release-1.5, commit e88e2ce08b6e52c89b93a38f72511c5cfc0dbffc

chegewara, 2019-06-20:
After further investigation i found that code cant pass this line:
https://github.com/aws/amazon-freertos/blob/master/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c#L792

It can be issue in my code, which is a bit changed aws_iot_otaupdate_demo, or it can be problem with that code. 
It does not happen always, when esp32 disconnect from wifi then everything works good, but when is still connected to wifi, but disconnects from mqtt because of:
```
198 8012 [iot_thread] [ERROR][MQTT][80120] (MQTT connection 0x3ffbf7dc) Failed to send PINGREQ.
199 8012 [iot_thread] [WARN ][NET][80120] Failed to close connection.
200 8012 [iot_thread] [INFO ][MQTT][80120] (MQTT connection 0x3ffbf7dc) Network connection closed.
```
then this issue occurs.

chegewara, 2019-06-21:
Its better logging for comparison.
wifi connection lost:
```
I (54317) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 3
I (54327) my_disconnectedCallback: Disconnected callback
E (54327) _disconnectCallback: mqtt disconnect callback
E (54337) _disconnectCallback: OTA shutdown
64 5407 [iot_thread] [INFO ][MQTT][54070] (MQTT connection 0x3ffbfecc) Disconnecting connection.
65 5407 [iot_thread] [INFO ][MQTT][54070] (MQTT connection 0x3ffbfecc) Network connection closed.
66 5408 [iot_thread] [INFO ][Shadow][54080] Shadow library cleanup done.
67 5408 [iot_thread] [OTA_AgentShutdown] Start: 500 ticks
68 5409 [OTA Task] [prvOTAUpdateTask] Received shutdown event.
E (54347) prvAgentShutdownCleanup: before prvUnSubscribeFromJobNotificationTopic
E (54377) prvAgentShutdownCleanup: after prvUnSubscribeFromJobNotificationTopic
69 5412 [OTA Task] [WARN ][MQTT][54120] (MQTT connection 0x3ffbfecc) Attempt to use closed connection.
70 5412 [OTA Task] [ERROR][MQTT][54120] (MQTT connection 0x3ffbfecc) New operation record cannot be created for a closed connection
71 5412 [OTA Task] [prvUnSubscribeFromJobNotificationTopic] FAIL: $aws/things/alexa-iot-test/jobs/notify-next
72 5412 [OTA Task] [WARN ][MQTT][54120] (MQTT connection 0x3ffbfecc) Attempt to use closed connection.
73 5412 [OTA Task] [ERROR][MQTT][54120] (MQTT connection 0x3ffbfecc) New operation record cannot be created for a closed connection
74 5412 [OTA Task] [prvUnSubscribeFromJobNotificationTopic] FAIL: $aws/things/alexa-iot-test/jobs/$next/get/accepted
75 5413 [OTA Task] [prvOTA_Close] Context->0x3ffc6740
76 5414 [iot_thread] [OTA_AgentShutdown] End: 494 ticks
E (54397) _disconnectCallback: OTA shutdown => 0
```
mqtt disconnect because of PING timeout:
```
E (168057) _disconnectCallback: mqtt disconnect callback
E (168057) _disconnectCallback: OTA shutdown
151 16780 [iot_thread] [ERROR][MQTT][167800] (MQTT connection 0x3ffbe300) Failed to receive PINGRESP within 10000 ms.
152 16780 [iot_thread] [WARN ][NET][167800] Failed to close connection.
153 16780 [iot_thread] [INFO ][MQTT][167800] (MQTT connection 0x3ffbe300) Network connection closed.
154 16780 [iot_thread] [INFO ][Shadow][167800] Shadow library cleanup done.
155 16780 [iot_thread] [OTA_AgentShutdown] Start: 500 ticks
156 16781 [OTA Task] [prvOTAUpdateTask] Received shutdown event.
E (168067) prvAgentShutdownCleanup: before
E (173057) _disconnectCallback: OTA shutdown => 3
E (173057) _disconnectCallback: OTA shutdown
E (173057) _disconnectCallback: OTA shutdown => 3
```

pvyawaha, 2019-06-21:
Hi chegewara,

`I have code where i am calling OTA_AgentShutdown in loop as long as wont get status != eOTA_AgentState_NotReady(0).`

The OTA_AgentShutdown sends an event OTA_EVT_MASK_SHUTDOWN to the OTA update task and waits for it to complete the shutdown and the OTA agent state to be set to eOTA_AgentState_NotReady(0).

Once the OTA update task receives this event it sets the agent state to eOTA_AgentState_ShuttingDown(3) and the performs number of cleanup steps on setup that was performed when OTA_AgentInit was called  - 

1. Stop and delete timers.
2. Unsubscribe to OTA job notification topic 
3. Close OTA file
4. Delete any shared memory , queue used etc.

From the log it seems the unsubscribe is failing as connection is closed before that. Can you share more information on your changes in OTA demo. You can also call OTA_AgentShutdown followed by OTA_GetAgentState to confirm that state is changed eOTA_AgentState_NotReady(0).

chegewara, 2019-06-21:
Hi @pvyawaha 
ive been reading thru aws_iot_ota_agent code and i think i understand it now (at least more than yesterday). `OTA_AgentShutdown` works good most of the time (no matter if its called before or after ``), the only issue i have in one particular situation, when wifi is still connected, but there is no access to internet. To simulate this situation i have esp32 connected to smartphone router and after normal start when all is up and running i just disconnect smartphone from internet but esp32 is still connected to smartphone router. This is leading to PINGRESP timeout and mqtt connection is closed.

I will try the same situation with original OTA example (maybe this weekend), but here you can see my changes:
```
/*
 * Amazon FreeRTOS V201906.00 Major
 * Copyright (C) 2019 Amazon.com, Inc. or its affiliates.  All Rights Reserved.
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy of
 * this software and associated documentation files (the "Software"), to deal in
 * the Software without restriction, including without limitation the rights to
 * use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of
 * the Software, and to permit persons to whom the Software is furnished to do so,
 * subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in all
 * copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS
 * FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR
 * COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER
 * IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN
 * CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
 * http://aws.amazon.com/freertos
 * http://www.FreeRTOS.org
 */

/**
 * @file aws_ota_update_demo.c
 * @brief A simple OTA update example.
 *
 * This example initializes the OTA agent to enable OTA updates via the
 * MQTT broker. It simply connects to the MQTT broker with the users
 * credentials and spins in an indefinite loop to allow MQTT messages to be
 * forwarded to the OTA agent for possible processing. The OTA agent does all
 * of the real work; checking to see if the message topic is one destined for
 * the OTA agent. If not, it is simply ignored.
 */
/* The config header is always included first. */
#include "iot_config.h"

/* MQTT include. */
#include "iot_mqtt.h"
/* Standard includes. */
#include <stdio.h>
#include <string.h>

/* FreeRTOS includes. */
#include "FreeRTOS.h"
#include "task.h"
#include "semphr.h"

/* Demo network handling */
#include "aws_iot_demo_network.h"

/* Required to get the broker address and port. */
#include "aws_clientcredential.h"

/* Amazon FreeRTOS OTA agent includes. */
#include "aws_ota_agent.h"

#include "iot_network_manager_private.h"

/* Required for demo task stack and priority */
#include "aws_demo_config.h"
#include "aws_application_version.h"

[... 297 lines cut ...]
and this callback to control mqtt disconnect event (either called from `my_disconnectedCallback` or when PINGRESP timeout occurs):
```
static void _disconnectCallback( void * pCallbackContext, IotMqttCallbackParam_t * pCallbackParam )
{
    ESP_LOGE(__func__, "mqtt disconnect callback");
    tasksStatus();
    int status = 0;
    /**
     * Deinit shadow and OTA task
     */
    AwsIotShadow_Cleanup();
    do{
        ESP_LOGE(__func__, "OTA shutdown");
        status = OTA_AgentShutdown(500);
        ESP_LOGE(__func__, "OTA shutdown => %d", status);
    }while(status != eOTA_AgentState_NotReady);
    
    if(OTAhandle != NULL)
    {
        vTaskDelete(OTAhandle);
        OTAhandle = NULL;
    }
}
```
Much appreciated help and great work on library.

chegewara, 2019-06-24:
Hi,
i cant reproduce it with OTA example, because `DEMO_networkDisconnectedCallback` is never called, even if i disconnect smartphone from internet OTA stays in loop:
`668 65145 [iot_thread] State: Ready  Received: 1   Queued: 1   Processed: 1   Dropped: 0`.

chegewara, 2019-06-25:
Ok, i found a way to reproduce it on OTA example.

I changed 2 things in code. 
1. change `otaDemoKEEPALIVE_SECONDS` to 12 seconds (waiting 20 minutes is too long for test),
2. this disconnect callback:
https://github.com/aws/amazon-freertos/blob/master/demos/ota/aws_iot_ota_update_demo.c#L142
```
static IotNetworkError_t prvNetworkDisconnectCallback( void* pvContext )
{
    ( void ) pvContext;
    xNetworkConnected = pdFALSE;
    int status = 0;
    do{
        printf("OTA_AgentShutdown\n");
        status = OTA_AgentShutdown(500);
        vTaskDelay(100); // delay to let logs prints out
    }while(status != eOTA_AgentState_NotReady);
    return IOT_NETWORK_SUCCESS;
}
```

When device is still connected to wifi, but no access to internet im getting this logs:
```
88 6247 [iot_thread] State: Ready  Received: 1   Queued: 1   Processed: 1   Dropped: 0
OTA_AgentShutdown
89 6314 [iot_thread] [ERROR][MQTT][63140] (MQTT connection 0x3ffd3b94) Failed to receive PINGRESP within 10000 ms.
90 6314 [OTA Task] [prvOTAUpdateTask] Received shutdown event.
91 6347 [iot_thread] State: Ready  Received: 1   Queued: 1   Processed: 1   Dropped: 0
92 6447 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
93 6547 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
94 6647 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
95 6747 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
96 6847 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
OTA_AgentShutdown
97 6914 [iot_thread] [OTA_AgentShutdown] Nothing to do: Already in state 3
98 6947 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
OTA_AgentShutdown
99 7014 [iot_thread] [OTA_AgentShutdown] Nothing to do: Already in state 3
100 7047 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
OTA_AgentShutdown
101 7114 [iot_thread] [OTA_AgentShutdown] Nothing to do: Already in state 3
102 7147 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
OTA_AgentShutdown
103 7214 [iot_thread] [OTA_AgentShutdown] Nothing to do: Already in state 3
104 7247 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
OTA_AgentShutdown
105 7314 [iot_thread] [OTA_AgentShutdown] Nothing to do: Already in state 3
106 7347 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
OTA_AgentShutdown
107 7414 [iot_thread] [OTA_AgentShutdown] Nothing to do: Already in state 3
108 7447 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
OTA_AgentShutdown
109 7514 [iot_thread] [OTA_AgentShutdown] Nothing to do: Already in state 3
110 7547 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
OTA_AgentShutdown
111 7614 [iot_thread] [OTA_AgentShutdown] Nothing to do: Already in state 3
112 7647 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
OTA_AgentShutdown
113 7714 [iot_thread] [OTA_AgentShutdown] Nothing to do: Already in state 3
114 7747 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
OTA_AgentShutdown
115 7814 [iot_thread] [OTA_AgentShutdown] Nothing to do: Already in state 3
116 7847 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
OTA_AgentShutdown
117 7914 [iot_thread] [OTA_AgentShutdown] Nothing to do: Already in state 3
118 7947 [iot_thread] State: Shutting down  Received: 1   Queued: 1   Processed: 1   Dropped: 0
```

pvyawaha, 2019-07-02:
Hi Chegewara,

Thank you for sharing the logs. The call to the OTA_AgentShutdown seems to be faster than the wait times for the unsubscribe operations in the shutdown process.  The OTA agent is subscribe to two job notification topics so during shutdown it unsubscribes from these topics. The wait time for the unsubscribe operation is defined in OTA agent as 

`#define OTA_UNSUBSCRIBE_WAIT_MS            1000UL`

It basically calls IotMqtt_Wait with this timeout for  both unsubscribe operations.

chegewara, 2019-07-03:
Hi, thanks for response.
I changed `OTA_AgentShutdown` timeout to 5000 ticks, still the same. ( i tried `portMAX_DELAY` with same result)
```
    do{
        ESP_LOGE(__func__, "OTA shutdown");
        status = OTA_AgentShutdown(5000);
        IotClock_SleepMs(1000);
        status = OTA_GetAgentState();
        ESP_LOGE(__func__, "OTA shutdown => %d", status);
    }while(status != eOTA_AgentState_NotReady);
```
logs:
```
67 8239 [iot_thread] [OTA_AgentShutdown] Start: 5000 ticks
68 8240 [OTA Task] [prvOTAUpdateTask] Received shutdown event.
E (82667) prvAgentShutdownCleanup: before
E (82697) prvUnSubscribeFromJobNotificationTopic: 1. inside prvUnSubscribeFromJobNotificationTopic
E (82707) prvUnSubscribeFromJobNotificationTopic: 2. inside prvUnSubscribeFromJobNotificationTopic
69 13239 [iot_thread] [OTA_AgentShutdown] End: 0 ticks
E (133657) _disconnectCallback: OTA shutdown => 3
```
Like i said, code does stuck on this function:
```
   if( ( stUnSub.topicFilterLength > 0U ) && ( stUnSub.topicFilterLength < sizeof( pcJobTopic ) ) )
    {
ESP_LOGE(__func__, "2. inside prvUnSubscribeFromJobNotificationTopic");
        if( IotMqtt_Unsubscribe( xOTA_Agent.pvPubSubClient,
                                 &stUnSub,
                                 1, /* Subscriptions count */
                                 IOT_MQTT_FLAG_WAITABLE, /* flags */
								 NULL,
                                 &( unsubscribeOperation[ 0]) ) != IOT_MQTT_STATUS_PENDING )
        {
            OTA_LOG_L1( "[%s] FAIL: %s\n\r", OTA_METHOD_NAME, stUnSub.pTopicFilter );
        }
        else
        {
            OTA_LOG_L1( "[%s] OK: %s\n\r", OTA_METHOD_NAME, stUnSub.pTopicFilter );
        }
```
https://github.com/aws/amazon-freertos/blob/master/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c#L2938

chegewara, 2019-07-03:
I know OTA is working fine, because i tested it few times already with OTA example, but now i am trying with all changes i posted here and this is result after downloading new image with OTA Job:
```
I (95732) ota_pal: No such certificate file: xxx. Using aws_ota_codesigner_certificate.h.

2612 9542 [OTA Task] [prvIngestDataBlock] Received file block 1133, size 1024
2613 9542 [OTA Task] [prvIngestDataBlock] Received final expected block of file.
2614 9542 [OTA Task] [prvStopRequestTimer] Stopping file request timer.
2615 9542 [OTA Task] ERROR: Object with label xxx not found.
2616 9793 [OTA Task] [prvIngestDataBlock] File receive complete and signature is valid.
2617 9793 [OTA Task] [prvUpdateJobStatus] Msg: {"status":"IN_PROGRESS","statusDetails":{"self_test":"ready","updatedBy":"0x30001"}}
2618 9834 [OTA Task] [prvUpdateJobStatus] 'IN_PROGRESS' to $aws/things/alexa-iot-test/jobs/AFR_OTA-chaya_ota_v032_1/update
2619 9834 [OTA Task] [prvOTA_Close] Context->0x3ffcc720
2620 9841 [OTA Task] [prvUnSubscribeFromDataStream] OK: $aws/things/alexa-iot-test/streams/AFR_OTA-d9c4b79b-f9ea-4c98-aeaf-c042e7e9b083/data/cbor
2621 9841 [OTA Task] Received eOTA_JobEvent_Activate callback from OTA Agent.
E (98732) _disconnectCallback: mqtt disconnect callback
E (98742) _disconnectCallback: OTA shutdown
2622 9843 [OTA Task] [OTA_AgentShutdown] Start: 5000 ticks
2623 14926 [OTA Task] [OTA_AgentShutdown] End: 0 ticks
E (150572) _disconnectCallback: OTA shutdown => 2
```

I will close this issue for now, until i understand better whats going on.

closed 2019-07-03 by chegewara

ahtashambaig, 2020-01-11:
Hi i am working to setup OTA demo for ESP32 using Amazon FreeRTOS i have followed all the steps as said but at the output I am getting the my files are being received but not processed
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7312
ho 0 tail 12 room 4
load:0x40078000,len:12620
load:0x40080400,len:6708
entry 0x40080778
I (75) boot: Chip Revision: 1
I (75) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v3.3-163-g601a03e 2nd stage bootloader
I (41) boot: compile time 15:11:39
I (41) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 40MHz
I (50) boot: SPI Mode       : DIO
I (54) boot: SPI Flash Size : 4MB
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (77) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (84) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (92) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (99) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (107) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (114) boot: End of partition table
I (118) boot: ota rollback check done
I (123) boot_comm: chip revision: 1, min. application chip revision: 0
I (130) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x253dc (152540) map
I (192) esp_image: segment 1: paddr=0x00045404 vaddr=0x3ffbdb60 size=0x0336c ( 13164) load
I (198) esp_image: segment 2: paddr=0x00048778 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:\aws-freertos-backup\ahtasham_ota\freertos-1578549574684\AmazonFreeRTOS\build/../freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1685

I (200) esp_image: segment 3: paddr=0x00048b80 vaddr=0x40080400 size=0x07490 ( 29840) load
I (221) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xab74c (702284) map
0x400d0018: _flash_cache_start at ??:?

I (467) esp_image: segment 5: paddr=0x000fb76c vaddr=0x40087890 size=0x0f1e4 ( 61924) load
0x40087890: ld_inq_frm_isr_wrapper at ??:?

I (506) boot: Loaded app from partition at offset 0x20000
I (507) boot: Disabling RNG early entropy source...
I (507) cpu_start: Pro cpu up.
I (511) cpu_start: Application information:
I (515) cpu_start: Project name:     esp-idf
I (520) cpu_start: App version:      1
I (525) cpu_start: Compile time:     Jan  9 2020 15:11:14
I (531) cpu_start: ELF file SHA256:  c02c40b55ad5a2c1...
I (537) cpu_start: ESP-IDF:          v3.3-163-g601a03e
I (543) cpu_start: Single core mode
I (547) heap_init: Initializing. RAM available for dynamic allocation:
I (554) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (560) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (566) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (572) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (578) heap_init: At 3FFCFCB0 len 00010350 (64 KiB): DRAM
I (585) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (591) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (597) heap_init: At 40096A74 len 0000958C (37 KiB): IRAM
I (603) cpu_start: Pro cpu start user code
I (286) cpu_start: Starting scheduler on PRO CPU.
I (308) uart: queue free spaces: 10
0 1 [main] Initializing FreeRTOS TCP stack
I (308) PKCS11: Initializing NVS partition: "storage"
E (308) PKCS11: failed nvs get file size 4354 0
1 1 [main] Write certificate...
2 16 [iot_thread] [INFO ][DEMO][160] ---------STARTING DEMO---------

[... 516 lines cut ...]
482 12306 [iot_thread] State: WaitingForFileBlock  Received: 94   Queued: 0   Processed: 0   Dropped: 0
483 12346 [OTA Agent Task] [prvIngestDataBlock] Received file block 94, size 4096
484 12346 [OTA Agent Task] [prvIngestDataBlock] Remaining: 141
485 12347 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
486 12406 [iot_thread] State: WaitingForFileBlock  Received: 95   Queued: 0   Processed: 0   Dropped: 0
487 12438 [OTA Agent Task] [prvIngestDataBlock] Received file block 93, size 4096
488 12439 [OTA Agent Task] [prvIngestDataBlock] Remaining: 140
489 12439 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
490 12506 [iot_thread] State: WaitingForFileBlock  Received: 96   Queued: 0   Processed: 0   Dropped: 0
491 12532 [OTA Agent Task] [prvIngestDataBlock] Received file block 95, size 4096
492 12533 [OTA Agent Task] [prvIngestDataBlock] Remaining: 139
493 12533 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
494 12533 [OTA Agent Task] [INFO ][MQTT][125330] (MQTT connection 0x3fff0b00) MQTT PUBLISH operation queued.
495 12533 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/quickconnect_esp32/streams/AFR_OTA-8e649e99-a21f-46b8-a3df-ced243405fcc/get/cbor
496 12533 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock]
[... cut ...]
501 12706 [iot_thread] State: WaitingForFileBlock  Received: 98   Queued: 0   Processed: 0   Dropped: 0
502 12796 [OTA Agent Task] [prvIngestDataBlock] Received file block 97, size 4096
503 12797 [OTA Agent Task] [prvIngestDataBlock] Remaining: 137
504 12797 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]

Kindly respond, I am not sure code working properly or not i am generating software from Amazon Console