aws/amazon-freertos #817: Program to just listen to the delta topic

escanzanocloudcath, 2019-06-20:
Hi,

I have the 201906.00 Major version of the FreeRTOS. I have studied the shadow demo and understand it pretty well. I want a shadow program that just listens to the delta topic. In the demo, the _SendShadowUpdates function execute this for loop: 

for( i = 1; i <= AWS_IOT_DEMO_SHADOW_UPDATE_COUNT; i++ )

When done, the code exits. I want a program that does not exit but rather continuously listens to the delta topic. Would I make a function like this 

static int _background()
{
    for(;;)
    {
        IotClock_SleepMs( 3000 );
    }

    return EXIT_SUCCESS;
}

So, for example, would this code replace lines 877-880 of the was_iot_demo_shadow.c file?

Thanks

chegewara, 2019-06-21:
You can just remove part of code that makes shadow cleanup. This is how i do:
```
int _RunShadowDemo( const char * pIdentifier, IotMqttConnection_t mqttConnection )
{
    /* Return value of this function and the exit status of this program. */
    int status = 0;

    /* Length of Shadow Thing Name. */
    size_t thingNameLength = 0;
    IotSemaphore_t deltaSemaphore;

    /* Flags for tracking which cleanup functions must be called. */
    bool deltaSemaphoreCreated = false;

    /* Determine the length of the Thing Name. */
    if( pIdentifier != NULL )
    {
        thingNameLength = strlen( pIdentifier );

        if( thingNameLength == 0 )
        {
            IotLogError( "The length of the Thing Name (identifier) must be nonzero." );

            status = EXIT_FAILURE;
        }
    }
    else
    {
        IotLogError( "A Thing Name (identifier) must be provided for the Shadow demo." );

        status = EXIT_FAILURE;
    }

    /* Initialize the libraries required for this demo. */
    if( status == EXIT_SUCCESS )
    {
        status = _initializeDemo();
    }

    if( status == EXIT_SUCCESS )
    {
        /* Create the semaphore that synchronizes with the delta callback. */
        deltaSemaphoreCreated = IotSemaphore_Create( &deltaSemaphore, 0, 1 );

        if( deltaSemaphoreCreated == false )
        {
            status = EXIT_FAILURE;
        }
    }
    else
    {
        IotLogError("Failed to init shadow");
    }
    

    if( status == EXIT_SUCCESS )
    {   if(queue == NULL)
        {
            queue = xQueueCreate(5, sizeof(DeltaEvent_t));

            xTaskCreate(delta_event_handler, "deltaT", 2*1024, NULL, 5, &deltaHandle);
        }
        /* Set the Shadow callbacks for this demo. */
        status = _setShadowCallbacks( &deltaSemaphore, mqttConnection, pIdentifier, thingNameLength );
    }
    else
    {
        IotLogError("Failed to set callbacks shadow");
    }

    if( status == EXIT_SUCCESS )
    {
        /* Send Shadow updates. */
        status = _sendShadowUpdates( &deltaSemaphore, mqttConnection, pIdentifier, thingNameLength ); // <-- send only 1 update on device reboot

    }
    else
    {
        IotLogError("Failed to send shadow update");
    }

    return status;
}
```

escanzanocloudcath, 2019-06-21:
chegewara, Thanks for your response.

Your code does not compile for me. Issues with DeltaEvent_t, &deltaHandle. Care to give a more complete example?

Much appreciated.

chegewara, 2019-06-21:
This is code that you need to delete:
https://github.com/aws/amazon-freertos/blob/master/demos/shadow/aws_iot_demo_shadow.c#L899-L915

tgsong, 2019-06-22:
Hi, in this case, you can delete the cleanup code chegewara mentioned above, and replace it with a infinite loop like `while(1) { IotClock_SleepMs( 3000 ); }` to make it keep running.

qiutongs, 2019-07-15:
If you still have question, feel free to re-open this issue again.

closed 2019-07-15 by qiutongs