aws/amazon-freertos #818: Error in esp_spi_flash.h when building OTA demo

gregsochanik, 2019-06-21:
**Describe the bug**
I cannot build the binary when enabling the OTA demo

running `idf.py all` results in the following:

```
In file included from C:/msys32/home/Greg/amazon-freertos-201906.00_Major/vendors/espressif/esp-idf/components/spi_flash/include/esp_partition.h:22:0,
                 from C:/msys32/home/Greg/amazon-freertos-201906.00_Major/vendors/espressif/boards/esp32/ports/ota/aws_ota_pal.c:33:
C:/msys32/home/Greg/amazon-freertos-201906.00_Major/vendors/espressif/esp-idf/components/spi_flash/include/esp_spi_flash.h:45:1: error: expected ',' or ';' before 'void'
 void spi_flash_init();
 ^

```

Default MQTT demo works as expected.

**System information**
- Which hardware board or part numbers? **ESP32_LyraT**
- IDE used: **MinGW32**
- Operating System [**Windows**]
- Version of Amazon FreeRTOS **201906.00_Major-33-g757ff18c**
- Project [**OTA**]

**Expected behavior**
I expect the ap to build

**Screenshots or console output**
```
[ 96%] Building C object CMakeFiles/afr_ota.dir/vendors/espressif/boards/esp32/ports/ota/aws_ota_pal.c.obj
[ 96%] Built target afr_defender
In file included from C:/msys32/home/Greg/amazon-freertos-201906.00_Major/vendors/espressif/esp-idf/components/spi_flash/include/esp_partition.h:22:0,
                 from C:/msys32/home/Greg/amazon-freertos-201906.00_Major/vendors/espressif/boards/esp32/ports/ota/aws_ota_pal.c:33:
C:/msys32/home/Greg/amazon-freertos-201906.00_Major/vendors/espressif/esp-idf/components/spi_flash/include/esp_spi_flash.h:45:1: error: expected ',' or ';' before 'void'
 void spi_flash_init();
 ^
make[2]: *** [CMakeFiles/afr_ota.dir/build.make:102: CMakeFiles/afr_ota.dir/vendors/espressif/boards/esp32/ports/ota/aws_ota_pal.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:683: CMakeFiles/afr_ota.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
Setting IDF_PATH environment variable: C:/msys32/home/Greg/amazon-freertos-201906.00_Major/vendors/espressif/esp-idf
Running make in directory C:/msys32/home/Greg/amazon-freertos-201906.00_Major.build
Executing "make -j 14 all"...
make failed with exit code 2
```
**To reproduce**
Steps to reproduce the behavior:

1. Edit `aws_demo_config.h` and specify `#define CONFIG_OTA_UPDATE_DEMO_ENABLED`
2. Run the following command from the CLI `./vendors/espressif/esp-idf/tools/idf.py all -p COM3 -B ../amazon-freertos-201906.00_Major.build/`

tgsong, 2019-06-22:
Hi, are you runing `vendors/espressif/esp-idf/tools/idf.py` from MSYS? I tested building OTA demo with CMake for espressif and does not have any problem, can you try that instead?

Thanks

pvyawaha, 2019-07-31:
Hi gregsochanik ,

Closing this issue now , if you still have same problem please reopen the issue.

closed 2019-07-31 by pvyawaha

MikeLoh, 2020-08-14:
I had the same issue - however posting the solution. If anyone else gets this error, check that you put final semicolon on the code signer certificate in the aws_ota_codesigner_certificate.h file.