aws/amazon-freertos #819: [MQTT] [Shadow JSON]: Error parsing JSON: JSMN error code -3

escanzanocloudcath, 2019-06-21:
Hi All,

I am using V1.4.8.

I am testing with this payload and am getting JSMN error code -3 somewhere in the queue.c code. When I try it with "p": "5f303b88-8aaf-4c68-9697-8b43979437d4ABasdfasdfasdfadsadfasdfasdfe34" it works. Something about this one character difference matters. Am I overflowing a buffer somewhere? 

I am attempting to process the delta queue and write a reported response.

{
	"state": {
		"desired": {
			"p": "5f303b88-8aaf-4c68-9697-8b43979437d4ABasdfasdfasdfadsadfasdfasdfe345"
		}
	}
}

Any ideas?
Thanks

tgsong, 2019-06-22:
Hi, error code -3 means you're passing a partial JSON (missing enclosing bracket, curly braces, etc.). Can you give more context of what you're passing and which character did you change or add?

escanzanocloudcath, 2019-06-22:
This problem was caused by me sending a payload to the /update topic that was bigger than xShadowQueueData.ulDataLength (line 265, aws_shadow_lightbulb_on_off.c).

closed 2019-06-22 by escanzanocloudcath