aws/amazon-freertos #820: Included version of TinyCBOR does not compile in IAR

tombresson, 2019-06-21:
**Describe the bug**
The included modified version of tinycbor does **not** compile in IAR.

**System information**
- IDE used: IAR 8.32.1
- Version of Amazon FreeRTOS: 201906.00_Major

**Expected behavior**

The modified version of TinyCBOR does not include the [support that was added for the IAR compiler](https://github.com/ARMmbed/tinycbor/commit/d90395095b20a115533bf60d76b4c3a4d7fd7973) in June 2018 and subsequently amended [very recently](https://github.com/ARMmbed/tinycbor/commit/218eae1562974149b04f2a15514dd78a6c2e1494).

The AWS version of the library header appears to have been modified to support the TI compiler, but does not include the necessary data for IAR.

**To reproduce**
Steps to reproduce the behavior:

1. Attempt to build TinyCBOR with IAR
2. Compiler error thrown in [compilersupport_p.h](https://github.com/aws/amazon-freertos/blob/757ff18c526071940701e103bbfca1576cc5863b/libraries/3rdparty/tinycbor/compilersupport_p.h#L154).

**Recommended Fix**

Pull in the latest version of [TinyCBOR](https://github.com/ARMmbed/tinycbor) and apply the modifications required to the `compilersupport_p.h` file.

tgsong, 2019-06-22:
Thanks for reporting this, it is a known issue to us and we will check the upstream fix.

qiutongs, 2019-07-11:
We are actively working on it.

qiutongs, 2019-07-17:
We have updated the tinycbor to its latest release. Next step is to copy the change of IAR support from ARMmbed/tinycbor.

closed 2019-07-24 by qiutongs